fl4p/batmon-ha #1: HA installation

AntoineLaporte, 2022-02-17:
Hi,

your add-on is just what I was looking (BlueTooth) for but I don't know how to install it in HA with the graphic interface.
You have linked this other project that is over USB ([Daly: similar add-on](https://github.com/MindFreeze/dalybms)) and there is a link to an HA repository that is working for me:
https://github.com/MindFreeze/home-assistant-addons

Would it be possible to do the same for your add-on ?

Thank you !

fl4p, 2022-02-17:
Hi,

I just created a new repository for this add-on. 
Try https://github.com/fl4p/home-assistant-addons for HA repository link.

Cheers

AntoineLaporte, 2022-02-17:
Thanks a lot, it will be a really nice add-on to have in my van, I'll try it in the next days and tell you.

AntoineLaporte, 2022-02-17:
Actually I got an error in the logs
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
ifelse: fatal: unable to exec ./run.sh: Permission denied
[cmd] ./run.sh exited 126
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.`

fl4p, 2022-02-17:
Nice, I'm using it in my van too.

I just pushed an update, try this

AntoineLaporte, 2022-02-17:
Same
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
ifelse: fatal: unable to exec ./run.sh: Permission denied
[cmd] ./run.sh exited 126
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.`

AntoineLaporte, 2022-02-17:
Just got the update on ha now and it works, I don't understand all the "errors" in the logs but I see that I get the data from the BT.
Thanks again for your work.

P.S.: I would put the possility to put the name of the device (DL-5XXXXXXXXX), I think it's not very hard for you as I was seeing it in the logs assiciated to the mac address. Also the lasts ":" just after the mac address in the logs could be confusing for people who don't know how a mac address should be.

AntoineLaporte, 2022-02-17:
I have some new errors

fl4p, 2022-02-19:
Hi,
with the next update you can use the device name (DL-...) for the address. Also I changed the log output of the discovered devices, I agree the extra ":" is confusing

fl4p, 2022-02-19:
Yes same here. The Daly BMS has poor current measuring. It shows 0A for currents below 0.3A.

fl4p, 2022-02-19:
The error you posted is related to reading temperature values.

What is your Daly BMS model and how many temperature sensors does it show in the app?

AntoineLaporte, 2022-02-19:
I got the 120A with one temp sensor.
The problem I got is that it was reading 0 and -4 and then 0 and again -4, etc...
I tried the UART cable addon and I do not have this problem with it.

fl4p, 2022-02-19:
Please try the update I just pushed

AntoineLaporte, 2022-02-19:
I will probably tomorrow. Thanks

AntoineLaporte, 2022-02-28:
I'm sorry I couldn't test your update because I burnt my BMS, I'm waiting for the new one and I'll test it right away !

AntoineLaporte, 2022-03-16:
Hey,

I finally got the BMS and testing it right now.
It looks good and the data updates are looking great.
Just my SOC is not changing much right now and I guess if it is the same there is no update of the sensor ?
Thank you !

AntoineLaporte, 2022-03-18:
Hi,

I've tried a bit of everything and I found 2 problems:
- if you put the stay alive option the Daly BT dongle is not accessible by anything else (which is normal) but if I restart the server it is also not accessible by the server anymore. I'm not sure if it happends alll the time but I think so. Then I have to unplug the dongle and and put it back to be able to reconnect
- like you can read it the attached text when it cannot get the connection multiple times in a row the addon stops, and the watchdog is not restarting it, I have to do it manually. If I check with my smartphone the BT is accessible but it takes time to discover it like a bit more than 1 second, maybe it would be great to extend the timeout of the search before it is declared as not found.

> ERROR    [main.py:171] Stack: Traceback (most recent call last):
  File "/app/main.py", line 164, in main
    raise exceptions[0]
  File "/app/main.py", line 159, in main
    await t
  File "/app/main.py", line 63, in sample_bms
    async with bms:
  File "/app/bmslib/bt.py", line 42, in __aenter__
    await self.connect()
  File "/app/bmslib/daly.py", line 56, in connect
    await super().connect()
  File "/app/bmslib/bt.py", line 21, in connect
    await self.client.connect()
  File "/usr/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 297, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 96:69:08:04:0B:2D could not be found. Try increasing `timeout` value or moving the device closer.
too many errors, abort
[cmd] ./run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

>

fl4p, 2022-03-19:
Hi

> * if you put the stay alive option the Daly BT dongle is not accessible by anything else (which is normal) but if I restart the server it is also not accessible by the server anymore. I'm not sure if it happens all the time but I think so. Then I have to unplug the dongle and and put it back to be able to reconnect
> >

This appears to be a Daly BT Dongle issue. I have this issue with Daly and JBD BMSs.

The add-on intentionally exits itself for a clean re-start after too many errors. Try to disable the watchdog and enable it again.

fl4p, 2022-03-19:
Timeout is set 10, will increase it to 20 with the next update

AntoineLaporte, 2022-03-19:
Everything else is working good, thank you again for this addon, it's way easier to just have a look to HA than having to open the app, and I love stats !

AntoineLaporte, 2022-03-23:
I'm actually experiencing some troubles with my phone BT presence detection, do you think that batmon could use "too much" BT so my Pi has not time to try to detect my phone ?

closed 2022-06-10 by AntoineLaporte