fl4p/batmon-ha #4: Add BMS controls

fl4p, 2022-08-05:
* Toggle charge / discharge FET

JK : https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/switch/jk_switch.cpp

NeariX67, 2022-08-25:
## JBD

<details><summary>BMS has password</summary>

#### Test: Read write mode will fail because password is required

-> DD 5A 00 02 56 78 FF 30 77 
<- DD 00 81 00 00 00 77 (FAIL)

#### Enter Password (Password is 333333)

-> DD 5A 06 07 06 03 03 03 03 03 03 FF DB 77
<- DD 06 00 00 00 00 FF (OK)

##### Explanation:
[2] 06: Command Address
[3] 07: Data length
[4] 06: Password length
[5-10] 03: Password digit

Data length is 1 higher than password length

#### Read write mode

-> DD 5A 00 02 56 78 FF 30 77
<- DD 00 00 00 00 00 77 (OK)

</details>

<details><summary>BMS has no password</summary>

#### Read write mode

-> DD 5A 00 02 56 78 FF 30 77
<- DD 00 00 00 00 00 77 (OK)

</details>

PMPoulsen, 2022-09-08:
I would love to see the Charge and Discharge controls for the FET's in JK BMS's implementad here. In this way HA could perform the logic controls of this function given knowledge of information on surplus PV power and decide on charging/discharge the battery.
I was looking into and ESP BTsolution when Istumbled over this great integration, keep up the good work.

fl4p, 2022-09-09:
@NeariX67 thanks for posting the commands

Anyone knows how to implement a control switch for home assistant?
[the esp-home integration](https://github.com/syssi/esphome-jk-bms) uses a code gen function from esphome.

fl4p, 2022-10-24:
In the dev branch https://github.com/fl4p/batmon-ha/tree/dev you'll find the add-on with switches working for JBD (tested) and jikong (untested)

NeariX67, 2022-10-24:
Thanks, will test out!

PMPoulsen, 2022-10-25:
Looking forward to a new release, I am not skilled enough to test your dev branch.

dezfutak, 2022-10-31:
> In the dev branch https://github.com/fl4p/batmon-ha/tree/dev you'll find the add-on with switches working for JBD (tested) and jikong (untested)

What's the easiest way to test this out (Home Assistant newbie here :) )?

fl4p, 2023-02-01:
The JK password protection is actually broken, because authentication is done by the App (the JK just sends the password in clear text and the app compares it to user input).
Did you try it while charging&discharging?

closed 2023-03-26 by fl4p