fl4p/batmon-ha #7: Integration not working for me... attaching logs

randrcomputers, 2022-08-06:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/app/main.py", line 17, in <module>
    from bmslib.sampling import BmsSampler
  File "/app/bmslib/sampling.py", line 6, in <module>
    from mqtt_util import publish_sample, publish_cell_voltages, publish_temperatures, publish_hass_discovery
  File "/app/mqtt_util.py", line 7, in <module>
    from util import get_logger
ModuleNotFoundError: No module named 'util'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

fl4p, 2022-08-06:
Hi, i just pushed an update 0.0.28, which fixes this.

randrcomputers, 2022-08-06:
Thank you!
That did fix that error here is logs now... I have a "battery 1" so it does see it but cannot connect? i also have battery 2, battery 3 and a battery 4 same as battery 1 but not looking like it can "see" those?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:49] BT Discovery:
INFO     [main.py:54] BT Device   LED-084CC2   address=78:9C:E7:08:4C:C2
INFO     [main.py:54] BT Device   7C-10-15-00-3C-7F   address=7C:10:15:00:3C:7F
INFO     [main.py:54] BT Device   40-C6-E5-42-15-AA   address=40:C6:E5:42:15:AA
INFO     [main.py:54] BT Device   54-BD-79-86-F1-36   address=54:BD:79:86:F1:36
INFO     [main.py:54] BT Device   PROV_E658DCv   address=4C:11:AE:E6:58:DE
INFO     [main.py:54] BT Device   battery 1   address=A4:C1:38:02:8B:41
INFO     [main.py:54] BT Device   62-42-4E-DF-70-0A   address=62:42:4E:DF:70:0A
INFO     [main.py:126] connecting mqtt XXXXXXX@core-mosquitto
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
INFO     [main.py:126] connecting mqtt XXXXXXX@core-mosquitto
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
ERROR    [sampling.py:70] jbd_bms error: Device with address 8e:c2:30:38:c1:a4 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
ERROR    [sampling.py:70] jbd_bms error: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:66] Error (num 1) reading BMS: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:67] Stack: Traceback (most recent call last):
  File "/app/main.py", line 62, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 44, in connect
    await super().connect()
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 8e:c2:30:38:c1:a4 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
ERROR    [sampling.py:70] jbd_bms error: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:66] Error (num 2) reading BMS: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:67] Stack: Traceback (most recent call last):
  File "/app/main.py", line 62, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 44, in connect
    await super().connect()
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 8e:c2:30:38:c1:a4 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
ERROR    [sampling.py:70] jbd_bms error: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:66] Error (num 3) reading BMS: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:67] Stack: Traceback (most recent call last):
  File "/app/main.py", line 62, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 44, in connect
    await super().connect()
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 8e:c2:30:38:c1:a4 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)
ERROR    [sampling.py:70] jbd_bms error: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:66] Error (num 4) reading BMS: Device with address 8e:c2:30:38:c1:a4 was not found.
ERROR    [main.py:67] Stack: Traceback (most recent call last):
  File "/app/main.py", line 62, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 44, in connect
    await super().connect()
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 8e:c2:30:38:c1:a4 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(8e:c2:30:38:c1:a4)

fl4p, 2022-08-06:
Try to unplug and re-connect the Bluetooth dongle from the BMS. Can you see it in the xiaoxiang app?

From the logs I can see that battery 1 has address A4:C1:38:02:8B:41 , but the add-on tries to connect to 8e:c2:30:38:c1:a4 . Did you try to enter A4:C1:38:02:8B:41 instead?

randrcomputers, 2022-08-06:
[image]

randrcomputers, 2022-08-06:
Tried A4:C1:38:02:8B:41 instead yes but fails. My HA has locked up 2 times with this last update not sure why had to cold boot Raspberry Pi4 twice. odd!

randrcomputers, 2022-08-07:
Got it to work on "battery 2"
How do you add more batteries? manual add to yaml?

INFO     [sampling.py:33] connecting bms JbdBt(A4:C1:38:EA:04:0C)
INFO     [sampling.py:39] connected bms JbdBt(A4:C1:38:EA:04:0C)!
INFO     [sampling.py:53] jbd_bms result@2022-08-07T12:29:18.977735 BmsSample(U=13.4V,I=-0.74A,P=-10W,q=61.8Ah/100,mos=nan°C)
INFO     [sampling.py:60] jbd_bms volt=3350,3352,3354,3356 temp=[19.7, 20.2]
INFO     [sampling.py:73] JbdBt(A4:C1:38:EA:04:0C) times: connect=7.30s fetch=0.56s
INFO     [sampling.py:53] jbd_bms result@2022-08-07T12:29:20.638645 BmsSample(U=13.4V,I=-0.99A,P=-13W,q=61.8Ah/100,mos=nan°C)
INFO     [sampling.py:60] jbd_bms volt=3350,3352,3354,3359 temp=[19.7, 20.3]
INFO     [sampling.py:73] JbdBt(A4:C1:38:EA:04:0C) times: connect=0.00s fetch=0.35s
INFO     [sampling.py:53] jbd_bms result@2022-08-07T12:29:22.113719 BmsSample(U=13.4V,I=-0.90A,P=-12W,q=61.8Ah/100,mos=nan°C)
INFO     [sampling.py:60] jbd_bms volt=3350,3352,3354,3354 temp=[19.7, 20.2]

fl4p, 2022-08-08:
I just pushed version 0.0.36, which introduces a new options schema. There you can add multiple BMS of the same kind

randrcomputers, 2022-08-08:
This is looking very promising! will try it tonight after i get home from work. THANK YOU!

randrcomputers, 2022-08-08:
Ok so reading 4 different batteries does work! i am having a lot of timeouts and bluetooth does have good signal at least with phone in same location as Pi. Also add-on seems to crash(to many errors?) after a couple min(currently have watchdog off) here are some logs:
INFO     [sampling.py:33] connecting bms JbdBt(A4:C1:38:FC:A1:DE)
INFO     [sampling.py:39] connected bms JbdBt(A4:C1:38:FC:A1:DE)!
INFO     [sampling.py:53] battery3 result@2022-08-08T17:49:14.636690 BmsSample(U=13.6V,I=-4.20A,P=-57W,q=99.2Ah/100,mos=nan°C)
INFO     [sampling.py:60] battery3 volt=3392,3381,3388,3392 temp=[19.9, 20.5]
INFO     [sampling.py:73] JbdBt(A4:C1:38:FC:A1:DE) times: connect=5.36s fetch=0.30s
INFO     [sampling.py:33] connecting bms JbdBt(A4:C1:38:02:8B:41)
INFO     [sampling.py:39] connected bms JbdBt(A4:C1:38:02:8B:41)!
INFO     [sampling.py:53] battery1 result@2022-08-08T17:49:21.682390 BmsSample(U=13.6V,I=-1.04A,P=-14W,q=99.5Ah/100,mos=nan°C)
INFO     [sampling.py:60] battery1 volt=3389,3390,3385,3393 temp=[19.7, 19.8]
INFO     [sampling.py:73] JbdBt(A4:C1:38:02:8B:41) times: connect=4.40s fetch=0.51s
INFO     [sampling.py:33] connecting bms JbdBt(A4:C1:38:10:69:29)
ERROR    [sampling.py:70] battery4 error: Device with address A4:C1:38:10:69:29 was not found.
INFO     [sampling.py:33] connecting bms JbdBt(A4:C1:38:EA:04:0C)
ERROR    [sampling.py:70] battery2 error: [org.bluez.Error.Failed] le-connection-abort-by-local
ERROR    [main.py:206] 2 exceptions occurred fetching BMSs
ERROR    [main.py:93] Error (num 5) reading BMS: Device with address A4:C1:38:10:69:29 was not found.
ERROR    [main.py:94] Stack: Traceback (most recent call last):
  File "/app/main.py", line 89, in fetch_loop
    await fn()
  File "/app/main.py", line 207, in fn
    raise exceptions[0]
  File "/app/main.py", line 202, in fn
    await t
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 44, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address A4:C1:38:10:69:29 was not found.
WARNING  [main.py:96] too many errors, abort
INFO     [main.py:216] Disconnecting JbdBt(A4:C1:38:EA:04:0C)
INFO     [main.py:216] Disconnecting JbdBt(A4:C1:38:02:8B:41)
INFO     [main.py:216] Disconnecting JbdBt(A4:C1:38:FC:A1:DE)
INFO     [main.py:216] Disconnecting JbdBt(A4:C1:38:10:69:29)
INFO     [main.py:224] exit signal handler...
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

randrcomputers, 2022-08-09:
Here is screenshot of rough dashboard in HA
[image]

fl4p, 2022-08-09:
Ok, i'll increase the max error count before exit.

Try to enable `keep_alive`, this will evade connection errors (but you will no be able to connect to the BMS with your phone anymore)

fl4p, 2022-08-09:
> [image]

What app is this?

randrcomputers, 2022-08-09:
“Bms tool” in App Store its a clone of the “official xiaoxiang app” I prefer the clone app as you can see battery percentages on main Screen

randrcomputers, 2022-08-09:
So I was getting a lot of timeouts/errors so I installed external Bluetooth with 10’ extension and it is now working perfectly! No errors! Seems the Pi has a lot of interference for Bluetooth or something.

closed 2022-08-18 by fl4p