fl4p/batmon-ha #8: Remote Pi - Error Reading BMS - Already waiting

Ghostkls, 2022-08-10:
Firstly, thank you for your amazing work on this project!
I am having an issue with the remote Pi installation. Removing and re-adding the installation works after a few attempts until a device reboot. After which the "Error Reading BMS - Already Waiting" appears. Cannot run as a service as well. Attached is a screenshot.
[image]

RoySalisbury, 2022-08-12:
I attempted to run the code on MacOS and got the same error (3 already waiting).  The PI is hit or miss in connecting to the device, but I upped the max_error to 999 and it will finally connect.

Ghostkls, 2022-08-12:
I noted the same, had more "success" with disabling Bluetooth first, starting the script  and then starting Bluetooth while the script is looping. Very hit and miss though. Have not had nay success with the startup service so I leave it running in the terminal. Once running though, it runs great!

Ghostkls, 2022-08-12:
I can confirm the fix uploaded about an hour ago has resolved the issue. Working perfectly now as both manually running as well as via service.

RoySalisbury, 2022-08-12:
Same here... even working on my MacOS (manual, have not tried service).

fl4p, 2022-08-15:
Please try the latest git version

Ghostkls, 2022-08-15:
Working absolutely perfectly since the update. Grabbed the fix minutes after you uploaded and been running 100%!
Thank you for your quick fix!

fl4p, 2022-08-18:
Great to hear that. Just saw you already figured it days before my last comment 😄

closed 2022-08-18 by fl4p