fl4p/batmon-ha #10: soc-current/soc_power are inverted

RoySalisbury, 2022-08-12:
I would expect during discharge that the current and power would be negative (e.g, -3.5) and during charging they would be positive.  But here they are reversed..

When I read the data using the iOS app that is the way they are shown.  And when I read them in my own code (C# serial/rs485), they are correct as well.

fl4p, 2022-08-15:
This depends on how you define direction of battery current, which is arbitrary.
Here I assume that current flow is positive during discharge, as the technical flow is from bat(+) through the load to bat(-).
Some BMS and Charger use the inverse direction.

I might add an option so you can inverse the direction.

RoySalisbury, 2022-08-15:
OK.. for now I just went into the source code and removed the "-" in front of the current reading ... That fixes if for me.

All my shunts and devices report it this way (negative during discharge), so its now consistent for me.

fl4p, 2022-08-18:
I've just pushed version 0.0.37 which adds an option to invert the current