fl4p/batmon-ha #11: Entity rounding

proasnet, 2022-08-18:
Hi, first of all thanks for your work, the plugin is great. At the same time, I would like to request from the rounding of values ​​in entities that sometimes, but not always, have a lot of decimal places added, as can be seen in the attached picture from the HA panel.
I have JK BMS and in the near time will be the second the same time.
[image]

fl4p, 2022-08-18:
Hi, I just pushed an update here on github which rounds all numbers to 5 significant places

randrcomputers, 2022-08-18:
Not seeing any update still showing .36 as latest

proasnet, 2022-08-19:
I don't see any update either.
5 decimal places is too much. 3 decimal places would be ideal, no data from BMS has more.

fl4p, 2022-08-28:
I just published the update, please try 0.0.39

proasnet, 2022-08-28:
Hi, how can I update current version 0.0.36?
HA not inform me about update available of your addon.
I must uninstall and install new version again?
Thank you.

randrcomputers, 2022-08-28:
Thanks for the update. Testing it now.

fl4p, 2022-08-28:
> Hi, how can I update current version 0.0.36? HA not inform me about update available of your addon. I must uninstall and install new version again? Thank you. Dne ne 28. 8. 2022 11:57 uživatel Fabian ***@***.***> napsal:
> […](#)

In the add-on store click the 3 dots in the upper right corner and check for updates. Then wait a bit and reload the page and the update should appear