fl4p/batmon-ha #12: Two the same JK BMS B2A24S150P not working at the same time

proasnet, 2022-08-18:
Hi, today I trying add the second BMS and it not working.
Config:
devices:
  - address: C8:47:8C:E4:51:FA
    type: jk
    alias: battery1
    pin: "2580"
  - address: C2:6C:77:DD:4A:C7
    type: jk
    alias: battery2
    pin: "2580"
mqtt_user: jkbms
mqtt_password: mypassword
mqtt_broker: core-mosquitto
concurrent_sampling: true
keep_alive: true
verbose_log: true
sample_period: 1

Log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:76] BT Discovery:
INFO     [main.py:79]  - no devices found - 
INFO     [main.py:119] Verbose logging enabled
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:51:FA) keep alive enabled
INFO     [bt.py:53] BMS JKBt(C2:6C:77:DD:4A:C7) keep alive enabled
INFO     [main.py:160] connecting mqtt jkbms@core-mosquitto
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [sampling.py:33] connecting bms JKBt(C8:47:8C:E4:51:FA)
DEBUG    [jikong.py:94] starting scan
DEBUG    [scanner.py:132] cached devices: {'/org/bluez/hci0/dev_C8_47_8C_E4_51_FA': {'Address': 'C8:47:8C:E4:51:FA', 'AddressType': 'public', 'Name': 'Bat1', 'Alias': 'Bat1', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -92, 'Connected': True, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'Modalias': 'usb:v045Ep0040d0300', 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {2917: b'\x88\xa0\xc8G\x8c\xe4Q\xfa'}, 'ServicesResolved': True}}
DEBUG    [scanner.py:282] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
DEBUG    [jikong.py:110] retry after error Device C8:47:8C:E4:51:FA not discovered (set())
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
DEBUG    [jikong.py:110] retry after error Device C8:47:8C:E4:51:FA not discovered (set())
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
DEBUG    [jikong.py:110] retry after error Device C8:47:8C:E4:51:FA not discovered (set())
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
DEBUG    [jikong.py:110] retry after error Device C8:47:8C:E4:51:FA not discovered (set())
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
DEBUG    [jikong.py:110] retry after error Device C8:47:8C:E4:51:FA not discovered (set())

fl4p, 2022-08-18:
Hi, what device to you run your HA on?
Does it work when you remove the 2nd BMS?
It appears that the add-on cannot see any of the BMS. Try to find them in the mobile app and restart home assistant

proasnet, 2022-08-19:
Hi, thanks for your reply.
I have HA running on Generic x86 with HASOS. I have a BT module according to the recommendation for BT integration, ASUS USB-BT500.
When I have 1 BMS set up, it usually but not always connects. Working fine.
I used the BT scan app on my mobile yesterday and the BMS have the following addresses:
The first: C8:47:8C:E4:51:FA
The second, added: C8:47:8C:E4:52:9D
As you can see, I set the second MAC incorrectly in the first configuration. I fixed this but it didn't fix the problem. Although it finds the second BMS in the log, it does not connect.

I am sending the data again, current.

Config:
devices:
  - address: C8:47:8C:E4:51:FA
    type: jk
    alias: battery1
    pin: "2580"
  - address: C8:47:8C:E4:52:9D
    type: jk
    alias: battery2
    pin: "2580"
mqtt_user: jkbms
mqtt_password: mypassword
mqtt_broker: core-mosquitto
concurrent_sampling: false
keep_alive: true
verbose_log: false
sample_period: 1
----------------------------------------------------------------------------------------------------------------
Log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:76] BT Discovery:
INFO     [main.py:81] BT Device   Bat1   address=C8:47:8C:E4:51:FA
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:51:FA) keep alive enabled
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:52:9D) keep alive enabled
INFO     [main.py:160] connecting mqtt jkbms@core-mosquitto
INFO     [sampling.py:33] connecting bms JKBt(C8:47:8C:E4:52:9D)
ERROR    [sampling.py:70] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'C8:47:8C:E4:51:FA'})
INFO     [sampling.py:33] connecting bms JKBt(C8:47:8C:E4:51:FA)
INFO     [sampling.py:39] connected bms JKBt(C8:47:8C:E4:51:FA)!
INFO     [sampling.py:53] battery1 result@2022-08-19T06:42:48.816902 BmsSample(U=26.4V,I=-11.00A,P=-290W,q=189.8Ah/282,mos=29.0°C)
INFO     [sampling.py:60] battery1 volt=3296,3297,3297,3298,3299,3294,3296,3299 temp=[25.5, 26.8]
INFO     [sampling.py:73] JKBt(C8:47:8C:E4:51:FA) times: connect=27.32s fetch=0.58s
ERROR    [main.py:206] 1 exceptions occurred fetching BMSs
ERROR    [main.py:93] Error (num 1) reading BMS: Device C8:47:8C:E4:52:9D not discovered ({'C8:47:8C:E4:51:FA'})
ERROR    [main.py:94] Stack: Traceback (most recent call last):
  File "/app/main.py", line 89, in fetch_loop
    await fn()
  File "/app/main.py", line 207, in fn
    raise exceptions[0]
  File "/app/main.py", line 202, in fn
    await t
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 102, in connect
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'C8:47:8C:E4:51:FA'})
INFO     [sampling.py:33] connecting bms JKBt(C8:47:8C:E4:52:9D)
ERROR    [sampling.py:70] battery2 error: Device C8:47:8C:E4:52:9D not discovered (set())
INFO     [sampling.py:53] battery1 result@2022-08-19T06:42:59.906608 BmsSample(U=26.4V,I=-10.81A,P=-285W,q=189.8Ah/282,mos=29.0°C)
INFO     [sampling.py:60] battery1 volt=3296,3297,3297,3298,3298,3294,3296,3299 temp=[25.5, 26.8]
INFO     [sampling.py:73] JKBt(C8:47:8C:E4:51:FA) times: connect=0.00s fetch=0.21s
ERROR    [main.py:206] 1 exceptions occurred fetching BMSs
ERROR    [main.py:93] Error (num 2) reading BMS: Device C8:47:8C:E4:52:9D not discovered (set())
ERROR    [main.py:94] Stack: Traceback (most recent call last):
  File "/app/main.py", line 89, in fetch_loop
    await fn()
  File "/app/main.py", line 207, in fn
    raise exceptions[0]
  File "/app/main.py", line 202, in fn
    await t
  File "/app/bmslib/sampling.py", line 24, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 37, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 102, in connect
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered (set())
INFO     [sampling.py:53] battery1 result@2022-08-19T06:43:00.951845 BmsSample(U=26.4V,I=-11.00A,P=-290W,q=189.8Ah/282,mos=29.0°C)
INFO     [sampling.py:60] battery1 volt=3296,3298,3296,3297,3299,3294,3294,3298 temp=[25.5, 26.8]
INFO     [sampling.py:73] JKBt(C8:47:8C:E4:51:FA) times: connect=0.00s fetch=0.04s
INFO     [sampling.py:33] connecting bms JKBt(C8:47:8C:E4:52:9D)

fl4p, 2022-08-28:
Hi, please try latest version 0.0.39

proasnet, 2022-08-28:
HA not inform about addon update possibility.
I tried uninstall and install addon again and I have 0.0.36 again. Not 0.0.39

fl4p, 2022-08-28:
In the add-on store click the 3 dots in the upper right corner and check for updates. Then wait a bit and reload the page and the update should appear

proasnet, 2022-08-28:
Okay. Thanks. Updated. But after add secod bms to config, log show:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:65] BT Discovery:
INFO     [main.py:68]  - no devices found - 
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:51:FA) keep alive enabled
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:52:9D) keep alive enabled
INFO     [main.py:149] connecting mqtt jkbms@core-mosquitto
INFO     [main.py:166] Fetching 2 BMS + 0 others serially, period=5.00s, keep_alive=True
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
```
Now nothing work.

proasnet, 2022-08-28:
Is possible make the second addon as second instance? Every instace fot only one bms?

fl4p, 2022-08-28:
please uninstall the add-on first and then install the old version

proasnet, 2022-08-28:
Now I can install only 0.0.39, 36 is not in your repositary.
Oh. I dont know what is wrong.
My mobile phone with ENJPOWER working. If I start addon, old or new
version, I see that BMS is periodically connecting, led from flashing
lighting and double beep. If I stop addon, I see both bms with app.
I tried 2 day old total backup, but nothing. Not work.
I tried another usb port for BT module, not work.
Something is wrong and i dont know what.

proasnet, 2022-08-28:
verbose log:
```
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [client.py:387] Connection successful (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA)
DEBUG    [client.py:734] Waiting for ServicesResolved (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA)
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA): ['org.bluez.Device1', {'ServicesResolved': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [jikong.py:144] write b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11'
DEBUG    [client.py:967] Write Characteristic 0000ffe1-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011: b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11'
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'U\xaa\xeb\x90\x037JK-B2A24S15P\x00\x00\x00\x0010.XW\x00\x00\x0010.08\x00\x00\x00\xc8\x17\x11\x00\x05\x00\x00\x00Bat1\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x002580\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00220815\x00\x002042102022\x000000\x00Input Userdata\x00\x00801105\x00\x00\x00\x00')>}, []]
DEBUG    [jikong.py:60] header, clear buf bytearray(b'')
DEBUG    [jikong.py:65] bms msg(128) (buf128): 55 aa eb 90 3 37 4a 4b 2d 42 32 41 32 34 53 31 35 50 0 0 0 0 31 30 2e 58 57 0 0 0 31 30 2e 30 38 0 0 0 c8 17 11 0 5 0 0 0 42 61 74 31 0 0 0 0 0 0 0 0 0 0 0 0 32 35 38 30 0 0 0 0 0 0 0 0 0 0 0 0 32 32 30 38 31 35 0 0 32 30 34 32 31 30 32 30 32 32 0 30 30 30 30 0 49 6e 70 75 74 20 55 73 65 72 64 61 74 61 0 0 38 30 31 31 30 35 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
DEBUG    [jikong.py:65] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
DEBUG    [jikong.py:65] bms msg(128) (buf278): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00T')>}, []]
DEBUG    [jikong.py:65] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 54
DEBUG    [jikong.py:78] got response 3 (len300)
DEBUG    [jikong.py:144] write b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10'
DEBUG    [client.py:967] Write Characteristic 0000ffe1-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011: b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10'
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00D')>}, []]
DEBUG    [jikong.py:65] bms msg(20) (buf20): aa 55 90 eb c8 1 1 0 0 0 0 0 0 0 0 0 0 0 0 44
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'U\xaa\xeb\x90\x017X\x02\x00\x00(\n\x00\x00Z\n\x00\x00\x10\x0e\x00\x00\xde\r\x00\x00\x1e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\x808\x01\x00\x1e\x00\x00\x00<\x00\x00\x00\xf0I\x02\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xd0\x07\x00\x00\xf [... cut ...] 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 9e 63 0 0 21 8 5 0 7d cd
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E4_51_FA/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xff\xff\xfb\x00\t\x01$\x01\x00\x00\x00\x00\x00:[\x85\x02\x00\x90M\x04\x00')>}, []]
DEBUG    [jikong.py:65] bms msg(22) (buf150): ff ff fb 0 9 1 24 1 0 0 0 0 0 3a 5b 85 2 0 90 4d 4 0
```

fl4p, 2022-08-30:
Hi, I just pushed some fixes. please try the version from the git repo here.

fl4p, 2022-08-30:
I was not able to test the changes, since I currently don't have access to a JK BMS

proasnet, 2022-08-31:
Hi @fl4p ,
I tried install batmon back, not working.
Log:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:65] BT Discovery:
INFO     [main.py:68]  - no devices found - 
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:51:FA) keep alive enabled
INFO     [bt.py:53] BMS JKBt(C8:47:8C:E4:52:9D) keep alive enabled
INFO     [main.py:149] connecting mqtt jkbms@core-mosquitto
INFO     [main.py:166] Fetching 2 BMS + 0 others serially, period=1.00s, keep_alive=True
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:52:9D)
ERROR    [sampling.py:75] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'79:0B:8A:58:42:47'})
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
ERROR    [sampling.py:75] battery1 error: Device C8:47:8C:E4:51:FA not discovered ({'79:0B:8A:58:42:47'})
ERROR    [main.py:200] 2 exceptions occurred fetching BMSs
ERROR    [main.py:82] Error (num 1) reading BMS: Device C8:47:8C:E4:52:9D not discovered ({'79:0B:8A:58:42:47'})
ERROR    [main.py:83] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 92, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E4:52:9D was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 78, in fetch_loop
    await fn()
  File "/app/main.py", line 201, in fn
    raise exceptions[0]
  File "/app/main.py", line 196, in fn
    await t
  File "/app/bmslib/sampling.py", line 25, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 38, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 94, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/jikong.py", line 119, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'79:0B:8A:58:42:47'})
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:52:9D)
ERROR    [sampling.py:75] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'79:0B:8A:58:42:47'})
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
```

I tried using jk-bms-ble for ESP32 by @sissy, ​​with two bms connected via bluetooth and it works.
https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example-multiple-devices.yaml

fl4p, 2022-09-06:
I just pushed 0.0.44, please try this version. First with a single BMS, then add the second

fl4p, 2022-09-07:
If it still doesn't work, please try to use an external Bluetooth USB dongle

proasnet, 2022-09-07:
Hi, I have ASUS USB BT connected to HA, to Generic x86 as HASSOS.
I tried only one bms, periodically beeping ( connecting bt slave )
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:65] BT Discovery:
INFO     [main.py:70] BT Device   Bat1   address=C8:47:8C:E4:51:FA
INFO     [main.py:149] connecting mqtt jkbms@core-mosquitto
INFO     [main.py:166] Fetching 1 BMS + 0 others serially, period=5.00s, keep_alive=False
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
INFO     [jikong.py:111] normal connect failed, connecting with scanner
INFO     [sampling.py:40] connected bms JKBt(C8:47:8C:E4:51:FA)!
INFO     [sampling.py:57] battery1 result@2022-09-07T16:11:20.395555 BmsSample(U=26.7V,I=7.82A,P=209W,q=183.4Ah/282.0,mos=28.7°C)
INFO     [sampling.py:64] battery1 volt=3335,3336,3335,3335,3336,3341,3335,3335 temp=[23.7, 27.2]
INFO     [sampling.py:78] JKBt(C8:47:8C:E4:51:FA) times: connect=28.03s fetch=0.75s
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
INFO     [jikong.py:111] normal connect failed, connecting with scanner
```
And with two bms
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:65] BT Discovery:
INFO     [main.py:70] BT Device   20-AB-B1-90-EA-6F   address=20:AB:B1:90:EA:6F
INFO     [main.py:149] connecting mqtt jkbms@core-mosquitto
INFO     [main.py:166] Fetching 2 BMS + 0 others serially, period=5.00s, keep_alive=False
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
INFO     [sampling.py:40] connected bms JKBt(C8:47:8C:E4:51:FA)!
INFO     [sampling.py:57] battery1 result@2022-09-07T16:14:11.853280 BmsSample(U=26.7V,I=6.25A,P=167W,q=183.7Ah/282.0,mos=28.5°C)
INFO     [sampling.py:64] battery1 volt=3334,3334,3334,3336,3335,3338,3333,3334 temp=[23.7, 27.2]
INFO     [sampling.py:78] JKBt(C8:47:8C:E4:51:FA) times: connect=4.94s fetch=0.35s
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:52:9D)
INFO     [jikong.py:111] normal connect failed, connecting with scanner
ERROR    [sampling.py:75] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'20:AB:B1:90:EA:6F', 'C8:47:8C:E4:51:FA', '1E:51:28:D5:EE:8F'})
ERROR    [main.py:200] 1 exceptions occurred fetching BMSs
ERROR    [main.py:82] Error (num 1) reading BMS: Device C8:47:8C:E4:52:9D not discovered ({'20:AB:B1:90:EA:6F', 'C8:47:8C:E4:51:FA', '1E:51:28:D5:EE:8F'})
ERROR    [main.py:83] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 109, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 19, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E4:52:9D was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 78, in fetch_loop
    await fn()
  File "/app/main.py", line 201, in fn
    raise exceptions[0]
  File "/app/main.py", line 196, in fn
    await t
  File "/app/bmslib/sampling.py", line 25, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 38, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 40, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 112, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/jikong.py", line 136, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'20:AB:B1:90:EA:6F', 'C8:47:8C:E4:51:FA', '1E:51:28:D5:EE:8F'})
INFO     [sampling.py:34] connecting bms JKBt(C8:47:8C:E4:51:FA)
```

Still not working....
I remember, that jkbms on esphome working properly with both bmss.

proasnet, 2022-09-08:
I found that after testing, on both bms was recalibrated current sensors to zero and then not sensing current.
I am sorry, I am canceling testing with batmon, there are too much problems.
Thanks.

closed 2022-09-08 by proasnet

fl4p, 2022-09-09:
Try to enable keep_alive, this will never close the connection to the BMS.
Implementing BLE software on generic hardware is tricky, also some OSes appear to have poor BLE implementation.

No idea how the recalibration was triggered, batmon does not have the commands for this.

proasnet, 2022-10-05:
Today tested after your updates. Still not working with two the same JKBMS
```
Traceback (most recent call last):
  File "/app/main.py", line 79, in fetch_loop
    await fn()
  File "/app/main.py", line 206, in fn
    raise exceptions[0]
  File "/app/main.py", line 201, in fn
    await t
  File "/app/bmslib/sampling.py", line 29, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 42, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 80, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'5C:89:C4:FF:E2:59', 'C8:47:8C:E4:51:FA'})
WARNING  [mqtt_util.py:106] mqtt msg battery1/soc/total_voltage not published: (0, 39)
INFO     [sampling.py:61] battery1 result@2022-10-05T12:18:32.683396 BmsSample(U=25.9V,I=-17.13A,P=-444W,q=38.9Ah/280.0,mos=24.7°C)
INFO     [sampling.py:68] battery1 volt=3241,3242,3234,3237,3244,3240,3236,3244 temp=[22.0, 21.5]
INFO     [sampling.py:93] JKBt(C8:47:8C:E4:51:FA) times: connect=0.00s fetch=0.16s
INFO     [sampling.py:38] connecting bms JKBt(C8:47:8C:E4:52:9D)
INFO     [jikong.py:114] normal connect failed, connecting with scanner
ERROR    [sampling.py:90] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
ERROR    [main.py:205] 1 exceptions occurred fetching BMSs
ERROR    [main.py:83] Error (num 2) reading BMS: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
ERROR    [main.py:84] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 22, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E4:52:9D was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 79, in fetch_loop
    await fn()
  File "/app/main.py", line 206, in fn
    raise exceptions[0]
  File "/app/main.py", line 201, in fn
    await t
  File "/app/bmslib/sampling.py", line 29, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 42, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 80, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
WARNING  [mqtt_util.py:106] mqtt msg battery1/soc/total_voltage not published: (0, 51)
INFO     [sampling.py:61] battery1 result@2022-10-05T12:18:51.723148 BmsSample(U=25.9V,I=-18.58A,P=-481W,q=38.9Ah/280.0,mos=24.7°C)
INFO     [sampling.py:68] battery1 volt=3235,3236,3237,3235,3235,3240,3237,3237 temp=[22.0, 21.5]
INFO     [sampling.py:93] JKBt(C8:47:8C:E4:51:FA) times: connect=0.00s fetch=0.12s
INFO     [sampling.py:38] connecting bms JKBt(C8:47:8C:E4:52:9D)
INFO     [jikong.py:114] normal connect failed, connecting with scanner
ERROR    [sampling.py:90] battery2 error: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
ERROR    [main.py:205] 1 exceptions occurred fetching BMSs
ERROR    [main.py:83] Error (num 3) reading BMS: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
ERROR    [main.py:84] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 22, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E4:52:9D was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 79, in fetch_loop
    await fn()
  File "/app/main.py", line 206, in fn
    raise exceptions[0]
  File "/app/main.py", line 201, in fn
    await t
  File "/app/bmslib/sampling.py", line 29, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 42, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 80, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E4:52:9D not discovered ({'25:B6:C3:99:20:BA'})
INFO     [sampling.py:38] connecting bms JKBt(C8:47:8C:E4:52:9D)
INFO     [jikong.py:114] normal connect failed, connecting with scanner
```

proasnet, 2022-11-13:
I have to praise you. After some time I tested if reading from 2 BMS works and it works without problem.
Thank you for the good work.

proasnet, 2022-11-13:
The only thing I found lacking is the number of battery cycles, but all other working.