fl4p/batmon-ha #14: If disconected BMS device, HA device has not status unavailable

proasnet, 2022-08-19:
Hi, the third my issue is, that if is BMS unavailable with BATMON, and Addon running, the device and entities has not status unavailable, and in the history, I see the last values. In the graphs, I see horizontal lines.
Please use availability option for the device.
Thank you.

fl4p, 2022-08-28:
How do you send "unavailable" status messages through MQTT? I could not find it in the Home assistant docs

AndyRap, 2022-09-12:
Hi,

Check out the MQTT info here, with specific relation to the availability_topic: and its options:

https://www.home-assistant.io/integrations/sensor.mqtt

Something like:

`# Example configuration.yaml entry
mqtt:
  sensor:
    - name: "RSSI"
      state_topic: "home/sensor1/infojson"
      unit_of_measurement: "dBm"
      value_template: "{{ value_json.RSSI }}"
      availability:
        - topic: "home/sensor1/status"
      payload_available: "online"
      payload_not_available: "offline"
      json_attributes_topic: "home/sensor1/attributes"`

Hope that helps.

fl4p, 2022-09-19:
@AndyRap thanks. I just pushed an update here which uses the `expire_after` API. so values invalidate after 30 seconds or so.