fl4p/batmon-ha #16: Daly BMS Error

alextasmota, 2022-09-04:
Hi There,

Thank you for creating this project.  It is exactly what I was looking for :).

I'm attempting to connect to a new Daly BMS that arrived about a week ago.  I was initially receiving an error stating "BleakError: Client is already connected."  After turning the loggin up I can see that the actual error (as deep into the code that I can see) is "reading BMS: Characteristic with UUID 17 could not be found!"

This occurs at line 'await self.client.start_notify(self.UUID_RX, self._notification_callback)' in bmslib/daly.py

I can see that the UUID 17 is defined further up that file but that's as far as I have got so far.

Any help would be appreciated.  I'm trying to research how you determine those values.

Thanks, Alex

alextasmota, 2022-09-04:
Output from service_explorer.py

INFO:__main__:Connected: True
INFO:__main__:[Service] 0000fff0-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific
INFO:__main__:  [Characteristic] 0000fff2-0000-1000-8000-00805f9b34fb (Handle: 19): Vendor specific (read,write-without-response,write), Value: b''
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 21): Client Characteristic Configuration) | Value: b''
INFO:__main__:          [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 22): Characteristic User Description) | Value: b'spss_rx_desc'
INFO:__main__:  [Characteristic] 0000fff1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (read,notify), Value: b''
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration) | Value: b'\x00\x00'
INFO:__main__:          [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 18): Characteristic User Description) | Value: b'spss_tx_desc'
INFO:__main__:[Service] 02f00000-0000-0000-0000-00000000fe00 (Handle: 23): Unknown
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff03 (Handle: 24): Unknown (read), Value: b'\x02\x02'
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff00 (Handle: 29): Unknown (read), Value: b''
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff01 (Handle: 31): Unknown (write-without-response,write), Value: None
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff02 (Handle: 26): Unknown (read,notify), Value: b''
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration) | Value: b''
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff05 (Handle: 36): Unknown (read,write-without-response,write), Value: b'\x00\x00'
INFO:__main__:  [Characteristic] 02f00000-0000-0000-0000-00000000ff04 (Handle: 33): Unknown (read,write-without-response,write,notify), Value: b''
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 35): Client Characteristic Configuration) | Value: b''
INFO:__main__:[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
INFO:__main__:  [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11): Service Changed (read,indicate), Value: b'\x01\x00\xff\xff'
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration) | Value: b'\x02\x00'

alextasmota, 2022-09-04:
Ok I have it reading the data by altering the UUID as follows:

UUID_RX = 15
UUID_TX = 19

and then I found a hardcoded value on the line that starts:

await self.client.write_gatt_char(

I changed the first parameter to 31.

I suspect a new hardware or software version has been released.

fl4p, 2022-09-06:
Great that you figured it out, that helped a lot. I'll push an update soon

fl4p, 2022-09-06:
I just pushed version 0.0.44, which tries to detect the correct uuids. Please let me know whether it works for you

alextasmota, 2022-09-06:
Hi, Yes.  After pulling down the latest version it connects to the BMS without any modiciations to the code.  Looks like a fair few changes in there as a few files have changed significantly in size.  Thanks.

fl4p, 2022-09-07:
Ok great, i am closing this one

closed 2022-09-07 by fl4p