fl4p/batmon-ha #17: MQQT diconnects on timeout if bms not answer servel times.

Niceman35, 2022-09-16:
MQQT connection is closing on my system after several hours. 
In MQQT logs where is a message "Disconnected on timeout"
This most likely happens if BMS is not sending or not updating values for some time.

fl4p, 2022-09-19:
Hi,
can you please post the exact log (anonymised) here?

Niceman35, 2022-09-19:
Hi. I don't have this issue now.
I changes your code locally and added mqtt_client.loop_start() after connect in main.py.
Works stable after this fix. I think you should add this too. This is the correct connect routine used in documentation.

    try:
        mqtt_client.connect(user_config.mqtt_broker, port=1883)
>>        mqtt_client.loop_start()
    except Exception as ex:
        logger.error('mqtt connection error %s', ex)

fl4p, 2022-09-20:
Thanks, I've included this in the current release 0.0.45

closed 2022-10-10 by fl4p