fl4p/batmon-ha #20: Error on install

JeffyBeepBoop, 2022-09-25:
"The command '/bin/ash -o pipefail -c apk add --no-cache python3 bluez py-pip git' returned a non-zero code: 4"

fl4p, 2022-10-10:
Never heard of exit code 4. Maybe a permission / disk space error? 

To track this down, please edit `Dockerfile` and replace the lines
```
RUN \
  apk add --no-cache \
    python3 bluez py-pip git
```

with 
```
RUN apk add --no-cache python3
RUN apk add --no-cache py-pip 
RUN apk add --no-cache bluez 
RUN apk add --no-cache git

```

JeffyBeepBoop, 2022-11-04:
Sorry, I'm kinda new to how add-ons work. How would I edit that file within the flow of using the Add-on/repo? When I add the repo, does it download it somewhere? I can also install a separate Docker container. 

Just checked again and still getting the same error after clicking install. Could it be something to do with 32bit Raspberry PiOS?

JeffyBeepBoop, 2022-11-04:
Made a fork of the repo to test the changes you suggested. Same error. 

My disk, according to HA is only 40% full, so should have space. 

[image]

DryZus, 2022-12-09:
I have the same issue.

fl4p, 2023-02-03:
Please try version 0.0.50, which doesn't fix the error but gives more insights on what package actually causes the error

fl4p, 2023-02-23:
Closed due to inactivity. Please create a new issue if still present

closed 2023-02-23 by fl4p