fl4p/batmon-ha #22: Is it possible to use RS485 to USB Serial converter ?

prezes210, 2022-10-22:
I have JKBMS with successfully connected RS485 to USB Serial converter. This usb device is visible on HA as device `/dev/USB000`.
Is it possible to use this connection instead of Bluetooth? I might be more stable than bluetooth

I have used converters tested by Andy from offgrid garage in this video: https://www.youtube.com/watch?v=sHIlP2oJ3DU

prezes210, 2022-11-04:
For ppl searching the possibility to use a wired connection to JK-BMS this seems to be an interesting option:
https://github.com/syssi/esphome-jk-bms

aremishevsky, 2025-02-09:
Do we have any luck/progress with this feature?

typxxi, 2025-03-27:
is there any update if we can replace the bluetooth connection with the RS485 provided by the JK BMS ?

I have the jk inverter BMS V15 with the 2 ampere onboard balancer but I have every second day for minutes or hours bat-mon outages even thought the bt adapter antenna sits in front of the battery next to the BMS which is inside the EEL case.

I would like to use 1 of the 2 ports on the right labeled as RS485-2 (usually used for the communication between multiple battery packs.

[image]

thanks for any suggestions

fl4p, 2025-03-28:
> Do we have any luck/progress with this feature?

There has been some progress on this, see https://github.com/fl4p/batmon-ha/issues/225#issuecomment-2761926278

typxxi, 2026-05-29:
Ok , thanks a lot for your great and fast reply. 
Here the physical ports of the JK inverter BMS V15 for a better orientiation where the RJ11 RS232 single port is located in the middle.

[image]

1. I have understood that the modbus / RS485 chain / string **does not work** 
2. Hence I would need 2 cables and FTDI adapters to support the laptop and the Pi5 Home Assistant.
3. I also would not change my already running RJ 45 / RS485-2 cable USB FTDI adapter for the Windows Laptop

Therefore I would go for a RJ11 port (JK BMS V15 ) RS232 to USB A FTDI adapter to the Pi5 for the JK BMS Batmon , right ?
I would have to look for such [RJ11 connector with USB A to FTDI to adapter](https://www.ebay.de/itm/287030668219)
"Domotica P1 Poort Kabel FTDI USB-RJ11 1,8 m" which looks like this (8€ free delivery on ebay.de - 4€ cheaper than on ali Express and half the price of Amazon):

[image]

Seems to be used as reader for a smartmeter, but should do the trick for JK BMS too.

[image]

Thanks a lot

fl4p, 2026-09-03:
The original request is implemented: `type: jk_uart` with `address: serial`, `adapter: /dev/ttyUSB0` and `uart: true`, see the Serial / RS485 section of the README. Closing this one.

For the remaining threads: the JK RJ11 RS232 port on V15 is GPS-only, use the RJ45 RS485-2 port with a USB-RS485 adapter. The JK RS485 protocol is not Modbus, so it cannot pass through the evcc Modbus proxy. The device fields (`note`, `current_calibration`, `adapter`) are in the add-on schema since 1.98; if the editor shows a stale value after save + restart, reload the page.

Anything new, please open a fresh issue.

closed 2026-09-03 by fl4p