fl4p/batmon-ha #24: Fehlermeldung

trollingtom, 2022-10-22:
Hallo ich bekomme immer folgende Meldung:

INFO     [jikong.py:114] normal connect failed, connecting with scanner

Woran kann das liegen? Gruss Thomas

trollingtom, 2022-10-22:
Hier mal das laufende Protokoll:

Also auf der App von jk wird das bms nicht mehr angezeigt. Hier Mal das Protokoll:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:66] BT Discovery:
INFO     [main.py:71] BT Device   Werkstatt   address=D2:24:68:EA:C4:D6
INFO     [main.py:71] BT Device   BootLissy   address=A4:C1:38:2D:D9:87
INFO     [main.py:71] BT Device   63-11-63-06-E9-0B   address=63:11:63:06:E9:0B
INFO     [main.py:71] BT Device   SunSolar   address=FB:52:2F:CA:45:90
INFO     [bt.py:93] BMS JKBt(C8:47:8C:F7:A8:B2) keep alive enabled
INFO     [main.py:150] connecting mqtt homeassistant@core-mosquitto
INFO     [main.py:171] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
INFO     [sampling.py:38] connecting bms JKBt(C8:47:8C:F7:A8:B2)
INFO     [jikong.py:114] normal connect failed, connecting with scanner
ERROR    [sampling.py:90] battery1 error: Device C8:47:8C:F7:A8:B2 not discovered ({'D2:24:68:EA:C4:D6', 'A4:C1:38:2D:D9:87', '63:11:63:06:E9:0B', 'FB:52:2F:CA:45:90'})
ERROR    [main.py:205] 1 exceptions occurred fetching BMSs
ERROR    [main.py:83] Error (num 1) reading BMS: Device C8:47:8C:F7:A8:B2 not discovered ({'D2:24:68:EA:C4:D6', 'A4:C1:38:2D:D9:87', '63:11:63:06:E9:0B', 'FB:52:2F:CA:45:90'})
ERROR    [main.py:84] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 22, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:F7:A8:B2 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 79, in fetch_loop
    await fn()
  File "/app/main.py", line 206, in fn
    raise exceptions[0]
  File "/app/main.py", line 201, in fn
    await t
  File "/app/bmslib/sampling.py", line 29, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 42, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 80, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:F7:A8:B2 not discovered ({'D2:24:68:EA:C4:D6', 'A4:C1:38:2D:D9:87', '63:11:63:06:E9:0B', 'FB:52:2F:CA:45:90'})
INFO     [sampling.py:38] connecting bms JKBt(C8:47:8C:F7:A8:B2)
INFO     [jikong.py:114] normal connect failed, connecting with scanner
ERROR    [sampling.py:90] battery1 error: Device C8:47:8C:F7:A8:B2 not discovered ({'A4:C1:38:2D:D9:87', 'D2:24:68:EA:C4:D6', 'FB:52:2F:CA:45:90'})
ERROR    [main.py:205] 1 exceptions occurred fetching BMSs
ERROR    [main.py:83] Error (num 2) reading BMS: Device C8:47:8C:F7:A8:B2 not discovered ({'A4:C1:38:2D:D9:87', 'D2:24:68:EA:C4:D6', 'FB:52:2F:CA:45:90'})
ERROR    [main.py:84] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 22, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:F7:A8:B2 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 79, in fetch_loop
    await fn()
  File "/app/main.py", line 206, in fn
    raise exceptions[0]
  File "/app/main.py", line 201, in fn
    await t
  File "/app/bmslib/sampling.py", line 29, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 42, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 80, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:F7:A8:B2 not discovered ({'A4:C1:38:2D:D9:87', 'D2:24:68:EA:C4:D6', 'FB:52:2F:CA:45:90'})
INFO     [sampling.py:38] connecting bms 

Gruß Thomas

rromrrom, 2022-10-22:
I have exactly same issue. But it happens not after start, but after certain time (varies 10-20 minutes). After that script has to be restarted to work properly.

trollingtom, 2022-10-23:
Also wenn ich Batmon neu starte passiert das gleiche.

fl4p, 2022-10-24:
Can you find the BMS using the app? 
Also try to reboot your Home Assistant host. Sometimes bluetooth connections get stalled and stay open even when you restart the program.
BMS restart can help too.

fl4p, 2023-03-26:
Closing due to inactivity

closed 2023-03-26 by fl4p