fl4p/batmon-ha #25: Docker container

yue31, 2022-10-22:
Installation requires HA addons. My HA system uses Docker container and it is impossible to use Addons installation. Could you provide instructions on installation with Docker system?

yue31, 2022-10-29:
I tested standalone and it did work. Cell voltages are showing up on screen now. I'll need to figure out how to integrate into docker home assistant. 
Great work and thank you very much!

yue31, 2022-11-06:
Just let you know everything works great with standalone setup and docker HA!

rromrrom, 2022-12-21:
@fl4p could you please explain how you build docker container? Especially what is base image (BUILD_FROM). I see it requires alpine, s6 and bashio which is pretty specific requirement (do you really need all that?) and I don't see any "official" image with all that onboard . Thanks in advance.

rromrrom, 2022-12-21:
@fl4p ah, sorry, now I got all details. I've been assuming before that this is generic docker image, but not homeassistant specific. All clear now, thanks.

robwolff3, 2023-01-23:
@rromrrom So a docker container cant be built from this repository? I am not using home assistant OS so I cannot use the traditional add-on method, I dont think.

robwolff3, 2023-01-23:
Im running docker on a amd64 archetecute computer. The following worked for me:

1. Change the first line in run.sh to `#!/bin/sh`
2. `docker build --build-arg BUILD_FROM=alpine -f Dockerfile -t batmon-ha .`

and then I ran it in docker compose with:

```
  batmon-ha:
    container_name: batmon-ha
    image: batmon-ha
    restart: unless-stopped
    privileged: true
    volumes:
      - ${CONFIGDIR}/batmon-ha/options.json:/app/options.json
      - /var/run/dbus/:/var/run/dbus/:z
```

fl4p, 2023-02-25:
Thanks, I'll reference this in the Standalone readme

closed 2023-02-25 by fl4p

Sayrin, 2023-12-14:
> I'm running docker on a amd64 architecture computer. The following worked for me:
> 
>     1. Change the first line in run.sh to `#!/bin/sh`
> 
>     2. `docker build --build-arg BUILD_FROM=alpine -f Dockerfile -t batmon-ha .`
> 
> 
> and then I ran it in docker compose with:
> 
> ```
>   batmon-ha:
>     container_name: batmon-ha
>     image: batmon-ha
>     restart: unless-stopped
>     privileged: true
>     volumes:
>       - ${CONFIGDIR}/batmon-ha/options.json:/app/options.json
>       - /var/run/dbus/:/var/run/dbus/:z
> ```

run.sh has bin renamed to addon_main.sh

Dukat-Gul, 2024-01-01:
Apparently the errors:
```
fetch http://dl-cdn.alpinelinux.org/alpine/edge/main/x86_64/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/edge/main: temporary error (try again later)
```
require this for raspberrypi 

```
wget http://ftp.de.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.5.4-2+b1_armhf.deb
dpkg -i libseccomp2_2.5.4-2+b1_armhf.deb
```

coconup, 2024-02-19:
I just went through the process as well and created a somewhat cleaner setup which avoids the extra `install_bleak.py` step currently pointing at an old version patched for windows (?). It also sets environment variables through `docker-compose` instead of `addon_main.sh`. Finally, it avoids running the container in privileged mode.

I'm running this on a Raspberry Pi (i.e. Linux Debian), but I don't think it matters much.

`./services/batmon/Dockerfile:`
```
FROM alpine

WORKDIR /app

RUN apk update \
  && apk add --no-cache \
    python3 \
    bluez \
    py-pip \
    git

RUN rm /usr/lib/python3.11/EXTERNALLY-MANAGED

COPY ./requirements.txt requirements.txt
RUN pip3 install -r requirements.txt
```

`./services/batmon/requirements.txt`
```
bleak==0.20.2
paho-mqtt>=2.0.0
backoff
crcmod
```

`./docker-compose.yml`
```
  batmon:
    container_name: batmon
    build:
      context: ./services/batmon
    restart: unless-stopped
    environment:
      MQTT_HOST: "mosquitto"
      MQTT_USER: ""
      MQTT_PASSWORD: ""
    volumes:
      - /run/dbus:/run/dbus:ro
      - ./volumes/batmon-ha/:/app # Clone the repo in ./volumes/batmon-ha
    command: ["/bin/sh", "-c", "python3 ./main.py"]
```

ThomasCr, 2024-07-18:
It would be really nice, if there would be a official maintained container on docker-hub or so.
I also dot use HASS OS and also cant use HA addons the normal way.
Currently I have no BMS to monitor, but I am planning to buy and found this very interesting integration for HA.