fl4p/batmon-ha #33: Does not work with my version of Daly Smart BMS Bluetooth

m0jek, 2022-11-09:
Hi,

I have four Sterling Power LiFePO batteries that have built in Daly Smart BMS BT, however they use a different message header to what you are using. There is more information here; https://github.com/roccotsi2/esp32-smart-bms-simulation/blob/main/README.md

I'm in the process of trying to get to test the stand alone code, but use not working too well, and I am not a Python programmer so is step learning curve for me/

extract;

**Data format of used Smart BMS from Daly:**

This section does not contain the full data format for Daly Smart BMS (there seems to be different Smart BMS that uses another data format, anyway), because I found no official documentation on this and therefore I analyzed the data packets. The results of my analysis are documented in this section. It seems to be the case that Smart BMS is waiting for some data (because Android App does send commands to my simulation while connected to it), but I do not know how to interpret these data yet. This section will be updated if I get new insights on the data format.

**Daly Smart BMS expects following commands (data from client to Smart BMS):**

Request RunInfo (tab status in Smart BMS app): D2 03 00 00 00 3E D7 B9
Request some unknown values (tab status in Smart BMS app): D2 03 00 A9 00 20 87 91
Request some unknown values (tab parameter in Smart BMS app): D2 03 00 80 00 29 96 5F
Daly Smart BMS uses the following format to send data (data from Smart BMS to a client):

<HEADER_READ><CONTENT_LENGTH><CONTENT><CRC>
<-----------CRC relevant------------->
HEADER_READ: always 0xD203
CONTENT_LENGTH: content length in bytes
CONTENT: the content (depends on the information which is sent)
CRC: checksum (2 bytes). The checksum is calculated over HEADER_READ, CONTENT_LENGTH and CONTENT. See function "getCRC" to see how CRC is calculated
The content lengths are in bytes:

Run Info: 124
Run Info Last Battery Value: 32
Set Info: 82
Version Info: 64
PWD: 6

**Content format Run Info**

All positions needs 2 bytes:

Position 0-31: Battery volt (in mV)
Position 32-39: Battery temperature in Celsius (offset: 40)
Position 40: current V (in 0,1V)
Position 41: current A (offset: 30000 in 0,1A)
Position 42: SOC - State Of Charge in Percent (in 0,1 percent)
Position 43: max cell voltage (in mV)
Position 44: min cell voltage (in mV)
Position 45: ???
Position 46: ???
Position 47: ???
Position 48: ???
Position 49: count battery volt
Position 50: count battery temperatures (max. 8)
Position 51: cycle
Position 52: JH on/off
Position 53: CDMOS on/off (1 = on, everything else off)
Position 54: FDMOS on/off
Position 55: average voltage (in mV)
Position 56: differential voltage (in mV)
Position 57: current KW (in W)
Position 58: Alarm1
Position 59: Alarm2
Position 60: Alarm3
Position 61: Alarm4

**Content format Version Info**

<App Version: 16 bytes as String> (reversed), <MCU Version: 16 bytes as String>, <Machine Version: 32 bytes as String>

**Content format Run Info Last Battery Value**

16 values (each 2 byte): Battery volt (in mV)

tomatensaus, 2022-11-21:
I also could not get this code to work on my Daly. I cloned another project and with some effort managed to get it talking. Maybe this can help in solving the issues

fl4p, 2022-11-21:
I am working on this.

@tomatensaus What project did you use?

tomatensaus, 2022-11-21:
I have published my changes here. https://github.com/tomatensaus/python-daly-bms

psgcooldog, 2023-12-23:
I'm having what appears to be a similar problem.  This is from a 500A Daly Smart BMS that's about 16 months old.

I've attached logs.  Let me know if there's anything else that would be helpful.

[_addon_2af0a32d_batmon_logs (1).txt](https://github.com/fl4p/batmon-ha/files/13760046/_addon_2af0a32d_batmon_logs.1.txt)

closed 2025-10-02 by fl4p