fl4p/batmon-ha #34: esphome bluetooth proxy?

the-butterfry, 2022-11-10:
Been trying to get a remote daly to show up to a working HA batmon. I can see and pull data just fine when the HA computer is close, but when they're separated and have an esphome proxy attempting to forward connections - no device found. Am I being stupid here? or should this setup work?

I can see the device in the proxy logs, just fine

tangmonster911, 2022-11-17:
My understanding is that batmon is not working with bluetooth proxy. I am checking in on this project twice weekly in anticipation for that to work. Untill then I cannot use it.

on the main page:

TODO
use the new Bluetooth integration since HA 2022.8 https://www.home-assistant.io/integrations/bluetooth/

fl4p, 2022-11-21:
batmon uses standard BLE connection protocol, so it might work. Your bluetooth proxy must be discoverable with a Bluetooth Low-Energy scan. No compatibility to esphome though, as it uses WiFi for communication.

fl4p, 2023-02-01:
Actually, this is would require a custom integration (through HACS), as batmon just communicates through mqtt with HA and establishes bluetooth connections through DBus with the bluez stack.

MityaDVG, 2023-02-03:
What custom integration (through HACS)?

fl4p, 2023-02-25:
This needs to be implemented, porting batmon from add-on to an Integration. I have no experience with that though

fl4p, 2023-04-03:
Planned to port batmon to MicroPython, which supports ESP32

goldriver2, 2024-04-09:
I see that all discussion around this request to add ESPHome Proxy seems to be closed (  #179, #187 and 34) any one knows if this resquest is still an Open one or is this been rejected ?

Mikrofarad, 2024-08-21:
For me it was easier to use an ESP32 and follow this [Beginner's guide](https://github.com/syssi/esphome-jbd-bms/blob/main/docs/beginners-guide.md). Runs perfect with Bluetooth Proxies 👍