fl4p/batmon-ha #40: Keeping status of sensors not provided after unavailable BMS

Razsmithii, 2022-12-01:
First: really great project an relatively eas to install. Worked for me in 10 min! Thanks....
As the BMS goes in to sleep mode, all values get lost to "unknown".
In the changelog i can see, since  v 0.45, the behavior should be different.
What´s the problem here?

fl4p, 2022-12-01:
Values expire if the BMS is not reachable for 30 seconds.

What BMS do you use and how do you put it into sleep mode?

Razsmithii, 2022-12-01:
Ist there any option to change the expire-time?
Its a daly. It goes to sleep mode after a specific time (which can be adjusted), when current = 0.

fl4p, 2022-12-06:
In [bmslib/bms.py](https://github.com/fl4p/batmon-ha/blob/master/bmslib/bms.py#L5) you can change 
```
MIN_VALUE_EXPIRY = 20
```

fl4p, 2022-12-06:
I might add an option so you can configure it.

fl4p, 2023-02-01:
This will be in the next release. https://github.com/fl4p/batmon-ha/commit/f3e8169536e2602087dc2ba396143951fc2e4a57

closed 2023-02-01 by fl4p