fl4p/batmon-ha #41: Ancel Battery Monitor

chappyboy, 2022-12-05:
Hi there, I got Ancel Battery Monitor in my car. Will this be supported? https://www.anceltech.com/product/detail?id=1243380677060063232

fl4p, 2022-12-06:
Hi, batmon doesn't work with this battery. If there are a few people with interest I might implement it (given that someone can provide captured bluetooth frames).

chappyboy, 2022-12-07:
This is a popular Battery Monitoring Device in Amazon. I can provide you with Bluetooth frames if you can guide me on how to get it.

chappyboy, 2022-12-07:
Didn't mean to close it.

fl4p, 2026-09-03:
The Ancel BM200 is a rebranded Quicklynks BM2, so I added `type: bm2` (1aeccaf, next release). It decrypts the frames the device pushes on FFF4 with the BM2 key found by KrystianD and haxrob and publishes voltage and SOC; the BM2 frame carries no temperature. I have no device, and it is possible Ancel changed the key, so please try it and post a verbose_log: if every notification logs as "undecodable", the key differs and I would need a capture from the Ancel app.

fl4p, 2026-09-03:
2.20 is out with `type: bm2`. Untested on hardware, please try it and post the log if it does not read.

BigDurl, 2026-11-07:
I'm unsure if this is where to say this but. I have an Ancel BM200 and `type: bm2` doesn't work with it however. `type: bm6` does report the voltage, soc and temperature just fine.

fl4p, 2026-11-10:
Thanks, that's useful. Your BM200 speaks the BM6 protocol, not BM2, so `type: bm6` is the right one, and it gives you temperature too, which `bm2` can't.

I fixed the docs, and from the next release `bm2` suggests `bm6` when it hears nothing, and `type: auto` tells the two apart by itself.

closed 2026-11-10 by fl4p