fl4p/batmon-ha #46: High CPU usage, no readings

igg, 2023-01-03:
Apparently all of the connections are OK, but there is no battery info.

JK BMS: JK_BD4A8S4P
HomeAssistant on Odriod-C2 with Edimax Bluetooth USB dongle
* Home Assistant 2022.12.8
* Supervisor 2022.12.1
* Operating System 9.4
* Frontend 20221213.1 - latest

### batmon log:
INFO     [main.py:178] connecting mqtt batmon@core-mosquitto
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=False
INFO     [sampling.py:41] connecting bms JKBt(20:22:08:11:0A:35)
DEBUG    [client.py:181] Connecting to device @ 20:22:08:11:0A:35 with hci0
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
...the last two lines repeat indefinitely

### Supervisor log:
23-01-02 21:16:57 INFO (SyncWorker_2) [supervisor.docker.addon] Starting Docker add-on 2af0a32d/aarch64-addon-batmon with version 0.0.46
23-01-02 21:17:04 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'batmon'
23-01-02 21:17:04 INFO (MainThread) [supervisor.auth] Successful login for 'batmon'

### Devices/batmon:
#### Sensors (updating live)
* CPU percent 23.98%
* Memory percent 1.11%
* Newest version 0.0.46
* Running Running
* Version 0.0.46

And that's it - no battery info.
Did the MQTT not auto-configure? I don't see autoconfigure messages while listening to # in MQTT
Is 25% CPU usage normal?

sbox99, 2023-01-03:
I have the exact same issue. Would be great to get a solution.

INFO     [main.py:178] connecting mqtt @10.0.1.14
DEBUG    [client.py:2529] Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=False
INFO     [sampling.py:41] connecting bms JKBt(C8:57:8C:E8:54:92)
DEBUG    [client.py:181] Connecting to device @ C8:57:8C:E8:54:92 with hci0
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP

fl4p, 2023-01-10:
I just tested with Home Assistant 2023.1.2 and mosquitto add-on v6.1.3 and it works.

I just pushed a patch to the master branch  here that might fixes the issue. (it patches the _handle_pingresp of paho.Client to not reset `_ping_t` to 0)

igg, 2023-01-11:
I'm out of town for a couple of weeks, but will try when I get back. Were you able to reproduce this issue with your setup?

fl4p, 2023-01-12:
No, I could not reproduce the issue. What MQTT broker version do you use?

I can see a potential issue in Paho MQTT client implementation, where the ping timer (_ping_t ) is used for scheduling periodic pings to the server and also to detect server timeouts. This might lead to a spinning loop, because once the client receives a ping response from the server (_ping_t assigned 0) , the client will immediately send another ping request, to which the server will respond again.

This is just my guess from quickly looking at the MQTT client code.

klarjr, 2023-01-17:
Thank you for hard work on this. I believe I have the same issue as well. I'm unfamiliar with github etiquette and I don't want to hijack so if I'm overstepping, or should be doing something different, please let me know. I've tried both the addon and standalone and I'm getting the same result as mentioned above with the addon. I too am using Mosquitto broker addon v6.1.3. With the standalone install on a separate raspberry pi, when I ctrl-c/end the program it produces a rather long log and does send one update successfully over mqtt with valid information which populates the sensors only if I end the program early. If I wait for several attempt cycles, it gets in a never ending loop of trying to disconnect from an already disconnected Bluetooth adapter which I believe is a separate bug. Below a debug log for the stand alone install and Mosquitto from the same run. Let me know if I can test or share any other information that might help find our issue. Thank you again for your dedicated work.  It truly is appreciated. 

Stand alone install debug log
`pi@raspberrypi:~ $ cd batmon-ha
pi@raspberrypi:~/batmon-ha $ sudo python3 main.py
error reading /data/options.json, trying options.json [Errno 2] No such file or directory: '/data/options.json'
INFO     [main.py:69] BT Discovery:
INFO     [main.py:74] BT Device   JK-B2A24S20P   address=C8:47:8C:E2:DA:FB
INFO     [main.py:74] BT Device   HTZM   address=FF:FF:3A:00:D4:6C
INFO     [main.py:74] BT Device   50-95-5B-78-1D-33   address=50:95:5B:78:1D:33
INFO     [main.py:74] BT Device   Raycon E55 Earbuds   address=11:22:33:44:9E:6D
INFO     [main.py:137] Verbose logging enabled
INFO     [main.py:179] connecting mqtt mqtt_broker@192.168.0.200
DEBUG    [mqtt_util.py:325] applied paho monkey patch _handle_pingresp
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:204] Fetching 1 BMS + 0 others serially, period=15.00s, keep_alive=False
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:E2:DA:FB)
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:181] Connecting to device @ C8:47:8C:E2:DA:FB with hci0
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP (patched)
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP (patched)
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
DEBUG    [client.py:2529] Received CONNACK (0, 0)
^CINFO     [main.py:266] exit signal handler...
INFO     [jikong.py:114] normal connect failed, connecting with scanner
DEBUG    [bt.py:39] starting scan
DEBUG    [scanner.py:132] cached devices: {}
DEBUG    [scanner.py:282] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [client.py:507] Disconnecting (None)
DEBUG    [client.py:513] already disconnected (None)
`

Mosquitto broker debug log
`[20:13:53] INFO: Successfully send discovery information to Home Assistant.
[20:13:54] INFO: Successfully send service information to the Supervisor.
time="2023-01-16T20:13:55-06:00" level=debug msg="checking acl cache for mqtt_broker"
time="2023-01-16T20:13:55-06:00" level=debug msg="to auth record: [97 99 108 45 109 113 116 116 95 98 114 111 107 101 114 45 104 111 109 101 97 115 115 105 115 116 97 110 116 47 115 116 97 116 117 115 45 52 103 83 52 116 110 82 105 106 69 77 74 105 120 74 120 117 108 83 122 75 101 45 50 218 57 163 238 94 107 75 13 50 85 191 239 149 96 24 144 175 216 7 9]\n"
time="2023-01-16T20:13:55-06:00" level=debug msg="Superuser check with backend Files"
time="2023-01-16T20:13:55-06:00" level=debug msg="Superuser check with backend HTTP"
time="2023-01-16T20:13:55-06:00" level=debug msg="http request approved for mqtt_broker"
time="2023-01-16T20:13:55-06:00" level=debug msg="superuser mqtt_broker acl authenticated with backend HTTP"
time="2023-01-16T20:13:55-06:00" level=debug msg="setting acl cache (granted = true) for mqtt_broker"
time="2023-01-16T20:13:55-06:00" level=debug msg="to auth record: [97 99 108 45 109 113 116 116 95 98 114 111 107 101 114 45 104 111 109 101 97 115 115 105 115 116 97 110 116 47 115 116 97 116 117 115 45 52 103 83 52 116 110 82 105 106 69 77 74 105 120 74 120 117 108 83 122 75 101 45 50 218 57 163 238 94 107 75 13 50 85 191 239 149 96 24 144 175 216 7 9]\n"
time="2023-01-16T20:13:55-06:00" level=debug msg="Acl is true for user mqtt_broker"
time="2023-01-16T20:14:33-06:00" level=debug msg="checking auth cache for mqtt_broker"
time="2023-01-16T20:14:33-06:00" level=debug msg="to auth record: [97 117 116 104 45 109 113 116 116 95 98 114 111 107 101 114 45 104 111 109 101 97 115 115 105 115 116 97 110 116 218 57 163 238 94 107 75 13 50 85 191 239 149 96 24 144 175 216 7 9]\n"
time="2023-01-16T20:14:33-06:00" level=debug msg="found in cache: mqtt_broker"
time="2023-01-16T20:16:35-06:00" level=debug msg="checking auth cache for mqtt_broker"
time="2023-01-16T20:16:35-06:00" level=debug msg="to auth record: [97 117 116 104 45 109 113 116 116 95 98 114 111 107 101 114 45 104 111 109 101 97 115 115 105 115 116 97 110 116 218 57 163 238 94 107 75 13 50 85 191 239 149 96 24 144 175 216 7 9]\n"
[... cut ...]
time="2023-01-16T20:32:32-06:00" level=debug msg="to auth record: [97 99 108 45 109 113 116 116 95 98 114 111 107 101 114 45 104 111 109 101 97 115 115 105 115 116 97 110 116 47 98 105 110 97 114 121 95 115 101 110 115 111 114 47 98 97 116 116 101 114 121 50 47 100 105 115 99 104 97 114 103 101 47 99 111 110 102 105 103 45 52 103 83 52 116 110 82 105 106 69 77 74 105 120 74 120 117 108 83 122 75 101 45 49 218 57 163 238 94 107 75 13 50 85 191 239 149 96 24 144 175 216 7 9]\n"
time="2023-01-16T20:32:32-06:00" level=debug msg="Acl is true for user mqtt_broker"`

igg, 2023-02-08:
> Hi, please try just released 0.0.50, it should fix this issue

This release fixed the CPU utilization issue, but there are no measurements of anything other than the version number.

fl4p, 2023-04-03:
Please open a new issue if you still experience problems

closed 2023-04-03 by fl4p