fl4p/batmon-ha #47: bms jk not work, does not connect

amgstone, 2023-01-10:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   [TV] Samsung Q60 Series (55)   address=B8:BC:5B:A9:98:AB
INFO     [bt.py:91] BMS JKBt(C8:47:8C:E8:FF:73) keep alive enabled
INFO     [main.py:178] connecting mqtt roman@core-mosquitto
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=10.00s, keep_alive=True
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:E8:FF:73)
INFO     [main.py:99] mqtt watchdog loop started with timeout 120.0s
INFO     [jikong.py:114] normal connect failed, connecting with scanner
ERROR    [sampling.py:97] JK-BD6A20S10P error: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:240] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 1) reading BMS: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 22, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E8:FF:73 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 241, in fn
    raise exceptions[0]
  File "/app/main.py", line 236, in fn
    await t
  File "/app/bmslib/sampling.py", line 32, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 45, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 78, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 35, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})

amgstone, 2023-01-10:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   ANT-BLE16ZMUB   address=16:AA:22:02:23:45
INFO     [main.py:136] Verbose logging enabled
INFO     [bt.py:91] BMS JKBt(C8:47:8C:E8:FF:73) keep alive enabled
INFO     [main.py:178] connecting mqtt roman@core-mosquitto
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=10.00s, keep_alive=True
DEBUG    [client.py:2529] Received CONNACK (0, 0)
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:E8:FF:73)
DEBUG    [client.py:181] Connecting to device @ C8:47:8C:E8:FF:73 with hci0
INFO     [main.py:99] mqtt watchdog loop started with timeout 120.0s
DEBUG    [scanner.py:132] cached devices: {'/org/bluez/hci0/dev_6F_D5_7C_8D_F9_5D': {'Address': '6F:D5:7C:8D:F9:5D', 'AddressType': 'random', 'Alias': '6F-D5-7C-8D-F9-5D', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -78, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05q\x1c\xf3\x9b2'}, 'TxPower': 8, 'ServicesResolved': False}, '/org/bluez/hci0/dev_E0_51_AC_C2_03_D3': {'Address': 'E0:51:AC:C2:03:D3', 'AddressType': 'random', 'Alias': 'E0-51-AC-C2-03-D3', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -76, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x03'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F3_4D_B3_67_B6_66': {'Address': 'F3:4D:B3:67:B6:66', 'AddressType': 'random', 'Alias': 'F3-4D-B3-67-B6-66', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -75, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x01'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_69_C0_97_83_A7_47': {'Address': '69:C0:97:83:A7:47', 'AddressType': 'random', 'Alias': '69-C0-97-83-A7-47', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -79, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05q\x1c\xc0?R'}, 'TxPower': 8, 'ServicesResolved': False}, '/org/bluez/hci0/dev_ED_A2_83_BB_FB_26': {'Address': 'ED:A2:83:BB:FB:26', 'AddressType': 'random', 'Alias': 'ED-A2-83-BB-FB-26', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -78, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x03'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_4B_F5_4D_62_69_B5': {'Address': '4B:F5:4D:62:69:B5', 'AddressType': 'random', 'Alias': '4B-F5-4D-62-69-B5', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -80, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05q\x1c^\xc0N'}, 'TxPower': 8, 'ServicesResolved': False}, '/org/bluez/hci0/dev_D6_8B_D7_81_71_6E': {'Address': 'D6:8B:D7:81:71:6E', 'AddressType': 'random', 'Alias': 'D6-8B-D7-81-71-6E', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -79, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x01'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_59_C8_57_0B_60_E8': {'Address': '59:C8:57:0B:60:E8', 'AddressType': 'random', 'Alias': '59-C8-57-0B-60-E8', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -74, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05q\x1c\x91\x1f\x8e'}, 'TxPower': 8, 'ServicesResolved': False}, '/org/bluez/hci0/dev_D5_ED_0B_78_47_E5': {'Address': 'D5:ED:0B:78:47:E5', 'AddressType': 'random', 'Alias': 'D5-ED-0B-78-47-E5', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -76, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x02'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_5E_3D_D9_A7_9A_97': {'Address': '5E:3D:D9:A7:9A:97', 'AddressType': 'random', 'Alias': '5E-3D-D9-A7-9A-97', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -74, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05r\x1c{=d'}, 'TxPower': 7, 'ServicesResolved': False}, '/org/bluez/hci0/dev_E9_96_96_1F_5D_84': {'Address': 'E9:96:96:1F:5D:84', 'AddressType': 'random', 'Alias': 'E9-96-96-1F-5D-84', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -73, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x12\x02\x00\x02'}, 'ServicesResolved': False}, '/org/bluez/hci0/dev_F5_8D_C1_A6_E4_EA': {'Address': 'F5:8D:C1:A6:E4:EA', 'AddressType': 'random', 'Alias': 'F5-8D-C1-A6-E4-EA', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -77, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'Manufacturer [... cut ...] hci0', 'ServicesResolved': False}, '/org/bluez/hci0/dev_16_AA_22_02_23_45': {'Address': '16:AA:22:02:23:45', 'AddressType': 'public', 'Name': 'ANT-BLE16ZMUB', 'Alias': 'ANT-BLE16ZMUB', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -75, 'Connected': False, 'UUIDs': ['0000ffe0-0000-1000-8000-00805f9b34fb', '0000fee7-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {1623: b'\x88\xa0\x16\xaa"\x01#E'}, 'ServicesResolved': False}}`

fl4p, 2023-01-10:
Hi,
looks like your HA host cannot find the JK during a bluetooth scan. Try to connect to the BMS using the JK app, the close the app and re-start the batmon.

amgstone, 2023-01-10:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
ERROR    [main.py:126] Error discovering devices: [org.bluez.Error.NotReady] Resource Not Ready
INFO     [main.py:136] Verbose logging enabled
INFO     [bt.py:91] BMS JKBt(C8:47:8C:E8:FF:73) keep alive enabled
INFO     [main.py:178] connecting mqtt roman@core-mosquitto
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=10.00s, keep_alive=True
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:E8:FF:73)
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:181] Connecting to device @ C8:47:8C:E8:FF:73 with hci0
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP`

I did so, but neither the entity nor the device was added

amgstone, 2023-01-10:
when I disable the watchdog, it connects to the device
`INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   B8-BC-5B-A9-98-AB   address=B8:BC:5B:A9:98:AB
INFO     [main.py:136] Verbose logging enabled
INFO     [bt.py:91] BMS JKBt(C8:47:8C:E8:FF:73) keep alive enabled
INFO     [main.py:178] connecting mqtt roman@core-mosquitto
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=10.00s, keep_alive=True
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:E8:FF:73)
DEBUG    [client.py:2529] Received CONNACK (0, 0)
DEBUG    [client.py:181] Connecting to device @ C8:47:8C:E8:FF:73 with hci0
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP
DEBUG    [client.py:2529] Sending PINGREQ
DEBUG    [client.py:2529] Received PINGRESP`

amgstone, 2023-01-28:
is there any news I still can't connect to jk bms

fl4p, 2023-02-25:
Please try the new update https://github.com/fl4p/batmon-ha/commit/cc8ed5d1f9800da84cf00021779fac9e9bf8a79e

amgstone, 2023-02-25:
there is no update in hacs

fl4p, 2023-02-25:
Is just pushed it to the repository

amgstone, 2023-02-25:
so what to do next? I don't know where to put that code.

fl4p, 2023-02-25:
The update should appear in your add-on store, click on "check for updates"

amgstone, 2023-02-25:
File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 65, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 120, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 54, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E8:FF:73 not discovered (set())
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:E8:FF:73)
INFO     [jikong.py:114] normal connect failed (Device with address C8:47:8C:E8:FF:73 was not found.), connecting with scanner
ERROR    [sampling.py:125] Irenahouse_48 error: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 8) reading BMS: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 34, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E8:FF:73 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 65, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 120, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 54, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:E8:FF:73)
ERROR    [main.py:120] MQTT never published a message after 120s, exit
INFO     [jikong.py:114] normal connect failed (Device with address C8:47:8C:E8:FF:73 was not found.), connecting with scanner
ERROR    [sampling.py:125] Irenahouse_48 error: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 9) reading BMS: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 112, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 34, in connect
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:E8:FF:73 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 65, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 120, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 115, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 54, in _connect_with_scanner
    raise Exception('Device %s not discovered (%s)' % (self.client.address, discovered))
Exception: Device C8:47:8C:E8:FF:73 not discovered ({'B8:BC:5B:A9:98:AB'})
INFO     [main.py:282] Shutting down ...
INFO     [main.py:289] Disconnecting JKBt(C8:47:8C:E8:FF:73)
INFO     [main.py:298] exit signal handler...
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

fl4p, 2023-02-25:
Can you run a BLE scan from the command line and see the BMS?
https://ukbaz.github.io/howto/beacon_scan_cmd_line.html

amgstone, 2023-02-25:
http://joxi.ru/Vm63G0Oi4exqe2

amgstone, 2023-02-25:
http://joxi.ru/5mdny8OS3EvnbA

fl4p, 2023-02-25:
Sometimes it can take a minute until the device appears. If it doesn't I can't help you here. This is not related to batmon

fl4p, 2023-07-28:
please re-open if neede

closed 2023-07-28 by fl4p