fl4p/batmon-ha #48: Standalone JBD hangs on connect no response and no timeout

flambert860, 2023-01-20:
For some reason it seems like that the:
`await self.client.connect(timeout=timeout)`
hangs the process, if I ctrl-c it says connected after some time, and also it does not timeout after 20 sec or longer.

It's very weird because I had my own python code which talks with the bms also using Bleak, but found your implemtation more done and looked really mature.

Anyone who might have a clue how to prevent the connect to hang?

flambert860, 2023-01-20:
Seems to be a problem with the threading, when connecting in the main before starting the other threads, but then it get stucks when trying to write the GATTs

flambert860, 2023-01-20:
More findings, as suspected the 
`asyncio.create_task(background_loop(timeout=max(120., sample_period * 3) if watchdog_en else 0))`
Causes my issues, so when commenting it out it works, anyhow I can fix this?

fl4p, 2023-08-28:
this should've been fixed some time ago. please re-open if not

closed 2023-08-28 by fl4p