fl4p/batmon-ha #49: reading BMS: float division by zero

zingreen, 2023-01-22:
Hi, 
I added the addon to HA and have weird issue in log.

My configuration is: 

```
devices:
  - address: C8:47:8C:EA:BF:0D
    type: jk
    alias: battery1
mqtt_user: user
mqtt_password: passwd
mqtt_broker: core-mosquitto
concurrent_sampling: false
keep_alive: false
verbose_log: false
sample_period: 5
invert_current: false
watchdog: true
pin: 1234
```

and error in logs:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   JK_B2A8S20P   address=C8:47:8C:EA:BF:0D
INFO     [main.py:178] connecting mqtt user@core-mosquitto
INFO     [main.py:201] Fetching 1 BMS + 0 others serially, period=5.00s, keep_alive=False
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:EA:BF:0D)
INFO     [main.py:99] mqtt watchdog loop started with timeout 120.0s
INFO     [sampling.py:47] connected bms JKBt(C8:47:8C:EA:BF:0D)!
ERROR    [sampling.py:97] battery1 error: float division by zero
ERROR    [main.py:240] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 1) reading BMS: float division by zero
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 241, in fn
    raise exceptions[0]
  File "/app/main.py", line 236, in fn
    await t
  File "/app/bmslib/sampling.py", line 32, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 49, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 176, in fetch
    return BmsSample(
  File "/app/bmslib/bms.py", line 52, in __init__
    soc = round(charge / capacity * 100, 2)
ZeroDivisionError: float division by zero
```

Could you help me with it?
Thank you very much!

zingreen, 2023-01-22:
Additional information:

```
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:EA:BF:0D)
ERROR    [jikong.py:65] crc check failed, 147 != 113, bytearray(b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00DU\xaa\xeb\x90\x01?X\x02\x00\x00(\n\x00\x00Z\n\x00\x00\x10\x0e\x00\x00\xde\r\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\xa8a\x00\x00\x1e\x00\x00\x00<\x00\x00\x00\xe0"\x02\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xd0\x07\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\xbc\x02\x00\x00X\x02\x00\x008\xff\xff\xff\x9c\xff\xff\xff\xe8\x03\x00\x00 \x03\x00\x00\x04\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\xc0E\x04\x00\xdc\x05\x00\x00\xe4\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x00\x00\x00\x00\x00|\xf8\xff\xff\x9f\r\x00\x00\x00\x00\x00\x00\xa5')
WARNING  [jikong.py:87] crc check failed, header at 20, discarding start of bytearray(b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00DU\xaa\xeb\x90\x01?X\x02\x00\x00(\n\x00\x00Z\n\x00\x00\x10\x0e\x00\x00\xde\r\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\xa8a\x00\x00\x1e\x00\x00\x00<\x00\x00\x00\xe0"\x02\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xd0\x07\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\xbc\x02\x00\x00X\x02\x00\x008\xff\xff\xff\x9c\xff\xff\xff\xe8\x03\x00\x00 \x03\x00\x00\x04\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\xc0E\x04\x00\xdc\x05\x00\x00\xe4\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x00\x00\x00\x00\x00|\xf8\xff\xff\x9f\r\x00\x00\x00\x00\x00\x00\xa5')
INFO     [sampling.py:47] connected bms JKBt(C8:47:8C:EA:BF:0D)!
ERROR    [sampling.py:97] battery1 error: float division by zero
ERROR    [main.py:240] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 10) reading BMS: float division by zero
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 241, in fn
    raise exceptions[0]
  File "/app/main.py", line 236, in fn
    await t
  File "/app/bmslib/sampling.py", line 32, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 49, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 176, in fetch
    return BmsSample(
  File "/app/bmslib/bms.py", line 52, in __init__
    soc = round(charge / capacity * 100, 2)
ZeroDivisionError: float division by zero
INFO     [sampling.py:41] connecting bms JKBt(C8:47:8C:EA:BF:0D)
ERROR    [jikong.py:65] crc check failed, 153 != 113, 
```

fl4p, 2023-02-03:
Hi, please try 0.0.50, it fixes this issue. Let me know whether it is working

konstantinzinchenko, 2023-02-03:
Thank you for the fix,
I have another one(

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:

INFO     [main.py:73] BT Device   JK_B2A8S20P   address=C8:47:8C:EA:BF:0D

INFO     [bt.py:126] BMS JKBt(C8:47:8C:EA:BF:0D) keep alive enabled
INFO     [main.py:195] connecting mqtt user@core-mosquitto
WARNING  [main.py:214] Failed to load meter states: [Errno 2] No such file or directory: '/data/bms_meter_states.json'
INFO     [main.py:232] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:EA:BF:0D)
INFO     [main.py:106] mqtt watchdog loop started with timeout 120.0s
ERROR    [sampling.py:125] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 1) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 65, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 113, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 120, in connect
    await self._q(cmd=0x96, resp=(0x02, 0x01))  # device state (resp 0x01 & 0x02)
  File "/app/bmslib/jikong.py", line 138, in _q
    return await self._fetch_futures.wait_for(resp, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 41, in wait_for
    return await asyncio.gather(*tasks, return_exceptions=False)
  File "/app/bmslib/__init__.py", line 44, in wait_for
    return await asyncio.wait_for(self._futures[name], timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

zingreen, 2023-02-05:
I was able to fix this previous error by copying the files
/usr/share/hassio/addons/data/2af0a32_batmon/bms_meter_states.json 
/usr/share/hassio/addons/data/2af0a32_batmon/options.json 
to 
/usr/share/hassio/addons/data/bms_meter_states.json 
/usr/share/hassio/addons/data/options.json 

but after that I have the error:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
...
INFO     [main.py:73] BT Device   JK_B2A8S20P   address=C8:47:8C:EA:BF:0D
...
INFO     [bt.py:126] BMS JKBt(C8:47:8C:EA:BF:0D) keep alive enabled
INFO     [main.py:195] connecting mqtt user@core-mosquitto
INFO     [main.py:232] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:EA:BF:0D)
INFO     [main.py:106] mqtt watchdog loop started with timeout 120.0s
INFO     [sampling.py:67] connected bms JKBt(C8:47:8C:EA:BF:0D)!
ERROR    [sampling.py:125] battery1 error: <class 'AssertionError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 1) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 176, in fetch
    return BmsSample(
  File "/app/bmslib/bms.py", line 57, in __init__
    assert math.isfinite(soc)
AssertionError
ERROR    [sampling.py:125] battery1 error: <class 'AssertionError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 2) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 176, in fetch
    return BmsSample(
  File "/app/bmslib/bms.py", line 57, in __init__
    assert math.isfinite(soc)
AssertionError

Thank you!

fl4p, 2023-02-25:
I fixed this and pushed it to the master branch

zingreen, 2023-02-25:
it works now. thank you!

closed 2023-02-25 by zingreen