fl4p/batmon-ha #50: Connection problem

dotevo, 2023-02-03:
Hello,

I'm using DalyBMS and it works with this addon, but after some time (sometimes a few hours, sometimes few days) it has an error. Only workaround is to restart addon from HA. After that it is connecting to bms again and it works for a some time.

Is it possible to reconnect to bms if this error is detected? I have 2 BMSes connected and I can see that sometimes only one works, usually if this error happens, I cannot see a new data on entity:

```
ERROR    [sampling.py:125] daly1 error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:85] Error (num 9) reading BMS: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 124, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 150, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 107, in _q
    await self.client.write_gatt_char(self.UUID_TX, msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 945, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [sampling.py:125] DalyBMS24 error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:85] Error (num 10) reading BMS: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 124, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 150, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 107, in _q
    await self.client.write_gatt_char(self.UUID_TX, msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 945, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [sampling.py:125] daly1 error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:85] Error (num 10) reading BMS: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 124, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 150, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 107, in _q
    await self.client.write_gatt_char(self.UUID_TX, msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 945, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [sampling.py:125] DalyBMS24 error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:85] Error (num 11) reading BMS: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 124, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 150, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 107, in _q
    await self.client.write_gatt_char(self.UUID_TX, msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 945, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
```

bruceborrett, 2023-02-13:
I am having the exact same issue, works for a couple hours then the above error starts to appear. After googling around a bit it seems it may have something to do with the bluez version (and ultimately, the version of homeassistant). I am going to try and debug further and will post my results.

fl4p, 2023-09-02:
I have similar issue running on native arch linux (armv7a, RPI3). Can't power cycle BT hardware, even with `rfkill`. After power down  and up `bluetoothctl` fails with NotReady error. The only solution I found was to reboot.

fl4p, 2023-10-20:
Enabling option `install_newer_bleak` probably fixes this.

closed 2023-10-20 by fl4p