fl4p/batmon-ha #51: JK-B2A24S20P connection probs

Catchmydrift, 2023-02-05:
hi, im completely new to HA, github etc.

i have 2x JK-B2A24S20P bms's and i get this when trying to connect:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   61-24-07-A8-2D-6A   address=61:24:07:A8:2D:6A
INFO     [bt.py:126] BMS JKBt(C8:47:8C:E5:AB:CB) keep alive enabled
INFO     [main.py:195] connecting mqtt @core-mosquitto
INFO     [main.py:232] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:E5:AB:CB)
INFO     [main.py:106] mqtt watchdog loop started with timeout 120.0s
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
ERROR    [main.py:120] MQTT never published a message after 120s, exit
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
WARNING  [bt.py:18] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected!
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner
INFO     [jikong.py:114] normal connect failed, connecting with scanner

my config file is:

- address: C8:47:8C:E5:AB:CB
  type: jk
  alias: JK-B2A24S20P

im not sure if thats right, but what i have in alias is what shows up on the jk app  or bluetooth discovery.   the mac address is correct.

im not 100% sure i have mqqt setup correctly, as their didnt seem to be anywhere to create in a username and password? - do i even need one? 

any help appriciated thanks  :-)

Catchmydrift, 2023-02-05:
then it becomes undiscoverable and i have to switch the bms off and on again to see it via bluetooth...

fl4p, 2023-02-05:
On which hardware do you run HA?

Catchmydrift, 2023-02-05:
> On which hardware do you run HA?

Hi, rpi 3b+..

jb-studio, 2023-02-06:
Hi,
I am experiencing some difficulties connecting to JK BMS as well on my RPI 3B+
I disabled build in Bluetooth and connected TP-Link Bluetooth dongle. I managed to successfully connect Daly BMS to my HA using Batmon addon, but when I am trying to establish a connection to JK BMS JK-B2A8S20P I see errors

`INFO     [sampling.py:67] connected bms JKBt(C8:47:8C:EC:76:D1)!
ERROR    [sampling.py:125] jk-bms error: <class 'AssertionError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:274] 1 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 1) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 176, in fetch
    return BmsSample(
  File "/app/bmslib/bms.py", line 57, in __init__
    assert math.isfinite(soc)
AssertionError`

Any successions on how it can be fixed?
Thanks

Catchmydrift, 2023-02-06:
Ive sorted the mqtt stuff, should be ok, but still getting the same problem in the log..?

Catchmydrift, 2023-02-10:
> On which hardware do you run HA?

Any help?

fl4p, 2023-02-25:
I just pushed a fix https://github.com/fl4p/batmon-ha/commit/cc8ed5d1f9800da84cf00021779fac9e9bf8a79e

Catchmydrift, 2023-04-29:
> 

Thank you very much for the update 👍👍.      I have 2 jk bmses and they are both now picked up!   This is awsome 👍👍.    Thank you again 😁😁👏👏👏👌👌👌👌✌️✌️✌️👍👍

closed 2023-04-30 by fl4p