fl4p/batmon-ha #54: Cannot connect JK BMS

Docharty, 2023-02-09:
Hi - sorry if I'm missing something obvious here, but I would really like to get this working.

First it appears to connect ok (it has the correct bt address & correct Pin)

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   JKbms   address=C8:47:8C:E5:02:41
INFO     [main.py:73] BT Device   6C-A6-94-6C-7B-04   address=6C:A6:94:6C:7B:04
INFO     [main.py:73] BT Device   5A-A1-02-59-86-96   address=5A:A1:02:59:86:96
INFO     [main.py:73] BT Device   N09XA   address=E1:2F:A1:F9:CA:62
INFO     [main.py:73] BT Device   Tile   address=FE:04:2B:6C:AF:EC
INFO     [bt.py:126] BMS JKBt(C8:47:8C:E5:02:41) keep alive enabled
INFO     [main.py:195] connecting mqtt mosquitto@core-mosquitto
INFO     [main.py:232] Fetching 1 BMS + 0 others concurrently, period=1.00s, keep_alive=True
INFO     [sampling.py:61] connecting bms JKBt(C8:47:8C:E5:02:41)
INFO     [main.py:106] mqtt watchdog loop started with timeout 120.0s

But then I get an error message:

asyncio.exceptions.TimeoutError
ERROR    [sampling.py:125] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:85] Error (num 6) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 167, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 44, in wait_for
    return await asyncio.wait_for(self._futures[name], timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc

Do you have an idea what the cause of this could be? I have added the pin in the configuration pin:  "1234" - I believe it needs the "" or it doesn't work. Or should I be adding the pin some other way? Thanks for your help.

Catchmydrift, 2023-02-10:
Hi, i have no help for you unfortunately as im in the same boat, but, do you need to enter a pin when you connect with your phone? As i dont, just the first time if i remember? And only when changing settings..

Docharty, 2023-02-10:
Thanks for the suggestion. I've tried removing the password & also deliberately putting in an incorrect password and it seems it comes up with the same error. So not sure if it's just not reading the password correctly or if there's another problem.

yarosman, 2023-02-11:
I have the same problem but for the jbd bms
```
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
  File "/app/bmslib/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 44, in wait_for
    return await asyncio.wait_for(self._futures[name], timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

Docharty, 2023-02-11:
So I've double checked MQQT connection. This is definitely working - all ok in the MQQT log. I've tried both the correct pin, the incorrect pin, and no pin at all, and I'm getting the same error message each time. Which makes me think it's not reading the pin correctly? Can I check the pin format should be
  pin: "1234"
?

fl4p, 2023-02-23:
batmon doesn't use the pin for JK. Actually JK password protection is fake, as authentication is done within the app (the BMS just sends the password in clear text). So I doubt that this issue is related to password.

Did you try to toggle keep_alive ?

fl4p, 2023-10-05:
TimeoutErrors can occur when there is an issue with the driver. Reboot, restart BMS. Try another Bluetooth adapter

closed 2023-10-05 by fl4p