fl4p/batmon-ha #55: Charge/Discharge switches problem

ornolio, 2023-02-13:
Thanks a lot for this sw BATMON rel. 50. It works quite well with hassio, running on Raspi 3 B.
I use it on 3 JBD BMS with 3 batteries packs 
48V (1 x LIFEPO4 and 2 x Lion), but it has one major problem: the two charge/discharge switches don't work especially switching from off to on, i.e. it appears as they don't receive immediate confirmation from the MQTT/ BMS 'hardware (halo around the button) then after a couple if secs the switch returns off.
My setup for 3 BMS is: sample_period 5 secs, publish_period 60 secs, concurrent_sampling on, keep_alive on, expire_after 120 secs.
I must say that the Android APP also has this difficulty, but with the APP after a few attempts the switch is successful, while with BATMON almost never. Has anyone had the same problem? Any suggestion? Thank You.

fl4p, 2023-04-07:
Fixed with 0.0.57. Please re-open if you still encounter problems

closed 2023-04-07 by fl4p

ornolio, 2023-04-08:
Thank You very much.
Best Regards
p

pier giacomo ornolio
***@***.***
***@***.***
pec: ***@***.***
+39 335 78 31 433