fl4p/batmon-ha #57: asyncio.exceptions.TimeoutError  when enable 2 or more jbd bms in configuration

v222008, 2023-02-21:
ERROR    [sampling.py:125] xiaoxiang_BMS_8KWT error: <class 'asyncio.exceptions.TimeoutError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [sampling.py:125] xiaoxiang_BMS_11KWT error: <class 'asyncio.exceptions.TimeoutError'>
INFO     [sampling.py:51] bms debug data: %s
ERROR    [main.py:274] 2 exceptions occurred fetching BMSs
ERROR    [main.py:85] Error (num 6) reading BMS: 
ERROR    [main.py:86] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 81, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    raise exceptions[0]
  File "/app/main.py", line 270, in fn
    await t
  File "/app/bmslib/sampling.py", line 47, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 69, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
  File "/app/bmslib/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 44, in wait_for
    return await asyncio.wait_for(self._futures[name], timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

fl4p, 2023-02-23:
Thats odd. Did you try toggle keep_alive ?

fl4p, 2023-04-07:
Please re-open if issue still exists

closed 2023-04-07 by fl4p