fl4p/batmon-ha #59: JK BMS connected but not all values correctly displayed in HA

zingreen, 2023-02-26:
**Description:**

I have a JK BMS that is successfully connected to Home Assistant, but I'm experiencing issues with the data being displayed. Specifically, only the values for "battery1 cell_voltages (x)" are correctly displayed, while all other values are either blank or incorrect, as can be seen in the attached screenshots.

I have checked the logs, but I can't find any visible errors. I have also tried restarting both the BMS and Home Assistant, but the issue persists.

Any suggestions on how to fix this issue or troubleshoot further would be greatly appreciated.

**Steps to reproduce:**

Connect the BMS to Home Assistant
Observe the displayed values in Home Assistant
**Expected behavior:**

All values from the BMS should be correctly displayed in Home Assistant.

**Actual behavior:**

Only the values for "battery1 cell_voltages (x)" are correctly displayed, while all other values are either blank or incorrect.

**Screenshots:**

https://gyazo.com/79408aead6b5664cced4d571f7ed2f0a
https://gyazo.com/1812fdc967762fef66e10b414a5396c5

**Environment:**

BMS model: JK_B2A8S20P; Firmware: 11.25H; Hardware: 11.XW
Home Assistant version: Home Assistant 2023.2.5
Operating system: [insert operating system here]
Logs:
```

INFO     [sampling.py:94] battery1 result@2023-02-26T15:33:11.288634 BmsSample(U=0.0V,I=-0.00A,P=-0W,q=15204.4Ah/0.0,mos=0.0°C)
DEBUG    [mqtt_util.py:102] topic battery1/meter/total_energy data not changed
DEBUG    [mqtt_util.py:102] topic battery1/meter/total_energy_discharge data not changed
DEBUG    [mqtt_util.py:102] topic battery1/meter/total_energy_charge data not changed
DEBUG    [mqtt_util.py:102] topic battery1/meter/total_charge data not changed
DEBUG    [mqtt_util.py:102] topic battery1/cell_voltages/1 data not changed
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m7507), 'b'battery1/cell_voltages/2'', ... (5 bytes)
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m7508), 'b'battery1/cell_voltages/3'', ... (5 bytes)
DEBUG    [mqtt_util.py:102] topic battery1/cell_voltages/4 data not changed
DEBUG    [mqtt_util.py:102] topic battery1/temperatures/1 data not changed
DEBUG    [mqtt_util.py:102] topic battery1/temperatures/2 data not changed
INFO     [sampling.py:103] battery1 volt=3335,3335,3335,3336 temp=[0.0, 0.0]
INFO     [sampling.py:128] JKBt(C8:47:8C:EA:BF:0D) times: connect=0.00s fetch=0.11s
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"U\xaa\xeb\x90\x02S\x07\r\x07\r\x07\r\x08\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x07\r\x01\x00\x00\x02'\x00'\x00&\x00&\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00")>}, []]
DEBUG    [jikong.py:72] header, clear buf bytearray(b'')
DEBUG    [jikong.py:77] bms msg(128) (buf128): 55 aa eb 90 2 53 7 d 7 d 7 d 8 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 f 0 0 0 7 d 1 0 0 2 27 0 27 0 26 0 26 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe8\x00\x00\x00\x00\x00')>}, []]
DEBUG    [jikong.py:77] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e8 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"\x1c4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc7\x00\xc5\x00\x00\x00\x00\x00\x00\x00\x00a\x02%\x04\x00\xc0E\x04\x00\x05\x00\x00\x00J\x04\x18\x00d\x00\x00\x00K\x8c'\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x9d\x03\x00\x00\x00\x00:\xccA@\x00\x00\x00\x006\x05\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00Z\x03\n\x00\x00\x00\x00\x00\xe8\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00")>}, []]
DEBUG    [jikong.py:77] bms msg(128) (buf278): 1c 34 0 0 0 0 0 0 0 0 0 0 c7 0 c5 0 0 0 0 0 0 0 0 61 2 25 4 0 c0 45 4 0 5 0 0 0 4a 4 18 0 64 0 0 0 4b 8c 27 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 9d 3 0 0 0 0 3a cc 41 40 0 0 0 0 36 5 0 0 0 1 0 1 0 5 0 0 5a 3 a 0 0 0 0 0 e8 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc/\x01\x01\x01\x00\x00\x00\x00J')>}, []]
DEBUG    [jikong.py:77] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc 2f 1 1 1 0 0 0 0 4a
DEBUG    [jikong.py:99] got response 2 (len300)
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"U\xaa\xeb\x90\x02T\x07\r\x07\r\x07\r\x07\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x07\r\x01\x00\x00\x02'\x00'\x00&\x00&\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00")>}, []]
DEBUG    [jikong.py:72] header, clear buf bytearray(b'')
DEBUG    [jikong.py:77] bms msg(128) (buf128): 55 aa eb 90 2 54 7 d 7 d 7 d 7 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 f 0 0 0 7 d 1 0 0 2 27 0 27 0 26 0 26 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe8\x00\x00\x00\x00\x00')>}, []]
DEBUG    [jikong.py:77] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e8 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"\x1c4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc7\x00\xc5\x00\x00\x00\x00\x00\x00\x00\x00a\x02%\x04\x00\xc0E\x04\x00\x05\x00\x00\x00J\x04\x18\x00d\x00\x00\x00L\x8c'\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x9d\x03\x00\x00\x00\x00:\xccA@\x00\x00\x00\x006\x05\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00_\x03\n\x00\x00\x00\x00\x00\xe8\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00")>}, []]
DEBUG    [jikong.py:77] bms msg(128) (buf278): 1c 34 0 0 0 0 0 0 0 0 0 0 c7 0 c5 0 0 0 0 0 0 0 0 61 2 25 4 0 c0 45 4 0 5 0 0 0 4a 4 18 0 64 0 0 0 4c 8c 27 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 9d 3 0 0 0 0 3a cc 41 40 0 0 0 0 36 5 0 0 0 1 0 1 0 5 0 0 5f 3 a 0 0 0 0 0 e8 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_BF_0D/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc/\x01\x01\x01\x00\x00\x00\x00P')>}, []]
DEBUG    [jikong.py:77] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc 2f 1 1 1 0 0 0 0 50
DEBUG    [jikong.py:99] got response 2 (len300)
INFO     [sampling.py:128] JKBt(C8:47:8C:EA:BF:0D) times: connect=0.00s fetch=0.06s
```

fl4p, 2023-02-27:
Hi,
thanks for the report.
Your JK appears to have support for 32 cells and has a slightly different protocol, which I did not implement in batmon yet. The frames are a bit larger for the extra cell data.
Will try to fix it in the next couple of days.

Personal note:
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L445

zingreen, 2023-02-27:
strange, but my bms model only supports a maximum of eight cells but it looks like the protocol is different.

Thank you very much for your support!

syssi, 2023-02-27:
Jikong did change their protocol. I assume your BMS (manufacturing date, hardware & software version) is pretty new. All models in future will come with the new protocol version probably, so they are prepared for future 32 cells models.

fl4p, 2023-03-01:
Pushed the fix to the master branch, will be in the next update

zingreen, 2023-03-01:
it works now. 
Thank you!

closed 2023-03-01 by zingreen