fl4p/batmon-ha #64: BlueZ doesnt work Raspi 4 with HA operation system

wowawabba, 2023-03-06:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
INFO     [main.py:68] BT Discovery:
INFO     [main.py:73] BT Device   40-16-08-01-YY-YY   address=40:16:08:01:YY:YY
INFO     [main.py:73] BT Device   DL-46640104XXXX   address=46:64:01:04:XX:XX
INFO     [main.py:153] Verbose logging enabled
INFO     [main.py:195] connecting mqtt batmon@192.168.1.100
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
DEBUG    [client.py:2529] Received CONNACK (0, 0)
INFO     [main.py:232] Fetching 1 BMS + 0 others concurrently, period=10.00s, keep_alive=False
INFO     [sampling.py:61] connecting bms DalyBt(46:64:01:04:XX:XX)
DEBUG    [client.py:181] Connecting to device @ 46:64:01:04:XX:XX with hci0
INFO     [main.py:106] mqtt watchdog loop started with timeout 120.0s
DEBUG    [scanner.py:132] cached devices: {'/org/bluez/hci0/dev_40_16_08_01_YY_YY': {'Address': '40:16:08:01:YY:YY', 'AddressType': 'random', 'Alias': '40-16-08-01-YY-YY', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -71, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {258: b'\x10@\x16\x08\x01.\x19'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_46_64_01_04_XX_XX': {'Address': '46:64:01:04:XX:XX', 'AddressType': 'public', 'Name': 'DL-46640104XXXX', 'Alias': 'DL-46640104XXXX', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -48, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'Modalias': 'usb:v045Ep0040d0300', 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {25670: b'\x01\x04\x01\xcd'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}}
DEBUG    [scanner.py:282] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [scanner.py:282] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_40_16_08_01_YY_YY): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -48)>}, []]
DEBUG    [scanner.py:282] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_46_64_01_04_XX_XX): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -42)>}, []]
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_46_64_01_04_XX_XX): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_46_64_01_04_XX_XX): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', False)>}, []]
DEBUG    [client.py:1182] Device disconnected (/org/bluez/hci0/dev_46_64_01_04_XX_XX)
DEBUG    [client.py:438] _remove_signal_handlers(/org/bluez/hci0/dev_46_64_01_04_XX_XX)
DEBUG    [client.py:465] _disconnect_message_bus(/org/bluez/hci0/dev_46_64_01_04_XX_XX)
ERROR    [sampling.py:125] batmon1 error: [org.bluez.Error.Failed] le-connection-abort-by-local
DEBUG    [client.py:438] _remove_signal_handlers(/org/bluez/hci0/dev_46_64_01_04_XX_XX)
DEBUG    [client.py:441] no bus object
DEBUG    [client.py:465] _disconnect_message_bus(/org/bluez/hci0/dev_46_64_01_04_XX_XX)
DEBUG    [client.py:468] already disconnected (/org/bluez/hci0/dev_46_64_01_04_XX_XX)
INFO     [sampling.py:61] connecting bms DalyBt(46:64:01:04:XX:XX)
DEBUG    [client.py:181] Connecting to device @ 46:64:01:04:XX:XX with hci0
DEBUG    [client.py:1123] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_46_64_01_04_XX_XX): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]

Is this related to the addon itself or the bluetooth of the Raspi4? Running the newest HA and batmon version

wowawabba, 2023-03-06:
bluetooth logs at console shows:
[CHG] Device 46:64:01:04:XX:XX RSSI: -42
[CHG] Device 46:64:01:04:XX:XX Connected: yes
[CHG] Device 46:64:01:04:XX:XX Connected: no
[CHG] Device 46:64:01:04:XX:XX RSSI: -42
[CHG] Device 46:64:01:04:XX:XX RSSI: -44
[CHG] Device 46:64:01:04:XX:XX Connected: yes
[CHG] Device 46:64:01:04:XX:XX Connected: no
[CHG] Device 46:64:01:04:XX:XX RSSI: -44
[CHG] Device 46:64:01:04:XX:XX RSSI: -41
[CHG] Device 46:64:01:04:XX:XX RSSI: -48
[CHG] Device 46:64:01:04:XX:XX Connected: yes
[CHG] Device 46:64:01:04:XX:XX Connected: no

so its always connected for a moment.

fl4p, 2023-03-06:
Did you try to power cycle the BMS and the RPI?
Also I just pushed an update, that fixes several things (although not related to this issue)

wowawabba, 2023-03-06:
I have restarted the Pi multiple times. After the upgrade its still the same issue. I will try to restart the BMS later today. I have also ordered a external supported bluetooth dongle just to verify the bluetooth is fine. Connection with my smartphone and the app works

wowawabba, 2023-03-08:
is it possible to configure which hci the addon should pick? It always pics hci0 which is the bluetooth of the PI but I want to use my bluetooth-adapter hci1

fl4p, 2023-03-10:
Will add this with the next release

fl4p, 2023-03-10:
You also might want to try disabling the on-board adapter https://github.com/fl4p/batmon-ha/issues/42#issuecomment-1377588841

Karatzie, 2024-03-23:
Hi Fabian.
is this issue with Raspi 4 and BlueZ with HA operation system solved by a new HA operation system?
I tryed to run your batmon HA with a Raspi 4.
I saw the same error messages from this issue in my batmon HA log.
I tried some configuration changes on my Raspi 4.
I disabled the onboard Bluetooth adapter over the config.txt file.
My Bluetooth dongle has now hci0.
I can use this Bluetooth dongle with my Bluetti AC200Max and get values from the Bluetti.
If I use bluetoothctl commands power off, power on and scan on the terminal I got no error messages in bluetoothctl.
The scan discover a bluetooth battery monitor located nearly by the Supervolt Batteries.
The Supervolt batteries are not discovered.

I replace my Raspi 3 because ESPHome runs into an error during building of a new FW.
The swap file is not big enough.
The next problem was the instable Bluetooth connection with my batteries.
After some time, the communication was stopped. 
Power cycle didnt help and at the end I mus reboot the Raspi 3.
That happens very often every day.

It will be greate if the batmon HA wil be running with my Raspi 4.

Can you support with this issue?

BR
Andreas