fl4p/batmon-ha #65: Problems with Daly BMS: asyncio.exceptions.TimeoutError

Quickred, 2023-03-06:
Bluetooth and MQTT are working correctly. But after batmon stopped after some minutes and I can not see any entities.
---
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
22:50:21 INFO [main] BT Discovery:
22:50:26 INFO [main] BT Device   55-F5-CF-D1-06-97   address=55:F5:CF:D1:06:97
22:50:26 INFO [main] BT Device   CF-4D-A3-04-73-E4   address=CF:4D:A3:04:73:E4
22:50:26 INFO [main] BT Device   57-DB-CB-A4-9C-45   address=57:DB:CB:A4:9C:45
22:50:26 INFO [main] BT Device   0F-12-21-51-CE-94   address=0F:12:21:51:CE:94
22:50:26 INFO [main] BT Device   ShellyPlus1PM-A8032ABBA0F4   address=A8:03:2A:BB:A0:F6
22:50:26 INFO [main] BT Device   38-AA-E2-1B-88-4A   address=38:AA:E2:1B:88:4A
22:50:26 INFO [main] BT Device   5A-28-31-87-52-06   address=5A:28:31:87:52:06
22:50:26 INFO [main] BT Device   44-5C-E9-8C-74-5D   address=44:5C:E9:8C:74:5D
22:50:26 INFO [main] BT Device   40-A3-24-FF-55-F3   address=40:A3:24:FF:55:F3
22:50:26 INFO [main] BT Device   F48F924E82DE3396EB   address=ED:ED:F8:9A:E7:62
22:50:26 INFO [main] BT Device   [TV] Fernseher   address=B8:BC:5B:A9:45:EF
22:50:26 INFO [main] BT Device   6A-65-2E-EB-48-D5   address=6A:65:2E:EB:48:D5
22:50:26 INFO [main] BT Device   0B-4A-CB-3D-65-32   address=0B:4A:CB:3D:65:32
22:50:26 INFO [main] BT Device   56-8A-02-A3-D3-F9   address=56:8A:02:A3:D3:F9
22:50:26 INFO [main] BT Device   CD-9A-DE-17-9D-02   address=CD:9A:DE:17:9D:02
22:50:26 INFO [main] BT Device   **DL-C66C09040D19   address=C6:6C:09:04:0D:19**
22:50:26 INFO [main] BT Device   98E7822CB2BC30AEE9   address=F3:33:CB:E0:36:2B
22:50:26 INFO [main] BT Device   98E7822E732230AEE9   address=F3:33:86:27:45:A1
22:50:26 INFO [main] BT Device   F2-6C-49-8F-38-27   address=F2:6C:49:8F:38:27
22:50:26 INFO [main] BT Device   C9-EE-8B-F8-D6-B4   address=C9:EE:8B:F8:D6:B4
22:50:26 INFO [main] BT Device   00-60-50-3B-82-C2   address=00:60:50:3B:82:C2
22:50:26 INFO [bt] BMS DalyBt(C6:6C:09:04:0D:19) keep alive enabled
22:50:26 INFO [main] connecting mqtt mqtt-user@core-mosquitto
22:50:26 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
22:50:26 INFO [sampling] connecting bms **DalyBt(C6:6C:09:04:0D:19)**
22:50:26 INFO [main] mqtt watchdog loop started with timeout 120.0s
22:50:31 INFO [daly] found rx uuid to be working: 17 (tx 15, sx 48)
22:50:31 INFO [sampling] connected bms DalyBt(C6:6C:09:04:0D:19)!
22:50:39 ERROR [sampling] battery1 error: <class '**asyncio.exceptions.TimeoutError**'>
22:50:39 ERROR [main] 1 exceptions occurred fetching BMSs
22:50:39 ERROR [main] Error (num 1, max 200) reading BMS: 
22:50:39 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 83, in fetch_loop
    await fn()
  File "/app/main.py", line 268, in fn
    raise exceptions[0]
  File "/app/main.py", line 263, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 72, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 122, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 149, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 108, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 46, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
22:50:48 ERROR [sampling] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
22:50:48 ERROR [main] 1 exceptions occurred fetching BMSs
22:50:48 ERROR [main] Error (num 2, max 200) reading BMS: 
22:50:48 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 83, in fetch_loop
    await fn()
  File "/app/main.py", line 268, in fn
    raise exceptions[0]
  File "/app/main.py", line 263, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 72, in sample
    sample = await bms.fetch()
  File "/app/bmslib/daly.py", line 122, in fetch
    status = await self._fetch_status()
  File "/app/bmslib/daly.py", line 149, in _fetch_status
    response_data = await self._q(0x93)
  File "/app/bmslib/daly.py", line 108, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 46, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
---
config in batmon:
- address: C6:6C:09:04:0D:19
  type: daly
  alias: battery1

---
Info from the SMAFT BMS app (iPhone):
Battery code: SH39F003
Software version: 210629
---
Is the version working with Batmon (version 0.0.53) ?

wshankles, 2023-03-21:
Same but with overkill/JBD.

fl4p, 2023-03-24:
Did you ever receive any sensor data ?

closed 2025-02-02 by fl4p