fl4p/batmon-ha #66: Can not add pin to victron shunt

sweety-muc, 2023-03-10:
Thank you for developing this wonderful software. My Daly-BMS worked oob :-)
My smart shunt is discovered, but when entering the pin in the config, it does not want to accept it.

- address: 40:16:06:01:xxxxx
  type: daly
  alias: Battery-0F
- address: 40:16:05:01:xxxx
  type: daly
  alias: Battery-DB
- address: D3:9F:05:AC:xxxx
  type: victron
  alias: SmartShunt
  pin: 000000

I get this: Failed to save add-on configuration, not a valid value. Got {'devices': [{'address': '40:16:06:01:11:0f', 'type': 'daly', 'alias': 'Battery-0F'}, {'address': '40:16:05:01:09:DB', 'type': 'daly', 'alias': 'Battery-DB'}, {'address': 'D3:9F:05:AC:88:DA', 'type': 'victron', 'alias': 'SmartShunt', 'pin': 0}], 'mqtt_user': 'homeassistant', 'mqtt_password': 'ChohHuuCh9cusae8eise2ouhi9si8gumoo4oCh7fee2eibefeighahzaibiengom', 'mqtt_broker': 'core-mosquitto', 'concurrent_sampling': False, 'keep_alive': True, 'verbose_log': False, 'sample_period': 1, 'publish_pe...

Do I use a wrong browser?

Thanks,

Chris

fl4p, 2023-03-10:
Hi, happy to hear it's working for your BMS.
Wrap the Pin with double quotes , e.g. `Pin: "000000"`

sweety-muc, 2023-03-10:
Thank you so much!

:-) I love your software for monitoring my cottage batterys.

Christoph

sweety-muc, 2023-03-10:
it saves the Pin “000000” now, but I get the same errors:

- address: D3:9F:05:AC:88:DA
  type: victron
  alias: VictronShunt
  pin: “000000"

(GATT is enabled)

22:23:17 ERROR [main] Error (num 21, max 200) reading BMS: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
22:23:17 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 83, in fetch_loop
    await fn()
  File "/app/main.py", line 268, in fn
    raise exceptions[0]
  File "/app/main.py", line 263, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 68, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 157, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 67, in connect
    self._values[k] = parse_value(await self.client.read_gatt_char(char['uuid']), char)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 823, in read_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
22:23:18 INFO [sampling] connecting bms SmartShuntBt(D3:9F:05:AC:88:DA)
22:23:19 ERROR [base_events] Task exception was never retrieved
future: <Task finished name='Task-115' coro=<SmartShuntBt._keep_alive_loop() done, defined at /app/bmslib/victron.py:55> exception=BleakDBusError('org.bluez.Error.Failed', 'Operation failed with ATT error: 0x0e')>
Traceback (most recent call last):
  File "/app/bmslib/victron.py", line 60, in _keep_alive_loop
    await self.client.write_gatt_char('6597ffff-4bda-4c1e-af4b-551c4cf74769', data, response=False)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 945, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
22:23:20 ERROR [sampling] VictronShunt error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
22:23:20 WARNING [bt] BMS SmartShuntBt(D3:9F:05:AC:88:DA) disconnected!

fl4p, 2023-03-11:
Did you enable GATT service in the Victron app?

fl4p, 2023-03-12:
Oh I see you did.
What hardware and OS are you running this on?

sweety-muc, 2023-03-13:
I am sorry I did not respond, I was in Italy for the weekend.

I am running it on an Raspberry Pi 4 with 2 GB, internal Bluetooth. Home Assistent actual Version without Docker.

Shunt is Firmware 4.12, Name: HQ2237QTG39
Bluetooth Interface is Rev2, Firmware 2.43
Bootloader 1.16, Instant Readout is activated.

Btw, The Dalys are also producing Errors, but most of the MQTT-Values are correct.

Are you interested in details and logs?

Thanks,

Christoph

fl4p, 2023-03-14:
Yes, you can post the logs here, please wrap it in a code block (```)

sweety-muc, 2023-03-30:
> Actually i just found a bug so the PIN was not passed to the pairing code. fixed with 0.0.55 <https://github.com/fl4p/batmon-ha/commit/6d7b6eee75ce25754d1b216b8ba2d20e82673b23>

Thank you. I will test it soon. I am suffering from corona right now ….

sweety-muc, 2023-04-01:
Hallo Fabian,

YOU ARE MY SUPERHERO! Thank you so much for this wonderful Integration. This is soooo cool!

It works flawless now!

Where can I send my coffeetip to?

Greetings,

Christoph

22:30:38 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:39 INFO [sampling] Battery-DB volt=3324,3321,3325,3325 temp=[12]
22:30:39 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:40 INFO [sampling] Battery-0F volt=3324,3325,3320,3323 temp=[10]
22:30:40 INFO [sampling] connecting bms SmartShuntBt(D3:9F:05:AC:88:DA)
22:30:40 INFO [bt] Device D3:9F:05:AC:88:DA asking for psk, giving '000000'
22:30:40 INFO [sampling] connected bms SmartShuntBt(D3:9F:05:AC:88:DA)!
22:30:40 INFO [sampling] VictronShunt: BmsSample(U=13.3V,I=-3.47A,P=-46W,q=-26.6Ah/-27.0,mos=nan°C)
22:30:41 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:42 INFO [sampling] Battery-0F volt=3324,3324,3322,3323 temp=[10]
22:30:42 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:43 INFO [sampling] Battery-DB volt=3324,3321,3325,3325 temp=[12]
22:30:44 INFO [sampling] VictronShunt: BmsSample(U=13.3V,I=-3.29A,P=-44W,q=-26.6Ah/-27.0,mos=nan°C)
22:30:44 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:45 INFO [sampling] Battery-0F volt=3324,3325,3322,3323 temp=[10]
22:30:46 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:47 INFO [sampling] Battery-DB volt=3324,3321,3325,3324 temp=[12]
22:30:48 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:48 INFO [sampling] Battery-0F volt=3325,3322,3323,3322 temp=[10]
22:30:49 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:49 INFO [sampling] Battery-DB volt=3324,3321,3325,3325 temp=[12]
22:30:49 INFO [sampling] VictronShunt: BmsSample(U=13.3V,I=-3.20A,P=-42W,q=-26.6Ah/-27.0,mos=nan°C)
22:30:51 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:51 INFO [sampling] Battery-0F volt=3324,3325,3319,3323 temp=[10]
22:30:52 INFO [sampling] VictronShunt: BmsSample(U=13.3V,I=-3.34A,P=-44W,q=-26.6Ah/-27.0,mos=nan°C)
22:30:52 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:53 INFO [sampling] Battery-DB volt=3325,3321,3325,3324 temp=[12]
22:30:53 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:54 INFO [sampling] Battery-0F volt=3324,3323,3323,3323 temp=[10]
22:30:55 INFO [sampling] Battery-DB: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)
22:30:56 INFO [sampling] Battery-DB volt=3324,3321,3324,3324 temp=[12]
22:30:56 INFO [sampling] VictronShunt: BmsSample(U=13.3V,I=-3.21A,P=-43W,q=-26.6Ah/-27.0,mos=nan°C)
22:30:56 INFO [sampling] Battery-0F: BmsSample(U=13.2V,I=0.00A,P=0W,q=276.6Ah/280.0,mos=nan°C)

Delphinus44, 2023-04-02:
First thanks for this great tool, I have 3 JK and it works with them and I also have a Victron Smart shunt and it never manage to connect.
After install .56 it doesn't connect and keep restarting the integration, I try pin with and without ""

01:01:48 INFO [sampling] connecting bms SmartShuntBt(F4:13:D2:6C:41:79)
01:01:48 INFO [main] mqtt watchdog loop started with timeout 120.0s
01:01:52 ERROR [sampling] LiFePo4 error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
01:01:52 ERROR [main] 1 exceptions occurred fetching BMSs
01:01:52 ERROR [main] Error (num 1, max 200) reading BMS: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
01:01:52 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 45, in fetch_loop
    await fn()
  File "/app/main.py", line 234, in fn
    raise exceptions[0]
  File "/app/main.py", line 229, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 68, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 164, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 67, in connect
    self._values[k] = parse_value(await self.client.read_gatt_char(char['uuid']), char)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 823, in read_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
01:01:52 WARNING [bt] BMS SmartShuntBt(F4:13:D2:6C:41:79) disconnected!

Tpalagyi82, 2023-04-03:
Hi, Thank you soooooooo much, finally with 0.56 my Victron Shunt is also works!!!!
Here is a sample log from 2 jk bms and the Victron Shunt:
06:52:12 INFO [sampling] victronshunt1: BmsSample(U=52.2V,I=-55.80A,P=-2911W,q=-140.9Ah/-184.0,mos=nan°C)
06:52:13 INFO [sampling] battery2: BmsSample(U=52.4V,I=-30.11A,P=-1579W,q=222.3Ah/285.0,mos=17.7°C)
06:52:13 INFO [sampling] battery2 volt=3280,3284,3282,3274,3275,3280,3282,3273,3279,3278,3274,3277,3280,3284,3279,3274 temp=[16.5, 13.3]
06:52:17 INFO [sampling] battery1: BmsSample(U=52.4V,I=-25.65A,P=-1343W,q=191.2Ah/261.9,mos=17.6°C)
06:52:17 INFO [sampling] battery1 volt=3276,3270,3276,3282,3275,3276,3280,3277,3270,3269,3270,3276,3277,3266,3264,3270 temp=[17.9, 14.8]
Best wishes!
Regards,
Tamas

Delphinus44, 2023-04-03:
Can one of you that have if now working paste the YAML of your config to compare with mine, thanks.

fl4p, 2023-04-03:
@Delphinus44 did you enable GATT service in the Victron app?

Delphinus44, 2023-04-03:
Yes GATT is enabled

sweety-muc, 2023-04-03:
I tried to edit the config on Mac with Safari (Ventura 13.3, Safari 16.4 (18615.1.26.11.22)). There it was not possible to edit the code lines. The regarding window on the config-page was not visible. Editing it with Chrome on Mac was flawless.

Delphinus44, 2023-04-04:
> @Delphinus44 did you enable GATT service in the Victron app?

Today I removed batmon and install it again and got the same result, then I try to edit the YAML file to put my config and I notice that if I edit only parts of it I can't save the configuration. If I paste the full YAML it allow save, also it never allow to put the pin after type it always go to the bottom of the file.

My YAML:
devices:
  - address: F4:13:D2:6C:41:79
    type: victron
mqtt_user: xxxxxx
mqtt_password: xxxxxx
mqtt_broker: core-mosquitto
concurrent_sampling: false
keep_alive: true
verbose_log: false
sample_period: 2
publish_period: 4
invert_current: true
watchdog: true
expire_values_after: 20
pin: xxxxxx

[image]

fl4p-old, 2023-04-04:
You yaml is not correct, see https://github.com/fl4p/batmon-ha/blob/master/config.yaml#L16 and follow the scheme

Delphinus44, 2023-04-04:
> You yaml is not correct, see https://github.com/fl4p/batmon-ha/blob/master/config.yaml#L16 and follow the scheme

Sorry but I don't understand why, this is your default yaml changed for my setup, I only add the pin at the end.

[image]

[image]

fl4p, 2023-04-07:
Pin goes under devices

closed 2023-04-07 by fl4p

Delphinus44, 2023-04-07:
I put pin under devices has I save it goes to the bottom line, victron pin still does'nt  work, just installed latest victron firmware v.4.14 .
It works with pin at bottom using JK BMS.
You made a good instance but still need to fix it for victron, probably you should had pin in configuration area.

Delphinus44, 2023-04-08:
It is working now with victron, I'm using 0.0.57 and reduce sample period to 3, don't know if related but the connection time was over 1 second, also must have keep alive otherwise from time to time I got the older error. 
Even with keep alive the victron app show the preview info :)
Thanks for this software.