fl4p/batmon-ha #67: Feature Request & Implementation: Add AppArmor support for supervised docker installation

MariusHerget, 2023-03-17:
Hey, 

I tried to install your add-on in a supervised docker installation and apparmor was denying me access to the dbus. I already had the same issue with another add-on and tried the same fix locally: Works perfectly! 

Steps:
1. Add `apparmor: true` to `config.yaml`
2. Copy the attached file `apparmor.txt` next to `config.yaml` in the main folder. It is an adapted version of [the other addon](https://github.com/lolouk44/hassio-addons/blob/master/mi-scale/apparmor.txt) where I helped to fix the same [issue](https://github.com/lolouk44/hassio-addons/issues/67).

[apparmor.txt](https://github.com/fl4p/batmon-ha/files/11005995/apparmor.txt)

Thanks in advance!

fl4p, 2023-03-18:
Hi Marius,
thanks for this! Will snap this for the next release.

fl4p, 2023-04-03:
https://github.com/fl4p/batmon-ha/commit/bddf1167ab653e8e4d8b89860c1d85731168bd0a

closed 2023-04-03 by fl4p