fl4p/batmon-ha #68: DALY connection abort by local

Flesky95, 2023-03-22:
Hi,
I configure the add-on correctly, sometimes start to comunicate and work for some minutes. After the addon stop works and lost communication. I try to change Bluetooth usb dongle on the thin client I use for HA, use an extension cable, different usb port (2.0 and 3.0), disconnect and reconnect several times the Bluetooth dongle on the bms with no change.
Bms Bluetooth works correctly when I try to use the smartphone app. Distance between bms and HA machine: 2mt 

This is the log:

12:44:56 INFO [sampling] connecting bms DalyBt(18:81:03:04:00:5B)
12:45:01 ERROR [sampling] BATT error: [org.bluez.Error.Failed] le-connection-abort-by-local
12:45:01 ERROR [main] 1 exceptions occurred fetching BMSs
12:45:01 ERROR [main] Error (num 6, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
12:45:01 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 83, in fetch_loop
    await fn()
  File "/app/main.py", line 268, in fn
    raise exceptions[0]
  File "/app/main.py", line 263, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 68, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 157, in __aenter__
    await self.connect()
  File "/app/bmslib/daly.py", line 68, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 69, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 40, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
12:45:01 WARNING [bt] BMS DalyBt(18:81:03:04:00:5B) disconnected!

And the configuration :

devices:
  - address: 18:81:03:04:00:5B
    type: daly
    alias: BATT
mqtt_user: ***
mqtt_password: ***
mqtt_broker: core-mosquitto
concurrent_sampling: false
keep_alive: true
verbose_log: false
sample_period: 2
publish_period: 2
invert_current: false
watchdog: true
expire_values_after: 60

Thx for your work and help

fl4p, 2023-03-24:
Is there current flowing through the BMS? The Daly has a sleep mode which makes BT unavailable until the BMS senses some current.

Flesky95, 2023-03-24:
I use the “magic value” 15300 for sleep time and the BMS works all the time with Bluetooth. No issue if I try with the app.
When I try to connect my HA I disable my phone Bluetooth too to avoid other problem

fl4p, 2023-03-24:
Ok, then its not the sleep mode.

Does restarting batmon help?

Flesky95, 2023-03-24:
Previously 1/100 times yes. Now not at all. I try several times to disconnect and reconnect the BMS Bluetooth dongle like you write in the docs, but nothing change. I had restart the server machine too and try to uninstall and reinstall Batmon

Flesky95, 2023-03-24:
This is a verbose log after several tries 

08:54:22 INFO [main] BT Discovery:
08:54:27 INFO [main] BT Device   64-07-F6-B8-99-37   address=64:07:F6:B8:99:37
08:54:27 INFO [main] BT Device  DL-18810304005B   address=18:81:03:04:00:5B 
08:54:27 INFO [main] BT Device   60-A6-91-E7-DF-4D   address=60:A6:91:E7:DF:4D
08:54:27 INFO [main] Verbose logging enabled
08:54:27 INFO [bt] BMS DalyBt(18:81:03:04:00:5B) keep alive enabled
08:54:27 INFO [main] connecting mqtt mqtt-user@core-mosquitto
08:54:27 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
08:54:27 DEBUG [client] Received CONNACK (0, 0)
08:54:27 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
08:54:27 INFO [sampling] connecting bms DalyBt(18:81:03:04:00:5B)
08:54:27 DEBUG [client] Connecting to device @ 18:81:03:04:00:5B with hci0
08:54:27 INFO [main] mqtt watchdog loop started with timeout 120.0s
08:54:27 DEBUG [scanner] cached devices: {'/org/bluez/hci0/dev_18_81_03_04_00_5B': {'Address': '18:81:03:04:00:5B', 'AddressType': 'public', 'Name': 'DL-18810304005B', 'Alias': 'DL-18810304005B', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -88, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'Modalias': 'usb:v045Ep0040d0300', 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {33048: b'\x03\x04\x00['}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_64_07_F6_B8_99_37': {'Address': '64:07:F6:B8:99:37', 'AddressType': 'public', 'Alias': '64-07-F6-B8-99-37', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -86, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {117: b'B\x04\x01\x80fd\x07\xf6\xb8\x997f\x07\xf6\xb8\x996\x01\x00\x00\x00\x00\x00\x00'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x00'}, '/org/bluez/hci0/dev_60_A6_91_E7_DF_4D': {'Address': '60:A6:91:E7:DF:4D', 'AddressType': 'random', 'Alias': '60-A6-91-E7-DF-4D', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -78, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05\x03\x18#\xd0\xa9'}, 'TxPower': 4, 'ServicesResolved': False, 'AdvertisingFlags': b'\x1a'}}
08:54:27 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
08:54:29 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_A6_91_E7_DF_4D): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -78)>}, []]
08:54:32 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_18_81_03_04_00_5B): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -88)>}, []]
08:54:32 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_64_07_F6_B8_99_37): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -84)>}, []]
08:54:37 DEBUG [client] _remove_signal_handlers(/org/bluez/hci0/dev_18_81_03_04_00_5B)
08:54:37 DEBUG [client] _disconnect_message_bus(/org/bluez/hci0/dev_18_81_03_04_00_5B)
08:54:37 ERROR [sampling] BATT error: [org.bluez.Error.Failed] le-connection-abort-by-local
08:54:37 ERROR [main] 1 exceptions occurred fetching BMSs
08:54:37 ERROR [main] Error (num 1, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
08:54:37 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 83, in fetch_loop
    await fn()
  File "/app/main.py", line 268, in fn
    raise exceptions[0]
  File "/app/main.py", line 263, in fn
    await t
  File "/app/bmslib/sampling.py", line 50, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 68, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 157, in __aenter__
    await self.connect()
  File "/app/bmslib/daly.py", line 68, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 69, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 40, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
08:54:38 INFO [sampling] connecting bms DalyBt(18:81:03:04:00:5B)
08:54:38 DEBUG [client] Connecting to device @ 18:81:03:04:00:5B with hci0
08:54:40 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_18_81_03_04_00_5B): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
08:54:40 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_18_81_03_04_00_5B): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', False)>}, [ [... cut ...]
08:54:46 DEBUG [client] Write Characteristic f000ffc2-0451-4000-b000-000000000000 | /org/bluez/hci0/dev_18_81_03_04_00_5B/service002b/char0030: bytearray(b'')
08:54:46 INFO [daly] found rx uuid to be working: 17 (tx 15, sx 48)
08:54:46 INFO [sampling] connected bms DalyBt(18:81:03:04:00:5B)!
08:54:46 DEBUG [daly] daly send: bytearray(b'\xa5\x80\x93\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc0')
08:54:46 DEBUG [client] Write Characteristic 0000fff2-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_18_81_03_04_00_5B/service000e/char000f: bytearray(b'\xa5\x80\x93\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc0')

fl4p, 2023-03-24:
To track this down further, please downgrade to [0.0.46](https://github.com/fl4p/batmon-ha/releases/tag/release0.0.46)
(Remove the add-on store repository and then install this as a local add-on)

Flesky95, 2023-03-24:
Ok, I try. Simple copy the batmon addon folder to /addons directory and restart?

Flesky95, 2023-03-24:
I do all but same problem with 0.0.46
You need logs too?

fl4p, 2023-03-26:
No logs needed, thanks.

Please try this version https://github.com/fl4p/batmon-ha/releases/tag/daly-scanner
It enables BT discovery before connecting.

Flesky95, 2023-03-27:
Ok, I’ll try it this evening. Thx

fl4p, 2023-04-10:
Closing this, please re-open if you still encounter problems

closed 2023-04-10 by fl4p