fl4p/batmon-ha #71: Feature Request - Convenient Standalone Upgrade Method

klarjr, 2023-03-31:
I'd like to first say how appreciative I am that precious time was spent developing this incredibly useful add-on and it works amazingly well.  

This is a low priority request but still one I think would have value. For reasons unimportant, I have to run this as a standalone service, remote from my Home Assistant installation. I'm able to ssh into the system and run a handful of commands to perform an upgrade when a new version is released. It would be great for the Home Assistant add-on to be configurable as either the primary service server or as a client for management of a remote standalone server. Ideally, I'd like to have full control within Home assistant of the standalone service - Configuring, Starting,Stopping, Upgrading, etc... Short of this complicated request, it would be amazing if a simple python upgrade file could be included that could simply be run through ssh to automate the upgrade process. 

Thanks for taking the time to read my long winded request and regardless if you decide to take on my request, know that I'm very grateful for the work you've already put in and for any anything you decide to do in the future.

fl4p, 2023-04-01:
You can write a shell script that polls the git repository daily
```
if [ -z "$(git pull 2> /dev/null | grep 'Already up to date.')" ]; then
	echo "changes on remote, updating"
	git pull
	... restart etc...
fi
```

Maybe you should consider some orchestration software like kubernetes, portainer.
I use [Cronicle](https://github.com/jhuckaby/Cronicle) a lot to control tasks remotely.

This request out of scope of batmon.

klarjr, 2023-04-02:
Understood and thanks for the tips. I'll look into the recommended solutions. Appreciate the fast and detailed response. Thanks again for making this available.

closed 2023-04-02 by klarjr