fl4p/batmon-ha #79: Energy Meters

Delphinus44, 2023-04-09:
I have 2 devices installed but can only see energy meters for one off them, have the readings in the log, Mosquitto broker only show one.

"15:47:50 INFO [sampling] LiFePo4: BmsSample(U=54.7V,I=87.90A,P=4804W,q=-49.2Ah/-55.0,mos=nan°C)
15:48:00 INFO [sampling] EV280 K: BmsSample(U=54.5V,I=52.88A,P=2884W,q=193.5Ah/263.3,mos=34.3°C)
15:48:01 INFO [sampling] EV280 K volt=3406,3420,3409,3399,3395,3410,3434,3424,3418,3413,3405,3395,3399,3409,3405,3394 temp=[30.6, 31.2]"

[image]

Delphinus44, 2023-04-10:
Looking at MQTT logs found the problem, we cannot use a space in alias, Mqtt discovery only accept standard characters.

closed 2023-04-10 by Delphinus44

fl4p, 2023-04-10:
I'll fix this with the next update