fl4p/batmon-ha #82: connection rejected due to security problem

ornolio, 2023-04-15:
4 jbd BMS, working fine until v.53, password not changed, all bts working fine with jbd APP on Android. 
sampling period 2 secs, publish_period 15secs, concurrent_sampling yes, keep_alive yes, expire_values_after 120secs. Raspi 3 B, internal bt antenna, distabce 50cm to 2 meters.
Could you help?
Thamk You.
B.Regards.

log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
07:09:10 INFO [main] BT Discovery:
07:09:15 INFO [main] BT Device   Lion 2,5 kWh   address=A4:C1:38:BB:0E:90
07:09:15 INFO [main] BT Device   Lion6kwh   address=A4:C1:38:E2:C4:86
07:09:15 INFO [main] BT Device   LIFEPO15kWh   address=A4:C1:37:30:7C:CA
07:09:15 INFO [main] BT Device   LIFEPO5kWh   address=A4:C1:38:46:6F:DA
07:09:15 INFO [bt] BMS JbdBt(A4:C1:38:46:6F:DA) keep alive enabled
07:09:15 INFO [bt] BMS JbdBt(A4:C1:38:E2:C4:86) keep alive enabled
07:09:15 INFO [bt] BMS JbdBt(A4:C1:38:BB:0E:90) keep alive enabled
07:09:15 INFO [bt] BMS JbdBt(A4:C1:37:30:7C:CA) keep alive enabled
07:09:15 INFO [main] connecting mqtt ornolio@core-mosquitto
07:09:15 INFO [main] Fetching 4 BMS + 0 others concurrently, period=2.00s, keep_alive=True
07:09:15 INFO [sampling] connecting bms JbdBt(A4:C1:38:46:6F:DA)
07:09:17 ERROR [client] Pairing /org/bluez/hci0/dev_A4_C1_38_46_6F_DA with hci0 failed (org.bluez.Error.AuthenticationFailed)
07:09:17 ERROR [bt] Pairing failed!
07:09:17 ERROR [sampling] LIFEPO5kWh error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
07:09:17 INFO [sampling] connecting bms JbdBt(A4:C1:38:E2:C4:86)
07:09:17 WARNING [bt] BMS JbdBt(A4:C1:38:46:6F:DA) disconnected!
07:09:18 ERROR [client] Pairing /org/bluez/hci0/dev_A4_C1_38_E2_C4_86 with hci0 failed (org.bluez.Error.AuthenticationFailed)
07:09:18 ERROR [bt] Pairing failed!
07:09:18 ERROR [sampling] LION6kWh error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
07:09:18 INFO [sampling] connecting bms JbdBt(A4:C1:38:BB:0E:90)
07:09:18 WARNING [bt] BMS JbdBt(A4:C1:38:E2:C4:86) disconnected!
07:09:19 ERROR [client] Pairing /org/bluez/hci0/dev_A4_C1_38_BB_0E_90 with hci0 failed (org.bluez.Error.AuthenticationFailed)
07:09:19 ERROR [bt] Pairing failed!
07:09:20 ERROR [sampling] LION2kWh error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
07:09:20 INFO [sampling] connecting bms JbdBt(A4:C1:37:30:7C:CA)
07:09:20 WARNING [bt] BMS JbdBt(A4:C1:38:BB:0E:90) disconnected!
07:09:21 ERROR [client] Pairing /org/bluez/hci0/dev_A4_C1_37_30_7C_CA with hci0 failed (org.bluez.Error.AuthenticationFailed)
07:09:21 ERROR [bt] Pairing failed!
07:09:21 ERROR [sampling] LIFEPO15kWh error: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)
07:09:21 INFO [sampling] connecting bms JbdBt(A4:C1:38:46:6F:DA)
07:09:21 INFO [sampling] connecting bms JbdBt(A4:C1:38:E2:C4:86)
07:09:21 INFO [sampling] connecting bms JbdBt(A4:C1:38:BB:0E:90)
07:09:21 INFO [sampling] connecting bms JbdBt(A4:C1:37:30:7C:CA)
07:09:21 ERROR [sampling] LIFEPO15kWh error: This service is already present in this BleakGATTServiceCollection!
07:09:21 ERROR [main] Error (num 1, max 0) reading BMS: This service is already present in this BleakGATTServiceCollection!
07:09:21 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 45, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 51, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 70, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 172, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 51, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 77, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 48, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 319, in connect
    self.services.add_service(BleakGATTServiceBlueZDBus(obj, path))
  File "/usr/lib/python3.10/site-packages/bleak/backends/service.py", line 128, in add_service
    raise BleakError(
bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!
07:09:21 WARNING [bt] BMS JbdBt(A4:C1:37:30:7C:CA) disconnected!

fl4p-old, 2023-04-16:
did you set PINs in the batmon options?
if yes, remove it

ornolio, 2023-04-17:
Thank you for your reply.
Yes I'd the pin on all 4 BMS configuration.
Then as for your suggestion I removed them all and now 3 of 4 ble are 
working fine. Next days I'll replace the last BMS ble and I'll let you know.

Thank You very much for your help.

P.S.
by the way, here is the old configuration, before removing the pin/psw.

- address: A4:C1:38:46:6F:DA
  type: jbd
  alias: LIFEPO5kWh
  pin: "123456"
- address: A4:C1:38:E2:C4:86
  type: jbd
  alias: LION6kWh
  pin: "1234567" <=this psw changed from default
- address: A4:C1:38:BB:0E:90
  type: jbd
  alias: LION2kWh <=this not working yet
  pin: "123456"
- address: A4:C1:37:30:7C:CA
  type: jbd
  alias: LIFEPO15kWh
  pin: "123456"

pier giacomo ornolio
***@***.***
***@***.***
pec: ***@***.***
+39 335 78 31 433

fl4p, 2023-04-19:
glad its working

closed 2023-04-19 by fl4p