fl4p/batmon-ha #83: JK BMS Error after connected

albertomontoro2023, 2023-04-18:
Hi all,
I see all issues and dont see any similar.
I have a JK-B1A20S15P
Hardware Ver: V10.XW
Soft Ver: V10.07
Manuf. Date: 2022-02-20
Version: V4.11.0
Its working fine for more than 1 year and now I just discover Home Assistant and I would like to connect my BMS with HA, but I dont understand where is my problem

My Home Assistant is version:
Home Assistant 2023.4.5
Supervisor 2023.04.0
Operating System 10.0
Frontend 20230411.1 - latest

MQTT is well configured and working fine with other components,

Configuration in Batmon (Version 0.0.57)
- address: 20:21:11:28:18:CE
  type: jk
  alias: JK-B1A20S15P
  debug: true

I try with PIN, without PIN, enabling and disabling keep alive, etc.....
Because I am not a progrmmer, can anyone help me with the next log?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
19:26:13 INFO [main] BT Discovery:
19:26:18 INFO [main] BT Device   JK-B1A20S15P   address=20:21:11:28:18:CE
19:26:18 INFO [main] BT Device   09-D0-C6-BF-57-65   address=09:D0:C6:BF:57:65
19:26:18 INFO [main] BT Device   5B-5E-31-AC-66-E9   address=5B:5E:31:AC:66:E9
19:26:18 INFO [main] BT Device   6C-5D-CF-20-91-06   address=6C:5D:CF:20:91:06
19:26:18 INFO [main] BT Device   49-CD-CF-DF-10-1B   address=49:CD:CF:DF:10:1B
19:26:18 INFO [main] BT Device   43-6E-50-51-AE-C9   address=43:6E:50:51:AE:C9
19:26:18 INFO [main] BT Device   Mi Smart Band 6   address=C6:77:22:7A:07:AE
19:26:18 INFO [main] BT Device   7F-7D-55-9E-BE-97   address=7F:7D:55:9E:BE:97
19:26:18 INFO [main] Verbose logging enabled
19:26:18 INFO [main] Verbose log for 20:21:11:28:18:CE enabled
19:26:18 INFO [bt] BMS JKBt(20:21:11:28:18:CE) keep alive enabled
19:26:18 INFO [main] connecting mqtt mqttuser@core-mosquitto
19:26:18 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
19:26:18 DEBUG [client] Received CONNACK (0, 0)
19:26:18 INFO [main] Fetching 1 BMS + 0 others serially, period=20.00s, keep_alive=True
19:26:18 INFO [sampling] connecting bms JKBt(20:21:11:28:18:CE)
19:26:18 DEBUG [client] Connecting to device @ 20:21:11:28:18:CE with hci0
19:26:18 INFO [main] mqtt watchdog loop started with timeout 120.0s
19:26:18 DEBUG [scanner] cached devices: {'/org/bluez/hci0/dev_7F_7D_55_9E_BE_97': {'Address': '7F:7D:55:9E:BE:97', 'AddressType': 'random', 'Alias': '7F-7D-55-9E-BE-97', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05\x02\x18>\xd6\x8a'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x1a'}, '/org/bluez/hci0/dev_C6_77_22_7A_07_AE': {'Address': 'C6:77:22:7A:07:AE', 'AddressType': 'public', 'Name': 'Mi Smart Band 6', 'Alias': 'Mi Smart Band 6', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fee0-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x03\xc6w"z\x07\xae'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_5B_5E_31_AC_66_E9': {'Address': '5B:5E:31:AC:66:E9', 'AddressType': 'random', 'Alias': '5B-5E-31-AC-66-E9', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05\x02\x1c<\x91U'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x1a'}, '/org/bluez/hci0/dev_20_21_11_28_18_CE': {'Address': '20:21:11:28:18:CE', 'AddressType': 'public', 'Name': 'JK-B1A20S15P', 'Alias': 'JK-B1A20S15P', 'Paired': False, 'Bonded': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000ffe0-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {19274: b'\x00\x01'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}}
19:26:18 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -73)>}, []]
19:26:18 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
19:26:18 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -78)>}, []]
19:26:18 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {}, ['RSSI']]
19:26:18 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', False)>}, []]
19:26:18 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -78)>}, []]
19:26:18 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
19:26:19 DEBUG [client] Connection successful (/org/bluez/hci0/dev_20_21_11_28_18_CE)
19:26:19 DEBUG [client] Waiting for ServicesResolved (/org/bluez/hci0/dev_20_21_11_28_18_CE)
19:26:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_20_21_11_28_18_CE): ['org.bluez.Device1', {'ServicesResolved': <dbus_next.signature.Variant ('b', True)>}, []]
19:26:19 ERROR [sampling] JK-B1A20S15P error: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.
19:26:19 ERROR [main] 1 exceptions occurred fetching BMSs
19:26:19 ERROR [main] Error (num 1, max 200) reading BMS: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.
19:26:19 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 45, in fetch_loop
    await fn()
  File "/app/main.py", line 240, in fn
    raise exceptions[0]
  File "/app/main.py", line 235, in fn
    await t
  File "/app/bmslib/sampling.py", line 51, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 70, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 172, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 122, in connect
    await self.client.start_notify(self.UUID_RX, self._notification_handler)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 1041, in start_notify
    characteristic = self.services.get_characteristic(char_specifier)
  File "/usr/lib/python3.10/site-packages/bleak/backends/service.py", line 185, in get_characteristic
    raise BleakError(
bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.

hddmax, 2023-04-19:
i have the same problem, since when i upgrade to 10.0 and 2023.4.5 at 18h  
[image]

albertomontoro2023, 2023-04-19:
My error was with prrvious version of Home Assistant. After upgrade to 10.0 same error.

fl4p, 2023-04-19:
> i have the same problem, since when i upgrade to 10.0 and 2023.4.5 at 18h [image]
> 
> 13:05:12 ERROR [sampling] BANK3 error: Device with address A4:C1:38:E6:3F:18 was not found. 13:05:12 INFO [sampling] connecting bms JbdBt(A4:C1:38:A2:EA:D8) 13:05:13 INFO [sampling] connected bms JbdBt(A4:C1:38:A2:EA:D8)! 13:05:13 INFO [sampling] BANK2: BmsSample(U=53.8V,I=6.22A,P=335W,q=268.0Ah/280.0,mos=nan°C) 13:05:14 INFO [sampling] BANK2 volt=3365,3363,3363,3363,3365,3362,3361,3367,3366,3364,3361,3364,3365,3364,3366,3364 temp=[19.2] 13:05:15 INFO [sampling] JbdBt(A4:C1:38:A2:EA:D8) times: connect=1.65s fetch=0.27s 13:05:15 INFO [sampling] connecting bms JbdBt(A4:C1:38:2A:95:DB) 13:05:36 ERROR [sampling] BANK1 error: Device with address A4:C1:38:2A:95:DB was not found. 13:05:36 ERROR [main] 2 exceptions occurred fetching BMSs 13:05:36 ERROR [main] Error (num 3, max 200) reading BMS: Device with address A4:C1:38:E6:3F:18 was not found. 13:05:36 ERROR [main] Stack: Traceback (most recent call last): File "/app/main.py", line 45, in fetch_loop await fn() File "/app/main.py", line 239, in fn raise exceptions[0] File "/app/main.py", line 234, in fn await t File "/app/bmslib/sampling.py", line 51, in **call** return await self.sample() File "/app/bmslib/sampling.py", line 70, in sample async with bms: File "/app/bmslib/bt.py", line 172, in **aenter** await self.connect() File "/app/bmslib/jbd.py", line 51, in connect await super().connect(**kwargs) File "/app/bmslib/bt.py", line 77, in connect await self._connect_client(timeout=timeout) File "/app/bmslib/bt.py", line 48, in _connect_client await self.client.connect(timeout=timeout) File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect raise BleakError( bleak.exc.BleakError: Device with address A4:C1:38:E6:3F:18 was not found. 13:05:51 INFO [sampling] connecting bms JbdBt(A4:C1:38:A2:EA:D8) 13:05:52 INFO [sampling] connected bms JbdBt(A4:C1:38:A2:EA:D8)! 13:05:52 INFO [sampling] BANK2: BmsSample(U=53.8V,I=7.65A,P=412W,q=268.1Ah/280.0,mos=nan°C) 13:05:52 INFO [sampling] BANK2 volt=3367,3364,3365,3365,3367,3362,3363,3369,3367,3367,3363,3364,3367,3366,3367,3365 temp=[19.3] 13:05:54 INFO [sampling] JbdBt(A4:C1:38:A2:EA:D8) times: connect=1.02s fetch=0.21s 13:05:54 INFO [sampling] connecting bms JbdBt(A4:C1:38:2A:95:DB) 13:06:15 ERROR [sampling] BANK1 error: Device with address A4:C1:38:2A:95:DB was not found. 13:06:15 INFO [sampling] connecting bms JbdBt(A4:C1:38:E6:3F:18) 13:06:35 ERROR [sampling] BANK3 error: Device with address A4:C1:38:E6:3F:18 was not found. 13:06:35 ERROR [main] 2 exceptions occurred fetching BMSs 13:06:35 ERROR [main] Error (num 4, max 200) reading BMS: Device with address A4:C1:38:2A:95:DB was not found. 13:06:35 ERROR [main] Stack: Traceback (most recent call last): File "/app/main.py", line 45, in fetch_loop await fn() File "/app/main.py", line 239, in fn raise exceptions[0] File "/app/main.py", line 234, in fn await t File "/app/bmslib/sampling.py", line 51, in **call** return await self.sample() File "/app/bmslib/sampling.py", line 70, in sample async with bms: File "/app/bmslib/bt.py", line 172, in **aenter** await self.connect() File "/app/bmslib/jbd.py", line 51, in connect await super().connect(**kwargs) File "/app/bmslib/bt.py", line 77, in connect await self._connect_client(timeout=timeout) File "/app/bmslib/bt.py", line 48, in _connect_client await self.client.connect(timeout=timeout) File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect raise BleakError( bleak.exc.BleakError: Device with address A4:C1:38:2A:95:DB was not found. 13:06:50 INFO [sampling] connecting bms JbdBt(A4:C1:38:2A:95:DB) 13:07:10 ERROR [sampling] BANK1 error: Device with address A4:C1:38:2A:95:DB was not found. 13:07:10 INFO [sampling] connecting bms JbdBt(A4:C1:38:A2:EA:D8) 13:07:11 INFO [sampling] connected bms JbdBt(A4:C1:38:A2:EA:D8)! 13:07:20 ERROR [sampling] BANK2 error: <class 'asyncio.exceptions.TimeoutError'> 13:07:20 INFO [sampling] bytearray(b'\xdd\x04\x00 \r'\r$\r%\r%\r'\r"\r#\r)\r'\r'\r#\r$\r'\r&\r'\r%\xfc\xb8w') bms debug data: BANK2 13:07:21 INFO [sampling] connecting bms JbdBt(A4:C1:38:E6:3F:18) 13:07:41 ERROR [sampling] BANK3 error: Device with address A4:C1:38:E6:3F:18 was not found. 13:07:41 ERROR [main] 3 exceptions occurred fetching BMSs 13:07:41 ERROR [main] Error (num 5, max 200) reading BMS: Device with address A4:C1:38:2A:95:DB was not found. 13:07:41 ERROR [main] Stack: Traceback (most recent call last): File "/app/main.py", line 45, in fetch_loop await fn() File "/app/main.py", line 239, in fn raise exceptions[0] File "/app/main.py", line 234, in fn await t File "/app/bmslib/sampling.py", line 51, in **call** return await self.sample() File "/app/bmslib/sampling.py", line 70, in sample async with bms: File "/app/bmslib/bt.py", line 172, in **aenter** await self.connect() File "/app/bmslib/jbd.py", line 51, in connect await super().connect(**kwargs) File "/app/bmslib/bt.py", line 77, in connect await self._connect_client(timeout=timeout) File "/app/bmslib/bt.py", line 48, in _connect_client await self.client.connect(timeout=timeout) File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect raise BleakError( bleak.exc.BleakError: Device with address A4:C1:38:2A:95:DB was not found.

This appear to be another issue. Also I don't believe this is due to recent updates. Did you try the things listed in the README / Troubleshooting?

fl4p, 2023-04-19:
@albertomontoro2023  The interesting part from your logs is
```
raise BleakError(
bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the handle attribute instead.
```
I'll release an update that tries to refer the Characteristics by handle soon

Honusnap, 2023-04-21:
Hello, 
My jkbms wotk with Home assistant and Batmon for 6 month without a problem.
Something strange i got for 3 days, my jkbms bluetooth work for a moment then do not respond anymore and it seems to be connected with those updates i did on HAssistant (i upgraded Batmon, and Home assistant core).
I was planning on going back to 0.0.56.
I can't assure it's from Batmon, i neeed to restart the BMS (can't right now, too early not enough sun).

hddmax, 2023-04-21:
so i have checked all,  restart all HA, BMS, Router, nothing helps.
i think this is a problem of the bluetooth integration in OS 10.0

i have a smal solution, but it works not all the time.
give the bluetooth more time. 

[image]

Honusnap, 2023-04-21:
Ok i restarted the BMS, i can confirm that if i let Batmon connect to it .. it will connect once... then the bluetooth device is kind killed, i can't access it anymore till à restart the BMS.
If i stop Batmon service, i can access the BMS normally with my phone, bluetooth stay available.

So, the way the bluetooth device is handled in Batmon/HA, "kill" the device
And i'm at 150 sample_pertiod and 300 publish .. which is not agressive, concurrent_sampling and keep_alive disabled.

fl4p, 2023-04-21:
@Honusnap you can dowgrade to this version and see if it changes anything: https://github.com/fl4p/batmon-ha/releases/tag/release0.0.46

Honusnap, 2023-04-21:
> @Honusnap you can dowgrade to this version and see if it changes anything: https://github.com/fl4p/batmon-ha/releases/tag/release0.0.46

Any guideline to downgrade while keeping settings ? Or do i ned to uninstall and reinstall 0.0.46
Rem : I was not having a problem with 0.0.56.

fl4p, 2023-04-21:
Not sure if this keeps the settings. You can backup it using the YAML Editor.
* Remove the add-on store repository
* Download an old release from https://github.com/fl4p/batmon-ha/tags
* Extract the zip file into Home Assistant's `/addons` folder (using FTP, Samba Share, etc.), so that there is a folder `/addons/batmon-ha`
* The add-on will appear in the add-on store under "Local Addons"

You can also try this version, which uses the latest bleak:
https://github.com/fl4p/batmon-ha/releases/tag/v0.0.60-bleak-0.20.2

albertomontoro2023, 2023-04-21:
Just try with the NEW version 0.0.60
Now I have the followind error:

19:53:08 ERROR [bt] 	[Characteristic] 0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 4): Vendor specific (read,notify), Value: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.
[...]
19:53:08 ERROR [sampling] JK-B1A20S15P error: Characteristic with UUID 17 could not be found!

This is the complete LOG:

s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
19:53:02 INFO [bt] BT Discovery:
19:53:07 INFO [bt] BT Device   16-12-2E-C9-F6-94   address=16:12:2E:C9:F6:94
19:53:07 INFO [bt] BT Device   59-99-99-60-2D-71   address=59:99:99:60:2D:71
19:53:07 INFO [bt] BT Device   JK-B1A20S15P   address=20:21:11:28:18:CE
19:53:07 INFO [bt] BT Device   Mi Smart Band 6   address=C6:77:22:7A:07:AE
19:53:07 INFO [bt] BT Device   62-97-99-CC-7C-CE   address=62:97:99:CC:7C:CE
19:53:07 INFO [bt] BT Device   77-65-77-12-E6-D4   address=77:65:77:12:E6:D4
19:53:07 INFO [main] Verbose logging enabled
19:53:07 INFO [bt] BMS JKBt(20:21:11:28:18:CE) keep alive enabled
19:53:07 INFO [main] connecting mqtt mqttuser@core-mosquitto
19:53:07 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
19:53:07 DEBUG [client] Received CONNACK (0, 0)
19:53:07 INFO [main] Fetching 1 BMS + 0 others concurrently, period=20.00s, keep_alive=True
19:53:07 INFO [sampling] connecting bms JKBt(20:21:11:28:18:CE)
19:53:07 DEBUG [client] Connecting to device @ 20:21:11:28:18:CE with hci0
19:53:07 INFO [main] mqtt watchdog loop started with timeout 900.0s
19:53:07 DEBUG [scanner] cached devices: {'/org/bluez/hci0/dev_62_97_99_CC_7C_CE': {'Address': '62:97:99:CC:7C:CE', 'AddressType': 'random', 'Alias': '62-97-99-CC-7C-CE', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {76: b'\x10\x05\x01\x18\nr\xfb'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x1a'}, '/org/bluez/hci0/dev_C6_77_22_7A_07_AE': {'Address': 'C6:77:22:7A:07:AE', 'AddressType': 'public', 'Name': 'Mi Smart Band 6', 'Alias': 'Mi Smart Band 6', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fee0-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x03\xc6w"z\x07\xae'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}, '/org/bluez/hci0/dev_20_21_11_28_18_CE': {'Address': '20:21:11:28:18:CE', 'AddressType': 'public', 'Name': 'JK-B1A20S15P', 'Alias': 'JK-B1A20S15P', 'Paired': False, 'Bonded': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000ffe0-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {19274: b'\x00\x01'}, 'ServicesResolved': False, 'AdvertisingFlags': b'\x06'}}
19:53:07 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
19:53:07 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_16_12_2E_C9_F6_94', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 16:12:2E:C9:F6:94)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 16-12-2E-C9-F6-94)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Bonded': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -74)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {6: <dbus_next.signature.Variant ('ay', b'\x01\t \x02.\x18\x1d\xc2\x07=\xb27\x80\xc4\xa4\xe8g\x96\xdb`\xc3\x81~Z[\xe8\xc1')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_next.signature.Variant ('ay', b'\x00')>}, 'org.freedesktop.DBus.Properties': {}}]
19:53:07 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_59_99_99_60_2D_71', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 59:99:99:60:2D:71)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 59-99-99-60-2D-71)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Bonded': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -67)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant (' [... cut ...]
  File "/app/bmslib/sampling.py", line 92, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 215, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 48, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

I hope this can help for next version.
Thanks for all your work

Honusnap, 2023-04-24:
Hello Guys,
Yesterday i upgraded  Home Assistant, and Batmon, the problem is solved for me, it works again... so i can't tell if it was a ptoblem on Home assistant side or Batmon... JkBMS is polled again every 150s by Batmon and it do not "lock up" it's Bluetooth anymore.

Thanks for this amazing plugin, thanks to it i can follow my everyday in and out power of my battery.
[image]

albertomontoro2023, 2023-04-25:
Same as you Honusnap !!!!!

With HA 10 and the last Batmon version 0.0.61 all is working fine
Super happy.
Thanks for all.

@fl4p I will pay you a pair of beers with paypal.
Your work is excellent
Thaks for all.

closed 2023-04-25 by albertomontoro2023

hddmax, 2023-04-27:
this was the initial problem:
https://github.com/home-assistant/operating-system/issues/2485
Upgrade to HA10.1 !