fl4p/batmon-ha #86: [Errno 2] No such file or directory: '/data/options.json' in standalone mode

KidrashVr, 2023-04-22:
Hello, 
i'm trying to run it on standalone mode. As per instruction i have created the options.json in the same folder but when i lauch python3 main.py i get this error:
deunan@victron-lab:~/batmon-ha$ python3 main.py
error reading /data/options.json, trying options.json [Errno 2] No such file or directory: '/data/options.json'
Traceback (most recent call last):
  File "/home/deunan/batmon-ha/bmslib/store.py", line 51, in load_user_config
    with open('/data/options.json') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/data/options.json'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/deunan/batmon-ha/main.py", line 28, in <module>
    user_config = load_user_config()
  File "/home/deunan/batmon-ha/bmslib/store.py", line 57, in load_user_config
    conf = dotdict(json.load(f))
  File "/usr/lib/python3.10/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.10/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 17 column 3 (char 349)
deunan@victron-lab:~/batmon-ha$

I have also created a subfolder named data and copied the same same inside it, but the error don't change.
What i'm doing wrong?

pedro042, 2023-04-22:
To me happens it too (missing /data/options.json), but it continues and than works.
Are you sure your options.json is valid? Try jq tool to validate yours options.json

fl4p, 2023-04-22:
yep, looks like your json is not valid

KidrashVr, 2023-04-23:
Found the error. Since i copied and modified the example options.json, there is an error in the example.
"invert_current": false misses a comma at the end.
Now works very well, thank you for pointing me in the right direction