fl4p/batmon-ha #87: Victron SmartShunt connection issue

djismgaming, 2023-04-26:
Hi, I'm experiencing an issue connecting to the SmartShunt. Working before on an ARM SBC (La Frite from Libre.Computer). The same bluetooth adapter is being used as before. The only change so far is the host which now is a VM inside Proxmox.

The JKbms "bateria-casa" works as expected. Sends data to HAOS and all.
The SmartShunt just says ```le-connection-abort-by-local``` so far while trying to connect.

### Logs:
```
ismael@solarmonitor:~/batmon-ha$ python3 main.py
19:10:49 INFO [bt] BT Discovery:
19:10:54 INFO [bt] BT Device   bateria-aires   address=DA:88:D3:60:A0:50
19:10:54 INFO [bt] BT Device   bateria-casa   address=C8:47:8C:E8:4F:07
19:10:54 INFO [bt] BT Device   08-90-E5-CD-E9-90   address=08:90:E5:CD:E9:90
19:10:54 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.64
19:10:54 INFO [main] connecting mqtt admin@192.168.0.2
19:10:54 INFO [main] Fetching 2 BMS + 0 others serially, period=10.00s, keep_alive=False
19:10:54 INFO [sampling] connecting bms JKBt(C8:47:8C:E8:4F:07)
19:10:54 INFO [main] mqtt watchdog loop started with timeout 900.0s
19:11:00 INFO [sampling] connected bms JKBt(C8:47:8C:E8:4F:07)!
19:11:01 INFO [sampling] bateria-casa subscribing for {'charge': True, 'discharge': True} switch change
19:11:01 INFO [mqtt_util] subscribe homeassistant/switch/bateria-casa/charge/set
19:11:01 INFO [mqtt_util] subscribe homeassistant/switch/bateria-casa/discharge/set
19:11:01 INFO [sampling] bateria-casa: BmsSampl(99.6%,U=54.5V,I=-2.32A,P=-126W,q=358.6Ah/360,mos=35.6°C)
19:11:01 INFO [sampling] bateria-casa volt=3410,3394,3409,3405,3394,3406,3395,3405,3410,3404,3398,3408,3408,3412,3406,3412 temp=[31.9, 32.2]
19:11:04 INFO [sampling] JKBt(C8:47:8C:E8:4F:07) times: connect=6.28s fetch=1.09s
19:11:04 INFO [sampling] connecting bms SmartShuntBt(DA:88:D3:60:A0:50)
19:11:06 ERROR [sampling] bateria-aires error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:11:06 INFO [sampling] connecting bms SmartShuntBt(DA:88:D3:60:A0:50)
19:11:07 ERROR [sampling] bateria-aires error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:11:07 INFO [sampling] connecting bms JKBt(C8:47:8C:E8:4F:07)
19:11:15 INFO [sampling] connected bms JKBt(C8:47:8C:E8:4F:07)!
19:11:15 INFO [sampling] bateria-casa: BmsSampl(99.6%,U=54.5V,I=-2.51A,P=-137W,q=358.6Ah/360,mos=35.5°C)
19:11:15 INFO [sampling] bateria-casa volt=3406,3398,3399,3405,3388,3409,3394,3408,3409,3405,3395,3409,3410,3405,3410,3409 temp=[31.9, 32.1]
19:11:18 INFO [sampling] JKBt(C8:47:8C:E8:4F:07) times: connect=7.98s fetch=0.71s
19:11:18 ERROR [main] 1 exceptions occurred fetching BMSs
19:11:18 ERROR [main] Error (num 1, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
19:11:18 ERROR [main] Stack: Traceback (most recent call last):
  File "/home/ismael/batmon-ha/main.py", line 35, in fetch_loop
    await fn()
  File "/home/ismael/batmon-ha/main.py", line 265, in fn
    raise exceptions[0]
  File "/home/ismael/batmon-ha/main.py", line 260, in fn
    await t()
  File "/home/ismael/batmon-ha/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/home/ismael/batmon-ha/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 237, in __aenter__
    await self.connect()
  File "/home/ismael/batmon-ha/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 140, in connect
    await self._connect_client(timeout=timeout)
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 108, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/home/ismael/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/home/ismael/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

19:11:28 INFO [sampling] connecting bms SmartShuntBt(DA:88:D3:60:A0:50)
19:11:29 ERROR [sampling] bateria-aires error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:11:29 INFO [sampling] connecting bms JKBt(C8:47:8C:E8:4F:07)
19:11:31 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:11:44 INFO [sampling] connected bms JKBt(C8:47:8C:E8:4F:07)!
^C19:11:44 INFO [main] exit signal handler... (2, <frame at 0x7fb67ca1fa40, file '/usr/lib/python3.10/selectors.py', line 469, code select>), {}, shutdown already False
19:11:45 INFO [sampling] bateria-casa: BmsSampl(99.6%,U=54.5V,I=-2.32A,P=-126W,q=358.6Ah/360,mos=35.6°C)
19:11:45 INFO [sampling] bateria-casa volt=3408,3397,3403,3405,3388,3409,3393,3408,3405,3406,3394,3409,3410,3404,3409,3408 temp=[31.9, 32.1]
19:11:53 INFO [sampling] JKBt(C8:47:8C:E8:4F:07) times: connect=14.53s fetch=0.75s
19:11:53 ERROR [main] 1 exceptions occurred fetching BMSs
19:11:53 ERROR [main] Error (num 2, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
19:11:53 ERROR [main] Stack: Traceback (most recent call last):
  File "/home/ismael/batmon-ha/main.py", line 35, in fetch_loop
    await fn()
  File "/home/ismael/batmon-ha/main.py", line 265, in fn
    raise exceptions[0]
  File "/home/ismael/batmon-ha/main.py", line 260, in fn
    await t()
  File "/home/ismael/batmon-ha/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/home/ismael/batmon-ha/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 237, in __aenter__
    await self.connect()
  File "/home/ismael/batmon-ha/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 140, in connect
    await self._connect_client(timeout=timeout)
  File "/home/ismael/batmon-ha/bmslib/bt.py", line 108, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/home/ismael/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/home/ismael/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

19:12:03 INFO [main] fetch_loop <function main.<locals>.fn at 0x7fb67bef67a0> ends
19:12:03 INFO [main] All fetch loops ended. shutdown is already True
19:12:03 INFO [main] Disconnecting JKBt(C8:47:8C:E8:4F:07)
19:12:03 INFO [main] Disconnecting SmartShuntBt(DA:88:D3:60:A0:50)
19:12:03 INFO [main] exit signal handler... (), {}, shutdown already True
ismael@solarmonitor:~/batmon-ha$
```

### lsusb:
```
ismael@solarmonitor:~/batmon-ha$ lsusb
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) ***USB BLUETOOTH***
Bus 002 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial ***LV5048***
Bus 002 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial ***LV6048***
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
ismael@solarmonitor:~/batmon-ha$
```

### options.json:
```
ismael@solarmonitor:~/batmon-ha$ cat /data/options.json
{
  "devices": [
    {
      "address": "C8:47:8C:E8:4F:07",
      "type": "jk",
      "alias": "bateria-casa"
    },
    {
      "address": "DA:88:D3:60:A0:50",
      "type": "victron",
      "alias": "bateria-aires",
      "pin": "000000"
    }
],
  "mqtt_broker": "hass-mqtt",
  "mqtt_user": "user",
  "mqtt_password": "pass",
  "concurrent_sampling": false,
  "keep_alive": false,
  "verbose_log": false,
  "sample_period": 10.0,
  "expire_values_after": 120,
  "install_newer_bleak": false,
  "invert_current": true,
  "watchdog": true
}
ismael@solarmonitor:~/batmon-ha$
```

damianarnold1993, 2023-04-27:
same here with a Liontron 100Ah also with the JK BMS. Seems working until yesterday, no changes on the Hardware

djismgaming, 2023-04-28:
I did change the hardware again from the VM to an Acer E15 so it would not end up in e-waste.

Made a docker container this time and all is working with the included Bluetooth on the PC.

No issues here now :)

@damianarnold1993 I would suggest opening a new issue with your logs so the developer and community can look at your issue ;)

closed 2023-04-28 by djismgaming