fl4p/batmon-ha #88: JKB2A8S20P  asyncio.exceptions.TimeoutError

restaun, 2023-04-29:
```
Hello. My BMS is JKB2A8S20P and i keep getting this error.

21:43:57 ERROR [sampling] JKBMSbatery error: <class 'asyncio.exceptions.TimeoutError'>
21:43:57 INFO [sampling] JKBMSbatery bms debug data: {3: bytearray(b'U\xaa\xeb\x90\x03\xbfJK_B2A8S20P\x00\x00\x00\x00\x0011.XG\x00\x00\x0011.14\x00\x00\x00\xf08N\x01\x12\x01\x00\x00JK_B2A8S20P\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00220818\x00\x002052317001\x000000\x00Input Userdata\x00\x00reg64297584\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xf3\x0f\x00\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04')}
21:43:57 ERROR [main] 1 exceptions occurred fetching BMSs
21:43:57 ERROR [main] Error (num 10, max 0) reading BMS: 
21:43:57 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 92, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 222, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 48, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

I have been receiving data from it, but updating the home assistant has no longer worked well.
All the best.
```

bjoernh, 2023-04-30:
Same Problem after replacing a faulty JK BMS with a new one. Looks like the have changed the protocol a little bit. When batman try to connect it only beeps one times.

fl4p, 2023-04-30:
Can you capture Bluetooth data sample when using the app? See https://github.com/fl4p/batmon-ha/blob/master/doc/BT%20Sniffing.md

restaun, 2023-04-30:
Hello, I have the application on android. Is there a program for them?

restaun, 2023-04-30:
[image]
Is this what you asked for?

closed 2025-02-02 by fl4p