fl4p/batmon-ha #89: MQTT connection Problem

schris91, 2023-04-30:
i am trying stand alone version to issue a connection via mqtt to iobroker. When i type in my MQTT Server adress and port (10.0.0.46:1884) i got a connection error.

fl4p, 2023-05-08:
Set the port with the option `mqtt_port`, and remove `:1884` from `mqtt_broker`

schris91, 2023-05-12:
there is no "mqtt_port". pls help :o

],
  "mqtt_broker": "homeassistant.local",
  "mqtt_user": "pv",
  "mqtt_password": "Offgrid",
  "concurrent_sampling": false,
  "keep_alive": true,
  "sample_period": 1.0,
  "publish_period": 1.0,
  "invert_current": false,
  "expire_values_after": 20,
  "verbose_log": false,
  "watchdog": false
}

fl4p, 2023-05-15:
Just add it to the json

closed 2023-07-30 by fl4p