fl4p/batmon-ha #90: No device fount - BT discovery

matata86, 2023-05-01:
```
12:27:31 INFO [bt] BT Discovery:
12:27:36 INFO [bt]  - no devices found - 
12:27:36 INFO [bt] BMS JbdBt(A4:C1:37:10:7B:E8) keep alive enabled
12:27:36 INFO [main] connecting mqtt matata@core-mosquitto
12:27:36 INFO [main] Fetching 1 BMS + 0 others concurrently, period=30.00s, keep_alive=True
12:27:36 INFO [sampling] connecting bms JbdBt(A4:C1:37:10:7B:E8)
12:27:36 INFO [main] mqtt watchdog loop started with timeout 900.0s
12:27:56 ERROR [sampling] battery1 error: Device with address A4:C1:37:10:7B:E8 was not found.
12:27:56 INFO [sampling] connecting bms JbdBt(A4:C1:37:10:7B:E8)```

Integration does not see any BT devices at all. I also have a Mi Band 6 and others, but it doesn't see any.

Bluetooth in Home Assistant is working fine. I use many BT thermometers and similar devices.

rdstevens, 2023-05-03:
Have you configured a device entry? Where did you get the device from?

the XiaoxiangBMS app on my iPhone shows the address bytes in reverse order, so if the iOS app says aa:bb:cc:dd:ee:ff, the address should be configured as ff:ee:dd:cc:bb:aa.

patman15, 2023-05-10:
I have the same issue very regularly. 
`~ bluetoothctl devices` does show all devices, the addon at the (almost) same time, does not. I think the whole setup is not very robust against unstable connections as it does not use https://bleak.readthedocs.io/en/latest/api/scanner.html#starting-and-stopping but only scans once and then assumes the BMS is available ... ?

fl4p, 2023-07-30:
The scan uses bleak, there are some issues with bleak. If the device does not appear in the scan, batmon will not be able to connect.
Try to enable `install_newer_bleak` and `bt_power_cycle`

fl4p, 2023-08-20:
closed due to inactivity. 
empty scan results is usually not a issue with batmon but with the OS, permission, container etc

closed 2023-08-20 by fl4p