fl4p/batmon-ha #95: Discharge Switch/Bluetooth connection

GRV0423, 2023-05-03:
Describe the issue here.
the discharge switch is not turning of my bms discharged output, charge switch is working though.
Bluetooth connection is taking longer to connect, and some time connection drop is more often after the latest update. 
Paste log output here.
```

fl4p, 2023-05-03:
Which BMS is it?

GRV0423, 2023-05-03:
Sorry I didn't mention which bms...
It a Daly 150A bms.

GRV0423, 2023-05-04:
Would be possible to put setting for SOC % for or calibrating...

closed 2023-05-09 by GRV0423

fl4p, 2023-05-09:
> Sorry I didn't mention which bms... It a Daly 150A bms.

I'll test this later, need to setup my Daly BMS.

> Would be possible to put setting for SOC % for calibrating...

Not sure if I understand you correctly. Usually you just need to charge to 100%, then completely discharge and if the SoC% is still above zero, decrease the capacity in the BMS settings by that percentage. 
For example: Capacity set to  280 Ah. SoC = 5% at complete discharge. Adjust capacity to 280 * (1-5%) = 266 Ah