fl4p/batmon-ha #96: Connection issue JK B1A24S15P

pooooom2, 2023-05-04:
I have issue about connection.
After use with keep_alive function 1day or half day batmon cannot connect to BMS.
Then I restart batmon, LED indicator of BMS like BMS is connected to batmon but from log it find a BMS bluetooth (that it cannot find BMS) then i disconnect bluetooth dongle from HA for 5 sec and plug in again sometime it work.

```
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 123, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 133, in _connect_with_scanner
    raise Exception('Device %s not discovered. Make sure it in range and is not being controled by '
Exception: Device C8:47:8C:E9:0B:B0 not discovered. Make sure it in range and is not being controled by another application. ({'43:28:FA:48:6A:3E', '04:B9:E3:4C:44:93', '6C:5B:4F:82:8B:97', '64:74:2E:02:52:88', 'C6:96:36:50:7E:2F', 'E3:1B:E4:F3:B8:2C', '5B:2B:D1:40:05:7F', 'C4:CD:82:08:14:12', '1B:1D:2C:85:59:08', '02:34:D2:A1:C2:7C', 'A4:C1:38:FC:19:C8'})
09:47:55 INFO [sampling] connecting bms JKBt(C8:47:8C:E9:0B:B0)
09:47:55 INFO [jikong] normal connect failed (Device with address C8:47:8C:E9:0B:B0 was not found. It may have been removed from BlueZ when scanning stopped.), connecting with scanner
09:48:05 ERROR [sampling] battery1 error: [org.bluez.Error.Failed] No discovery started
09:48:05 ERROR [main] 1 exceptions occurred fetching BMSs
09:48:05 ERROR [main] Error (num 13, max 200) reading BMS: [org.bluez.Error.Failed] No discovery started
09:48:05 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/bmslib/jikong.py", line 120, in connect
    await super().connect(timeout=4)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 81, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 239, in connect
    raise BleakDeviceNotFoundError(
bleak.exc.BleakDeviceNotFoundError: Device with address C8:47:8C:E9:0B:B0 was not found. It may have been removed from BlueZ when scanning stopped.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/bmslib/bt.py", line 133, in _connect_with_scanner
    raise Exception('Device %s not discovered. Make sure it in range and is not being controled by '
Exception: Device C8:47:8C:E9:0B:B0 not discovered. Make sure it in range and is not being controled by another application. ({'ED:C6:5D:AC:69:81'})
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/jikong.py", line 123, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 146, in _connect_with_scanner
    await scanner.stop()
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 186, in stop
    await self._backend.stop()
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/scanner.py", line 203, in stop
    await stop()
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 390, in stop
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] No discovery started
09:48:06 INFO [sampling] connecting bms JKBt(C8:47:8C:E9:0B:B0)
09:48:06 INFO [jikong] normal connect failed (Device with address C8:47:8C:E9:0B:B0 was not found. It may have been removed from BlueZ when scanning stopped.), connecting with scanner
09:48:20 INFO [sampling] connected bms JKBt(C8:47:8C:E9:0B:B0)!
09:48:20 INFO [sampling] battery1 subscribing for {'charge': True, 'discharge': True} switch change
09:48:20 INFO [mqtt_util] subscribe homeassistant/switch/battery1/charge/set
09:48:20 INFO [mqtt_util] subscribe homeassistant/switch/battery1/discharge/set
09:48:20 INFO [sampling] battery1: BmsSampl(60.1%,U=53.5V,I=-44.44A,P=-2376W,q=140.6Ah/234,mos=0.0°C)
09:48:20 INFO [sampling] battery1 volt=3807,3817,3808,3817,3819,3818,3823,3827,3827,3817,3832,3828,3814,3803 temp=[33.8, 34.0]
09:48:20 INFO [sampling] JKBt(C8:47:8C:E9:0B:B0) times: connect=14.27s fetch=0.55s
09:48:21 INFO [sampling] battery1: BmsSampl(60.1%,U=53.5V,I=-44.44A,P=-2376W,q=140.6Ah/234,mos=0.0°C)
09:48:21 INFO [sampling] battery1 volt=3806,3818,3808,3818,3821,3819,3823,3828,3825,3817,3833,3828,3814,3803 temp=[33.8, 34.0]
09:48:23 INFO [sampling] battery1: BmsSampl(60.1%,U=53.5V,I=-44.44A,P=-2376W,q=140.6Ah/234,mos=0.0°C)
09:48:23 INFO [sampling] battery1 volt=3806,3818,3808,3818,3821,3819,3823,3828,3825,3817,3833,3827,3814,3803 temp=[33.8, 34.0]
09:48:24 INFO [sampling] battery1: BmsSampl(60.1%,U=53.5V,I=-44.66A,P=-2388W,q=140.6Ah/234,mos=0.0°C)
09:48:24 INFO [sampling] battery1 volt=3807,3818,3807,3817,3821,3819,3827,3828,3823,3819,3830,3827,3814,3803 temp=[33.8, 34.0]
```

pooooom2, 2023-05-04:
Log when BMS disconnect

``````
19:36:27 INFO [sampling] battery1: BmsSampl(93.8%,U=56.0V,I=16.04A,P=899W,q=219.6Ah/234,mos=0.0°C)
19:36:27 INFO [sampling] battery1 volt=4008,4006,4004,4007,4007,4006,4002,3996,4001,4002,3999,3999,4001,4006 temp=[32.5, 32.5]
19:36:28 INFO [sampling] battery1: BmsSampl(93.8%,U=56.0V,I=16.04A,P=899W,q=219.6Ah/234,mos=0.0°C)
19:36:28 INFO [sampling] battery1 volt=4008,4003,4002,4003,4003,4004,4002,3999,4003,4008,3997,3998,4004,4006 temp=[32.5, 32.5]
19:36:30 INFO [sampling] battery1: BmsSampl(93.8%,U=56.0V,I=16.04A,P=899W,q=219.6Ah/234,mos=0.0°C)
19:36:30 INFO [sampling] battery1 volt=4008,4003,4002,4003,4003,4004,4002,3999,4003,4008,3997,3998,4004,4006 temp=[32.5, 32.5]
19:36:36 WARNING [bt] BMS JKBt(C8:47:8C:E9:0B:B0) disconnected after 35297.2s!
19:36:36 INFO [sampling] battery1 bms debug data: {3: bytearray(b'U\xaa\xeb\x90\x03\\JK_B1A24S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.25\x00\x00\x00\\\xe9n\x00\x0c\x00\x00\x00JK_B1A24S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230209\x00\x002092401204\x000000\x00Input Userdata\x00\x000850753323\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x8a'), 1: bytearray(b'U\xaa\xeb\x90\x01\\X\x02\x00\x00H\r\x00\x00\xac\r\x00\x00\x04\x10\x00\x00\xef\x0f\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb8\x0b\x00\x00\xf0I\x02\x00\x1e\x00\x00\x00<\x00\x00\x00\xf0I\x02\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xe8\x03\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\xbc\x02\x00\x00X\x02\x00\x008\xff\xff\xff\x9c\xff\xff\xff\xe8\x03\x00\x00 \x03\x00\x00\x0e\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x10\x92\x03\x00\xdc\x05\x00\x00\xb8\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x00\x00\x00\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\xaf'), 2: bytearray(b'U\xaa\xeb\x90\x02\x9e\xa8\x0f\xa3\x0f\xa2\x0f\xa3\x0f\xa3\x0f\xa4\x0f\xa2\x0f\x9f\x0f\xa3\x0f\xa8\x0f\x9d\x0f\x9e\x0f\xa4\x0f\xa6\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff?\x00\x00\xa3\x0f\x0b\x00\x00\n\x1e\x00\x1f\x00 \x00"\x00#\x00%\x00&\x00(\x00)\x00)\x00,\x00.\x00/\x00/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x89\x01\x00\x00\x00\x00\xeb\xda\x00\x00\x1e\xb8\r\x00U\xc1\xff\xffE\x01E\x01\x00\x00\x00\x00@\xfc\x02]\xb8Y\x03\x00\x10\x92\x03\x00\x1d\x00\x00\x00\xc0-i\x00d\x00\x00\x00\xebso\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00<\x04\x00\x00J\x00\x19\x8e@@\xd0\x00\x00\x00\xe4\x15\xa7\x1e\x00\x01\x00\x01\x00\x05\x00\x00\xf80y\x00\x00\x00\x00\x00\x89\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00\x9d')}
Traceback (most recent call last):
  File "/app/main.py", line 266, in main
    await fetch_loop(fn, period=sample_period, max_errors=max_errors)
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 92, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 222, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 48, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 296, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError
19:36:36 INFO [main] exit signal handler... (), {}, shutdown already False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
``````

fl4p, 2025-11-14:
The connection code has changed, please try with latest version and open a new issue if needed

closed 2025-11-14 by fl4p