fl4p/batmon-ha #97: Jk BMS disconnect a half minute.

tjanicsek, 2023-05-04:
Batmon starts fine, but a half minute later stops. 

```
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m42), 'b'battery1/cell_voltages/median'', ... (5 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m43), 'b'battery1/temperatures/1'', ... (4 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m44), 'b'battery1/temperatures/2'', ... (4 bytes)
22:01:31 INFO [sampling] battery1 volt=3876,3876,3874,3876,3876,3874,3874,3877,3877,3876,3877,3876,3876,3874 temp=[20.4, 20.1]
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m45), 'b'homeassistant/sensor/battery1/_soc_total_voltage/config'', ... (460 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m46), 'b'homeassistant/sensor/battery1/_soc_current/config'', ... (412 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m47), 'b'homeassistant/sensor/battery1/_soc_balance_current/config'', ... (476 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m48), 'b'homeassistant/sensor/battery1/_soc_soc_percent/config'', ... (413 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m49), 'b'homeassistant/sensor/battery1/_soc_power/config'', ... (423 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m50), 'b'homeassistant/sensor/battery1/_soc_capacity/config'', ... (360 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m51), 'b'homeassistant/sensor/battery1/_soc_cycle_capacity/config'', ... (384 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m52), 'b'homeassistant/sensor/battery1/_soc_num_cycles/config'', ... (425 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m53), 'b'homeassistant/sensor/battery1/_mosfet_status_capacity_ah/config'', ... (397 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m54), 'b'homeassistant/sensor/battery1/_mosfet_status_temperature/config'', ... (499 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m55), 'b'homeassistant/sensor/battery1/_bms_uptime/config'', ... (445 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m56), 'b'homeassistant/sensor/battery1/_cell_voltages_1/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m57), 'b'homeassistant/sensor/battery1/_cell_voltages_2/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m58), 'b'homeassistant/sensor/battery1/_cell_voltages_3/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m59), 'b'homeassistant/sensor/battery1/_cell_voltages_4/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m60), 'b'homeassistant/sensor/battery1/_cell_voltages_5/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m61), 'b'homeassistant/sensor/battery1/_cell_voltages_6/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m62), 'b'homeassistant/sensor/battery1/_cell_voltages_7/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m63), 'b'homeassistant/sensor/battery1/_cell_voltages_8/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m64), 'b'homeassistant/sensor/battery1/_cell_voltages_9/config'', ... (402 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m65), 'b'homeassistant/sensor/battery1/_cell_voltages_10/config'', ... (406 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m66), 'b'homeassistant/sensor/battery1/_cell_voltages_11/config'', ... (406 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m67), 'b'homeassistant/sensor/battery1/_cell_voltages_12/config'', ... (406 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m68), 'b'homeassistant/sensor/battery1/_cell_voltages_13/config'', ... (406 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m69), 'b'homeassistant/sensor/battery1/_cell_voltages_14/config'', ... (406 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m70), 'b'homeassistant/sensor/battery1/_cell_voltages_min/config'', ... (410 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m71), 'b'homeassistant/sensor/battery1/_cell_voltages_max/config'', ... (410 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m72), 'b'homeassistant/sensor/battery1/_cell_voltages_average/config'', ... (426 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m73), 'b'homeassistant/sensor/battery1/_cell_voltages_median/config'', ... (422 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m74), 'b'homeassistant/sensor/battery1/_cell_voltages_delta/config'', ... (418 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m75), 'b'homeassistant/sensor/battery1/_cell_voltages_min_index/config'', ... (379 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m76), 'b'homeassistant/sensor/battery1/_cell_voltages_max_index/config'', ... (379 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m77), 'b'homeassistant/sensor/battery1/_temperatures_1/config'', ... (408 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m78), 'b'homeassistant/sensor/battery1/_temperatures_2/config'', ... (408 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m79), 'b'homeassistant/sensor/battery1/_meter_total_energy/config'', ... (445 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m80), 'b'homeassistant/sensor/battery1/_meter_total_energy_charge/config'', ... (508 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m81), 'b'homeassistant/sensor/battery1/_meter_total_energy_discharge/config'', ... (520 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m82), 'b'homeassistant/sensor/battery1/_meter_total_charge/config'', ... (388 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m83), 'b'homeassistant/sensor/battery1/_meter_total_cycles/config'', ... (415 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m84), 'b'homeassistant/switch/battery1/charge/config'', ... (412 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m85), 'b'homeassistant/binary_sensor/battery1/charge/config'', ... (418 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m86), 'b'homeassistant/switch/battery1/discharge/config'', ... (427 bytes)
22:01:31 DEBUG [client] Sending PUBLISH (d0, q0, r0, m87), 'b'homeassistant/binary_sensor/battery1/discharge/config'', ... (433 bytes)
22:01:31 INFO [sampling] JKBt(C8:47:8C:EA:D7:78) times: connect=6.44s fetch=0.82s
22:01:31 INFO [sampling] connecting bms JKBt(C8:47:8C:E8:37:69)
22:01:31 DEBUG [client] Connecting to device @ C8:47:8C:E8:37:69
22:01:31 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
22:01:31 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E8_37_69): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -76)>}, []]
22:01:31 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E8_37_69): ['org.bluez.Device1', {}, ['RSSI']]
22:01:31 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
22:01:31 DEBUG [client] Connecting to BlueZ path /org/bluez/hci0/dev_C8_47_8C_E8_37_69
22:01:31 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_D7_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'U\xaa\xeb\x90\x02J%\x0f%\x0f%\x0f%\x0f%\x0f$\x0f$\x0f$\x0f$\x0f"\x0f$\x0f!\x0f$\x0f$\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff?\x00\x00$\x0f\x04\x00\x08\x04\xb4\x00\xbe\x00\xc3\x00\xb7\x00\xb6\x00\xc2\x00\xbb\x00\xad\x00\xa9\x00\xb2\x00\xbe\x00\xb8\x00\xb6\x00\xc3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
22:01:31 DEBUG [jikong] header, clear buf bytearray(b'')
22:01:31 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 4a 25 f 25 f 25 f 25 f 25 f 24 f 24 f 24 f 24 f 22 f 24 f 21 f 24 f 24 f 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 3f 0 0 24 f 4 0 8 4 b4 0 be 0 c3 0 b7 0 b6 0 c2 0  [... cut ...] }]
22:01:33 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EA_D7_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00\xb3'))>}, []]
22:01:33 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc f 1 0 0 0 0 0 0 b3
22:01:33 DEBUG [jikong] got response 2 (len300)
```

lucize, 2023-05-05:
seems that I have the same issue

```
1:10:49 INFO [sampling] JKBt(C8:47:xx:xx:xx:xx) times: connect=0.00s fetch=0.72s
21:10:57 WARNING [bt] BMS JKBt(C8:47:xx:xx:xx:xx) disconnected after 22.7s!
21:10:57 INFO [sampling] battery280 bms debug data: {3: bytearray(b'U\xaa\xeb\x90\x03\x12JK_B2A20S20P\x00\x00\x00\x0011.XW\x00\x00\x0011.26H\x00\x00\xdc\xfb\t\x00\x02\x00\x00\x00JK_B2A20S20P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230426\x00\x003010545094\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x9f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x007'), 1: bytearray(b'U\xaa\xeb\x90\x01\x12X\x02\x00\x00(\n\x00\x00Z\n\x00\x00\x10\x0e\x00\x00\xde\r\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00P\xc3\x00\x00\x1e\x00\x00\x00<\x00\x00\x00@\r\x03\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xd0\x07\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\xbc\x02\x00\x00X\x02\x00\x008\xff\xff\xff\x9c\xff\xff\xff\xe8\x03\x00\x00 \x03\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\xc0E\x04\x00\xdc\x05\x00\x00\xb8\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x00\x00\x00\x00\x00|\xf8\xff\xff\x9f\r\x00\x00\x00\x00\x00\x00\xab'), 2: bytearray(b'U\xaa\xeb\x90\x02&\xcc\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcb\x0c\xcc\x0c\xcb\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xcb\x0c\x01\x00\x00\x023\x003\x002\x003\x002\x003\x002\x002\x002\x002\x002\x002\x002\x002\x00?\x00=\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe2\x00\x00\x00\x00\x00\xb4\xcc\x00\x00\xc30\x03\x00j\xf0\xff\xff\xc9\x00\xc8\x00\x00\x00\x08\x00\x00\x00\x00,\xb5\xe1\x01\x00\xc0E\x04\x00\x00\x00\x00\x00\xe9\xd6\x03\x00d\x00\x00\x00\xe8\xfb\t\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x1a\x04\x00\x00\x13\x005b=@\x00\x00\x00\x00x\x14\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00|\xffA\x00\x00\x00\x00\x00\xe2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc/\x01\x01\x01\x00\x00\x00\x00Y')}Traceback (most recent call last):
  File "/app/main.py", line 266, in main
    await fetch_loop(fn, period=sample_period, max_errors=max_errors)
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 92, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jikong.py", line 222, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 48, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 296, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError
21:10:57 INFO [main] exit signal handler... (), {}, shutdown already False

```

hddmax, 2023-05-07:
like the the same as you since the 2023.5.1 update

23:24:15 ERROR [main] Error (num 107, max 200) reading BMS: Device with address A4:C1:38:E6:3F:18 was not found. It may have been removed from BlueZ when scanning stopped.
23:24:15 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 81, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 239, in connect
    raise BleakDeviceNotFoundError(
bleak.exc.BleakDeviceNotFoundError: Device with address A4:C1:38:E6:3F:18 was not found. It may have been removed from BlueZ when scanning stopped.
23:25:15 INFO [sampling] connecting bms JbdBt(A4:C1:38:A2:EA:D8)
23:25:15 ERROR [sampling] BANK2 error: Device with address A4:C1:38:A2:EA:D8 was not found. It may have been removed from BlueZ when scanning stopped.
23:25:15 INFO [sampling] BANK2 bms debug data: bytearray(b'\xdd\x04\x00 \x0c\xf3\x0c\xf1\x0c\xf1\x0c\xf2\x0c\xef\x0c\xf3\x0c\xf6\x0c\xf5\x0c\xf5\x0c\xf7\x0c\xf1\x0c\xf3\x0c\xef\x0c\xf2\x0c\xf5\x0c\xf4\xef\xf2w')
23:25:15 INFO [sampling] connecting bms JbdBt(A4:C1:38:2A:95:DB)
23:25:35 ERROR [sampling] BANK1 error: Device with address A4:C1:38:2A:95:DB was not found.
23:25:35 INFO [sampling] connecting bms JbdBt(A4:C1:38:E6:3F:18)
23:25:36 INFO [sampling] connected bms JbdBt(A4:C1:38:E6:3F:18)!
23:25:36 INFO [sampling] BANK3: BmsSampl(94.8%,U=53.0V,I=-4.89A,P=-259W,q=293.7Ah/310,mos=nan°C)
23:25:37 INFO [sampling] BANK3 volt=3319,3315,3315,3316,3315,3312,3317,3315,3312,3314,3313,3319,3314,3311,3312,3310 temp=[19.7]
23:25:39 INFO [sampling] JbdBt(A4:C1:38:E6:3F:18) times: connect=1.69s fetch=0.33s
23:25:39 ERROR [main] 2 exceptions occurred fetching BMSs
23:25:39 ERROR [main] Error (num 108, max 200) reading BMS: Device with address A4:C1:38:A2:EA:D8 was not found. It may have been removed from BlueZ when scanning stopped.
23:25:39 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 81, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 239, in connect
    raise BleakDeviceNotFoundError(
bleak.exc.BleakDeviceNotFoundError: Device with address A4:C1:38:A2:EA:D8 was not found. It may have been removed from BlueZ when scanning stopped.
23:26:39 INFO [sampling] connecting bms JbdBt(A4:C1:38:E6:3F:18)
23:26:39 ERROR [sampling] BANK3 error: Device with address A4:C1:38:E6:3F:18 was not found. It may have been removed from BlueZ when scanning stopped.
23:26:39 INFO [sampling] BANK3 bms debug data: bytearray(b'\xdd\x04\x00 \x0c\xf7\x0c\xf3\x0c\xf3\x0c\xf4\x0c\xf3\x0c\xf0\x0c\xf5\x0c\xf3\x0c\xf0\x0c\xf2\x0c\xf1\x0c\xf7\x0c\xf2\x0c\xef\x0c\xf0\x0c\xee\xef\xfbw')
23:26:39 INFO [sampling] connecting bms JbdBt(A4:C1:38:2A:95:DB)
23:26:59 ERROR [sampling] BANK1 error: Device with address A4:C1:38:2A:95:DB was not found.
23:26:59 INFO [sampling] connecting bms JbdBt(A4:C1:38:A2:EA:D8)
23:27:00 INFO [sampling] connected bms JbdBt(A4:C1:38:A2:EA:D8)!
[... 4 lines cut ...]
23:27:03 ERROR [main] Error (num 109, max 200) reading BMS: Device with address A4:C1:38:E6:3F:18 was not found. It may have been removed from BlueZ when scanning stopped.
23:27:03 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 81, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 239, in connect
    raise BleakDeviceNotFoundError(
bleak.exc.BleakDeviceNotFoundError: Device with address A4:C1:38:E6:3F:18 was not found. It may have been removed from BlueZ when scanning stopped.