fl4p/batmon-ha #99: Victron Smart shunt

GRV0423, 2023-05-08:
Describe the issue here.
i have 3 daly bms and 1 victron smart shunt, 3 bms will connect but the victron  smart shunt will not, i think because victron have pin? how to disable the pin in victron?

Paste log output here.
06:09:30 WARNING [bt] this bleak version has no pairing agent, pairing with a pin will likely fail!
06:09:30 INFO [bt] BMS SmartShuntBt(C1:E5:9D:F6:EF:75) keep alive enabled
06:09:30 INFO [bt] BMS DalyBt(C7:7C:03:02:0D:E9) keep alive enabled
06:09:30 INFO [bt] BMS DalyBt(B7:7B:02:06:06:6D) keep alive enabled
06:09:30 INFO [bt] BMS DalyBt(B7:7B:02:02:09:65) keep alive enabled

fl4p, 2023-05-08:
Disable option `install_newer_bleak` and try again

GRV0423, 2023-05-08:
how do i disable option?
i think bleak is up to date

- address: C7:7C:03:02:0D:E9
  type: daly
  alias: batteryi
- address: B7:7B:02:06:06:6D
  type: daly
  alias: batteryii
- address: B7:7B:02:02:09:65
  type: daly
  alias: batteryiii
- address: C1:E5:9D:F6:EF:75
  type: victron
  alias: victron ss
  pin: "042388"
 
OS Generic x86-64

thank you

'07:54:43 INFO [sampling] connecting bms SmartShuntBt(C1:E5:9D:F6:EF:75)
07:54:44 ERROR [sampling] victron ss error: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 ERROR [main] Error (num 27, max 200) reading BMS: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 99, in _connect_client
    res = await self.client.pair(callback=get_passkey)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 555, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 466, in pair
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 WARNING [bt] BMS SmartShuntBt(C1:E5:9D:F6:EF:75) disconnected after 0.0s!'

GRV0423, 2023-05-08:
how do i disable option?
i think bleak is up to date

- address: C7:7C:03:02:0D:E9
  type: daly
  alias: batteryi
- address: B7:7B:02:06:06:6D
  type: daly
  alias: batteryii
- address: B7:7B:02:02:09:65
  type: daly
  alias: batteryiii
- address: C1:E5:9D:F6:EF:75
  type: victron
  alias: victron ss
  pin: "042388"
 
OS Generic x86-64

thank you

07:54:43 INFO [sampling] connecting bms SmartShuntBt(C1:E5:9D:F6:EF:75)
07:54:44 ERROR [sampling] victron ss error: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 ERROR [main] Error (num 27, max 200) reading BMS: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 86, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 207, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 113, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 99, in _connect_client
    res = await self.client.pair(callback=get_passkey)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 555, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 466, in pair
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed
07:54:44 WARNING [bt] BMS SmartShuntBt(C1:E5:9D:F6:EF:75) disconnected after 0.0s!

fl4p, 2023-05-08:
Try this version:
`pip3 install git+https://github.com/jpeters-ml/bleak@feature/windowsPairing`

Latest bleak still doesn't support pairing with a PIN I guess.

GRV0423, 2023-05-08:
I think pin not supported yet... thanks

GRV0423, 2023-05-09:
is this normal?

'  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 874, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
16:42:34 INFO [sampling] batteryiii volt=3409,3398,3408,3408,3398,3399,3398,3408,3400,3399,3449,3282,3471,3400,3399,3391 temp=[31, 31]
16:42:34 INFO [sampling] DalyBt(B7:7B:02:02:09:65) times: connect=0.00s fetch=2.33s
16:42:34 ERROR [message_bus] A message handler raised an exception: invalid state.
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/dbus_fast/message_bus.py", line 811, in _process_message
    result = user_handler(msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 874, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
16:42:34 INFO [sampling] batteryi volt=3408,3405,3398,3400,3398,3398,3398,3398,3398,3401,3409,3393,3398,3405,3402,3396 temp=[28, 28]
16:42:34 INFO [sampling] DalyBt(C7:7C:03:02:0D:E9) times: connect=0.00s fetch=2.33s
16:42:35 INFO [sampling] batteryiii: BmsSampl(100.0%,U=54.4V,I=0.30A,P=16W,q=295.0Ah/295,mos=nan°C)
16:42:36 INFO [sampling] batteryi: BmsSampl(100.0%,U=54.4V,I=0.30A,P=16W,q=260.0Ah/260,mos=nan°C)
16:42:36 INFO [sampling] batteryii: BmsSampl(99.8%,U=54.4V,I=0.00A,P=0W,q=279.4Ah/280,mos=nan°C)
16:42:37 INFO [sampling] batteryii volt=3404,3402,3402,3400,3402,3403,3403,3402,3402,3402,3403,3401,3402,3403,3402,3400 temp=[28, 28]
16:42:37 ERROR [message_bus] A message handler raised an exception: invalid state.
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/dbus_fast/message_bus.py", line 811, in _process_message
    result = user_handler(msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 874, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
16:42:37 INFO [sampling] batteryiii volt=3409,3398,3409,3400,3403,3399,3399,3409,3399,3399,3400,3409,3398,3398,3399,3390 temp=[31, 31]
16:42:37 INFO [sampling] DalyBt(B7:7B:02:02:09:65) times: connect=0.00s fetch=2.33s
16:42:38 ERROR [message_bus] A message handler raised an exception: invalid state.
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/dbus_fast/message_bus.py", line 811, in _process_message
    result = user_handler(msg)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 874, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
16:42:38 INFO [sampling] batteryi volt=3408,3405,3398,3400,3389,3415,3392,3398,3398,3405,3398,3399,3398,3405,3402,3396 temp=[28, 28]
16:42:38 INFO [sampling] DalyBt(C7:7C:03:02:0D:E9) times: connect=0.00s fetch=2.33s
16:42:39 INFO [sampling] batteryiii: BmsSampl(100.0%,U=54.4V,I=0.30A,P=16W,q=295.0Ah/295,mos=nan°C)
16:42:39 INFO [sampling] batteryi: BmsSampl(100.0%,U=54.4V,I=0.30A,P=16W,q=260.0Ah/260,mos=nan°C)
16:42:40 INFO [sampling] batteryii: BmsSampl(99.8%,U=54.4V,I=0.00A,P=0W,q=279.4Ah/280,mos=nan°C)
16:42:40 INFO [sampling] batteryii volt=3404,3402,3403,3400,3403,3402,3402,3402,3402,3402,3403,3401,3402,3402,3402,3400 temp=[28, 28]
16:42:41 ERROR [message_bus] A message handler raised an exception: invalid state.
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/dbus_fast/message_bus.py", line 811, in _process_message
    result = user_handler(msg)
[... 2 lines cut ...]
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
[... cut ...]
    callback(bytearray(value))
  File "/app/bmslib/daly.py", line 67, in _notification_callback
    self._fetch_futures.set_result(command, response_bytes)
  File "/app/bmslib/__init__.py", line 30, in set_result
    fut.set_result(value)
asyncio.exceptions.InvalidStateError: invalid state
16:42:41 INFO [sampling] batteryi volt=3408,3405,3398,3400,3397,3398,3397,3398,3398,3396,3417,3390,3398,3405,3402,3396 temp=[28, 28]
16:42:41 INFO [sampling] DalyBt(C7:7C:03:02:0D:E9) times: connect=0.00s fetch=2.33s
''

fl4p, 2023-05-09:
I'll fix this with the next update