fl4p/batmon-ha #100: Interface org.bluez.Adapter1 doesn't exist

hemocode, 2023-05-10:
07:57:09 INFO [bt] BT Discovery:
07:57:09 ERROR [main] Error discovering devices: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
07:57:09 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
07:57:09 INFO [bt] BMS DalyBt(96:69:08:01:0C:18) keep alive enabled
07:57:09 INFO [bt] BMS DalyBt(17:71:06:04:02:E0) keep alive enabled
07:57:09 INFO [bt] BMS DalyBt(17:71:06:06:02:64) keep alive enabled
07:57:09 INFO [main] connecting mqtt mqtt@core-mosquitto
07:57:09 INFO [main] Fetching 3 BMS + 0 others serially, period=1.00s, keep_alive=True
07:57:09 INFO [sampling] connecting bms DalyBt(96:69:08:01:0C:18)
07:57:09 INFO [main] mqtt watchdog loop started with timeout 900.0s
07:57:09 INFO [daly] normal connect failed ([org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
), connecting with scanner
07:57:09 ERROR [sampling] bat1 error: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
07:57:09 INFO [sampling] Bleak version 0.13.1a1
07:57:09 INFO [sampling] connecting bms DalyBt(17:71:06:04:02:E0)
07:57:09 INFO [daly] normal connect failed ([org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
), connecting with scanner
07:57:09 ERROR [sampling] bat2 error: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
07:57:09 INFO [sampling] Bleak version 0.13.1a1
07:57:09 INFO [sampling] connecting bms DalyBt(17:71:06:06:02:64)
07:57:09 INFO [daly] normal connect failed ([org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
), connecting with scanner
07:57:09 ERROR [sampling] bat3 error: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
07:57:09 INFO [sampling] Bleak version 0.13.1a1
07:57:09 INFO [sampling] connecting bms DalyBt(96:69:08:01:0C:18)
07:57:09 INFO [daly] normal connect failed ([org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
), connecting with scanner

fl4p, 2023-05-10:
https://github.com/fl4p/batmon-ha#troubleshooting

hemocode, 2023-05-10:
Thank's, but the console output containы errors.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
12:06:59 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
Collecting bleak==0.20.2
  Downloading bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 608.1 kB/s eta 0:00:00
Collecting async-timeout<5,>=3.0.0
  Downloading async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
Collecting dbus-fast<2.0.0,>=1.83.0
  Downloading dbus_fast-1.86.0-cp310-cp310-musllinux_1_1_x86_64.whl (5.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.0/5.0 MB 648.4 kB/s eta 0:00:00
Installing collected packages: async-timeout, dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed async-timeout-4.0.2 bleak-0.20.2 dbus-fast-1.86.0
**WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv**
12:07:13 INFO [main] Power cycle bluetooth hardware

fl4p, 2023-05-17:
Can you paste your BMS config ?

fl4p, 2023-07-30:
I guess that the problem here is the `adapter: hci1` setting. Try `adapter: hci0`

closed 2023-07-30 by fl4p