fl4p/batmon-ha #103: No connection to JBD device after updates from 0.0.57

BetaTobi, 2023-05-16:
Hi,

I have using batmon since quite a while, without any issues.
The last working version is 0.0.57.
But after a batmon update to a higher version (0.0.62, 0.0.63) the connection to my JBD device fails.
No connection possible.
A backup-restore to the 0.0.57 version fixes the issue and connection is again possible.
Another update to a higher version leads to the same connection issue.
See logs below.

Any idea?

Greetz

BEGIN 
20:09:06 ERROR [sampling] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
20:09:06 INFO [sampling] Bleak version 0.13.1a1
20:09:22 ERROR [sampling] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
20:09:22 INFO [sampling] Bleak version 0.13.1a1
20:09:22 ERROR [main] 1 exceptions occurred fetching BMSs
20:09:22 ERROR [main] Error (num 1, max 200) reading BMS: 
20:09:22 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 95, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
  File "/app/bmslib/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
20:09:39 ERROR [sampling] battery1 error: <class 'asyncio.exceptions.TimeoutError'>
20:09:39 INFO [sampling] Bleak version 0.13.1a1
20:09:39 ERROR [main] 1 exceptions occurred fetching BMSs
20:09:39 ERROR [main] Error (num 2, max 200) reading BMS: 
20:09:39 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 95, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
  File "/app/bmslib/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
END
```

damianarnold1993, 2023-05-22:
same here.....

Liontron 100Ah Battery with a JBD BMS

GiampaoloSica, 2023-06-04:
same for me:

09:54:53 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
09:54:53 ERROR [main] Main loop exception: 'JbdBt' object has no attribute 'logger'
09:54:53 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 312, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 155, in main
    bms_list.append(bms_class(addr,
                    ^^^^^^^^^^^^^^^
  File "/app/bmslib/jbd.py", line 32, in __init__
    self.logger.warning('JBD usually does not use a pairing PIN')
    ^^^^^^^^^^^
AttributeError: 'JbdBt' object has no attribute 'logger'
09:54:53 INFO [main] exit signal handler... (), {}, shutdown already False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

BetaTobi, 2023-07-30:
Issue still existing, even after update to 0.0.64.

fl4p, 2023-07-30:
did you try to remove the pin?

fl4p, 2023-07-30:
> Issue still existing, even after update to 0.0.64.

Does a downgrade to 0.0.57 help?

BetaTobi, 2023-07-31:
> did you try to remove the pin?

Which pin?

> Does a downgrade to 0.0.57 help?

I did not downgrade to 0.0.57 only.
I restored my whole Home Assistant using a backup, where Batmon is on 0.0.57.
This works fine (to test the new Batmon versions, but is a bit "unhandy" :-) )

fl4p, 2023-08-01:
> > did you try to remove the pin?
> 
> Which pin?

Nvm, this was for @GiampaoloSica 

> > Does a downgrade to 0.0.57 help?
> 
> I did not downgrade to 0.0.57 only. I restored my whole Home Assistant using a backup, where Batmon is on 0.0.57. This works fine (to test the new Batmon versions, but is a bit "unhandy" :-) )

Please follow the steps here https://github.com/fl4p/batmon-ha/blob/master/doc/Downgrade.md and try 0.0.60 .

BetaTobi, 2023-08-02:
> Please follow the steps here https://github.com/fl4p/batmon-ha/blob/master/doc/Downgrade.md and try 0.0.60

Okay, i did this.
However, it still does not work.

I did the following:

- Deinstall and downgrade, as discribed above
- Tested local add-on with 0.0.60
- Tested local add-on with 0.0.57

Nothing worked.

**The log of the non-working local 0.060 is this (complete, only personal data is renamed):**

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
09:01:52 INFO [bt] BT Discovery:
09:01:57 INFO [bt]  - no devices found - 
09:01:57 INFO [bt] BMS JbdBt(MAC-ADDRESS) keep alive enabled
09:01:57 INFO [main] connecting USER@core-mosquitto
09:01:58 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
09:01:58 INFO [sampling] connecting bms JbdBt(MAC-ADDRESS)
09:01:58 INFO [main] mqtt watchdog loop started with timeout 900.0s
09:02:22 ERROR [sampling] battery1 error: <class 'TimeoutError'>
09:02:22 INFO [sampling] connecting bms JbdBt(MAC-ADDRESS)
09:02:26 INFO [sampling] connected bms JbdBt(MAC-ADDRESS)!
09:02:42 ERROR [sampling] battery1 error: <class 'TimeoutError'>
09:02:42 ERROR [main] 1 exceptions occurred fetching BMSs
09:02:42 ERROR [main] Error (num 1, max 200) reading BMS: 
09:02:42 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    await fn()
  File "/app/main.py", line 264, in fn
    raise exceptions[0]
  File "/app/main.py", line 259, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 92, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 48, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

**The log of the non-working local 0.057 is this (complete, only personal data is renamed):**

[... 73 lines cut ...]
    raise exceptions[0]
  File "/app/main.py", line 234, in fn
    await t
  File "/app/bmslib/sampling.py", line 51, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 74, in sample
    sample = await bms.fetch()
  File "/app/bmslib/jbd.py", line 73, in fetch
    buf = await self._q(cmd=0x03)
  File "/app/bmslib/jbd.py", line 67, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 46, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
10:21:11 INFO [sampling] battery1 subscribing for {'discharge': True, 'charge': True} switch change
10:21:11 INFO [mqtt_util] subscribe homeassistant/switch/battery1/discharge/set
10:21:11 INFO [mqtt_util] subscribe homeassistant/switch/battery1/charge/set
10:21:11 INFO [sampling] battery1: BmsSample(U=26.0V,I=0.00A,P=0W,q=1.7Ah/20.0,mos=nan°C)
10:21:11 INFO [sampling] battery1 volt=3265,3252,3241,3257,3252,3261,3217,3232 temp=[21.8, 21.6]

I am happy to test other ideas :-)

Greetz

BetaTobi, 2023-09-07:
Issue still existing with 0.0.69.

I tried an update and a complete new install (aber deinstall of add-on).
I also tried the new option "install_newer_bleak".

But still no connection possible.

Only help is the restore of my HomeAssistant back to the batmon version 0.0.57

fl4p, 2023-09-07:
hm that's odd.
you restore your backup, it works, and then you update the add-on (and nothing else) and it stops working?

BetaTobi, 2023-09-07:
Yes, exactly.
It is working fine with 0.0.57.
But when I do an update to a higher version (independent to which one, I try this with every release), it stops working.

I do nothing else, just the update, and it stops working.. :-/

fl4p, 2023-09-07:
what happens if you un-install and re-install that same version 0.0.57?

fl4p, 2023-09-07:
Version 0.0.60 has a bug, you can see it in the logs. So please try again that version as well, with a modification in main.py:
comment out this line
```
meter_state=meter_states.get(bms.name), 
```
so it becomes
```
# meter_state=meter_states.get(bms.name), 
```

BetaTobi, 2023-09-07:
> what happens if you un-install and re-install that same version 0.0.57?

The "re-install" of 0.0.57 is then only possible via local-addon (as described in  https://github.com/fl4p/batmon-ha/blob/master/doc/Downgrade.md ), correct?

fl4p, 2023-09-07:
yes

BetaTobi, 2023-09-07:
OK.
I uninstalled the working 0.0.57.
Then I downloaded the local 0,0.57 and I changed the "meter_state" line in main.py (comment out, as described by you).
Then I installed the local 0.0.57

And it works.
Fine.

Next? :-)

Update to the current 0,069?
Or do you prefer an "intermediate step"?

Greetz

BetaTobi, 2023-09-07:
So, in the meantime I tried the following:

- Additional "normal" 0.0.69 from the add-on store --> does not work
- Local 0.0.60 --> does not work

So it seems, that something changed from 0.0.57 to 0.0.60.
Any ideas?

fl4p, 2023-09-14:
Hm there were quite some changes from v57 to v60.
related are:
* `jbd.py`: 
`TIMEOUT = 16`. you can set this back to 8 and try again

 * `requirements.txt`: 
`git+https://github.com/jpeters-ml/bleak@feature/windowsPairing`
became
`bleak==0.20.2`

you can revert these in you local copy.

are there any other errors than the `TimeoutError` in the logs?

BetaTobi, 2023-09-17:
The only other errors are the following:
14:26:55 ERROR [main] 1 exceptions occurred fetching BMSs
14:26:55 ERROR [main] Error (num 1, max 200) reading BMS: 
14:26:55 ERROR [main] Stack: Traceback (most recent call last):

But they are the same for both the 0,0.57 and 0.0.60.
See attached log-files.

I will try your ideas with the changes within the source code and will give a feedback then.

[local_batmon_0057.log](https://github.com/fl4p/batmon-ha/files/12642910/local_batmon_0057.log)
[local_batmon_0060.log](https://github.com/fl4p/batmon-ha/files/12642911/local_batmon_0060.log)

Ghostkls, 2023-10-09:
Currently experiencing the same issue using a remote PI for connection to Revov battery using Bluetooth. 0.54 is last working version. Upgrading to anything higher does not work currently.

BetaTobi, 2023-10-11:
> you can revert these in you local copy.

A question here: Is it sufficient to make the changes "after" installation of the add-on? Or do I have to make the changes before the installation?

Ghostkls, 2023-12-08:
Not sure if this will work for everyone but I have managed to get my JDB bluetooth working on build 1.74 but not before following a few additional steps detailed below
Removed/renamed the current batmon folder
Followed standalone guide to clone repository and run PIP
Copied/Created Options.Json file to the batmon directory
Ran the following command to delete the cached bluetooth device: sudo bluetoothctl -- remove AA:BB:CC:DD:EE:FF (Obviously replace with correct device MAC)
Start using either python3 main.py or systemctl start service

With the below said, I have noted that the service does not seem to survive a reboot and I need to run the bluetooth remove command again before starting the service which I still need to unpack further. For now a daily script to stop, remove and start is keeping this "alive"