fl4p/batmon-ha #104: Error connecting to victron SmartShunt

MaxiStarling56, 2023-05-18:
Hi, I've just installed the Batmon plugin to use with my Victron SmartShunt, the device seems to connect but the plugin cannot comunicate properly, any ideas? thanks.
(I've enabled GATT in the mobile app before trying to connect)

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
21:40:40 INFO [main] Power cycle bluetooth hardware
21:40:44 INFO [bt] BT Discovery:
21:40:49 INFO [bt] BT Device   SmartShunt HQ*******VF   address=E2:**:**:**:**:94
21:40:49 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
21:40:49 INFO [bt] BMS SmartShuntBt(E2:**:**:**:**:94) keep alive enabled
21:40:49 INFO [main] connecting mqtt mqtt_user@core-mosquitto
21:40:49 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
21:40:49 INFO [sampling] connecting bms SmartShuntBt(E2:**:**:**:**:94)
21:40:49 INFO [main] mqtt watchdog loop started with timeout 900.0s
21:40:50 ERROR [sampling] smart_shunt error: [org.bluez.Error.Failed] le-connection-abort-by-local
21:40:50 INFO [sampling] Bleak version 0.13.1a1
21:40:50 INFO [sampling] connecting bms SmartShuntBt(E2:**:**:**:**:94)
21:40:51 ERROR [sampling] smart_shunt error: [org.bluez.Error.Failed] le-connection-abort-by-local
21:40:51 INFO [sampling] Bleak version 0.13.1a1
21:40:51 ERROR [main] 1 exceptions occurred fetching BMSs
21:40:51 ERROR [main] Error (num 1, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
21:40:51 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 89, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 250, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 153, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 121, in _connect_client
    await self.client.connect(timeout=timeout)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
```

I think the main error is "le-connection-abort-by-local".

fl4p, 2023-05-21:
Have you tried to enable `bt_power_cycle` in the add-on configuration?

MaxiStarling56, 2023-05-22:
Yes, it was enabled.

MaxiStarling56, 2023-05-28:
Hi, i was trying to run the project on my laptop, now it's displaying the following error on loop:
```
ERROR [main] Error (num 3, max 200) reading BMS: __init__() missing 2 required positional arguments: 'voltage' and 'current'
```
I remember also seeing this error on HA but i wasn't able to reproduce it yet, is this usefull in any way? 
Thanks.

dominikandreas, 2023-05-29:
I've had the same above error before enabling GATT (see readme / product info page in victron connect app). After setting the pin in the configuration, I'm now able to connect to the smart shunt

dezfutak, 2023-06-08:
> I've had the same above error before enabling GATT (see readme / product info page in victron connect app). After setting the pin in the configuration, I'm now able to connect to the smart shunt

Confirmed here: Update the Victron Connect app & was able to enable GATT. Did a couple of smartshunt firmware updates too (not sure they were needed) & subsequent to the GATT-enabling, BATMON is now connecting to the smartshunt & pushing its data via MQTT.

closed 2023-10-20 by fl4p