fl4p/batmon-ha #105: Question on Energy dashboard, Inverted charging and discharging meter and Batteries grouping

Hare75, 2023-05-20:
Hi, 

I have 2 set of batteries running parallel with Daly BMS on each set. After some tinkering, I manage to get the BMS readings via this superb add-on to the HAos install on a NUC. I am totally new to all this and came across a few issues. Hope you could guide me.

Since getting the readings, I proceeded to add it to the “Batteries” section on energy dashboard but after 2 days of testing, nothing is showing up. Any idea what I did wrong? 

I have also notice when the batteries are “charging”, the value in “Total energy discharge meter” will add up while the value in “Total energy charge meter” will stay stationary. During battery “discharge” it will be the other way round. Is this normal? If not, how can I correct this?

I have also try to group the batteries with the codes provided, But after saving and restarting. Nothing seems to happen. I know this is an experimental function so i just wanna confirm if there anything I miss?

Thank You.

fl4p, 2023-05-21:
Hi,
did you add the meters under Settings / Dashboard / Energy / Add Battery System ?

Can you post your add-on configuration here? (Batmon / Configuration / "Edit in Yaml" (click on the three dots on the upper right)
When charging, what is the reading of the battery current batmon reports positive or negative?

Hare75, 2023-05-22:
Hello, 

i did add the meters to the energy dashboard as mention 

[image]

**My add-on configuration is as below:-**

devices:
  - address: C6:6C:09:05:01:1F
    type: daly
    alias: DL-C66C0905011F
  - address: C6:6C:09:02:07:66
    type: daly
    alias: DL-C66C09020766
mqtt_user: user1234
mqtt_password: pass1234
mqtt_broker: core-mosquitto
concurrent_sampling: true
keep_alive: true
verbose_log: false
sample_period: 1
publish_period: 2
invert_current: false
watchdog: true
expire_values_after: 20
install_newer_bleak: false
bt_power_cycle: false

**The "charging" current was originally showing a Negative value. I have used the "Invert current" function as my preferred setup.**

**When i start Batmon, i have also come across some errors on the the HA system log with many thousands of occurrences.** 

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:411
Integration: MQTT (documentation, issues)
First occurred: 2:04:20 AM (56884 occurrences)
Last logged: 12:44:01 PM

JSON result was not a dictionary

**The error below was also found once in the HA system log:-**

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:125
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 1:21:50 AM (1 occurrences)
Last logged: 1:21:50 AM

Failed to to call /addons/2af0a32d_batmon/stats - 404 Client Error for http+docker://localhost/v1.42/containers/addon_2af0a32d_batmon/json: Not Found ("No such container: addon_2af0a32d_batmon")

Thanks

fl4p, 2023-05-22:
I'll fix `JSON result was not a dictionary` the next update. You can ignore, it is not related to the issue.

What model of Daly BMS do you use? Looks like it reports the current inverse to the model if have.
Can you connect with the SmartBMS app to check the sign of the current when charging?
With the model i have it is negative when charging.

Hare75, 2023-05-22:
I am using a 8S 100A Daly Lipo4. 

Attach is the Screen shot from the SMARTBMS app while charging.

[image]

[image]

fl4p, 2023-06-08:
I fix that with the next release. it'll introduce a current correction factor which you can set to -1 to flip the sign of the current

joosthb, 2023-06-08:
That's awesome. Thanks for your great project, works like a charm in my campervan 🚐

fl4p, 2023-07-25:
0.0.64 adds `current_calibration`, set this to `-1`

closed 2023-07-25 by fl4p

mazurek62, 2023-08-19:
> I experience the same behaviour. I am measuring the in-/output currents externally with a shunt as well and i can confirm the total energy discharge meter and charge meter are mixed up. While charging the discharge meter will increase, while discharging, the charging meter will increase.
> 

After update 0.0.64 and setting `invert_current: true`  the current is inverted, but unfortunately this doesn't affect the charge meter and discharge meter. In my case it is the same as above, discharging meter goes up while charging and chagring meter goes up while discharging.

You mention here `current_calibration` parameter, but in the Yaml settings the `invert_current` parameter is available. Is it just a change of names?

BMS Model: Daly BMS Smart Li-Ion 7S 24V 100A

fl4p, 2023-08-20:
`current_calibration` is for each device. It will be multiplied with the current reading before metering. Use it for calibration and to fix the energy meters direction.

`invert_current` is applied after metering and to all devices. Use it to set your taste of current definition.

mazurek62, 2023-08-21:
The `current_calibration` doesn't seem to be affecting the `total energy charge meter` or  `total energy discharge meter`. 
Changing the value between `-1` and `1` affects only the current sign similarly to `invert_current`. 

Any combination of these two options doesn't flip the charge/discharge entities.

I'm doing all this while the battery pack is being discharged and the `total energy charge meter` goes up.

fl4p, 2023-08-21:
Are you sure you put `current_calibration` in the device settings? Please post your yaml here.

mazurek62, 2023-08-22:
Here is my yaml:

devices:
  - address: 40:17:05:01:5B:83
    type: daly
    alias: Magazyn
    current_calibration: -1
mqtt_user: mqtt
mqtt_password: ******
mqtt_broker: core-mosquitto
concurrent_sampling: false
keep_alive: true
verbose_log: false
sample_period: 1
publish_period: 2
invert_current: false
watchdog: true
expire_values_after: 20
install_newer_bleak: false
bt_power_cycle: true

Hare75, 2023-08-25:
Hello again, As mention above i do have 2 set of batteries running parallel with Daly BMS on each set. After awhile of running them, i notice that the figures on the "Total Charge meter" and "Total Energy Meter" of one bank is reducing while the other bank is adding up. I am not sure why i am having this problem. Maybe you could shed some light on this matter. My YAML is still the same as above.  Thanks for the "JSON result was not a dictionary" fix.

Cheers.

[image]

fl4p, 2023-08-29:
i pushed another fix, which should fix this @Hare75 (version 0.0.68)

Hare75, 2023-08-30:
After the version 0.0.68 update, both battery banks seems to be adding up correctly. Will keep observing it for awhile. Thanks for the fix.